syssi/esphome-seplos-bms #1: Has this been tested?

NosIreland, 2022-04-18:
Hi,
I have Seplos BMS and looking to for ways to monitor it. I have found this repo but I am lacking details. I see esphome template but this looks a bit strange to me since I do not see MODBUS anywhere and BMS is using MODBUS ASCII to communicate. Has this been verified as working? As I understand I do nee UART to RS485 converter between ESP and Seplos.
Thanks

syssi, 2022-04-18:
This project isn't ready yet. I'm looking for somebody willing to execute the config yaml example and generate same traffic. As soon some valid traffic captures are available it's easy to bring this project forward.

Do you like to give it a try? I will provide some instructions.

NosIreland, 2022-04-18:
Sure. Send me the details.

syssi, 2022-04-18:
Could you provide the model name of your BMS?

As first step please prepare something like this: https://raw.githubusercontent.com/syssi/esphome-soyosource-gtn-virtual-meter/main/images/circuit.jpg

Please ping me if you are ready.

NosIreland, 2022-04-19:
I have device created.

syssi, 2022-04-19:
Could you tell me the model name of your device? Do you know the which pin of the RJ45 jack is `A+` and `B-`? Please attach the RS485 module to your BMS.

The next step:

1. Install `esphome`
2. Download this example yaml: https://raw.githubusercontent.com/syssi/esphome-seplos-bms/main/esp8266-example.yaml
3. Compile and flash the yaml by `esphome run esp8266-example.yaml` to your ESP

You should see some log output now. The ESP sends periodically these 7 instructions to the BMS:

https://github.com/syssi/esphome-seplos-bms/blob/main/esp8266-example.yaml#L42-L66

The BMS should respond to every instruction with a message. Please provide the log containing the requests + responses!

NosIreland, 2022-04-28:
Sorry for delay. Here are the logs. The model is 1101-SP15
Here are the logs looks like it is reponding to some but no all requests:
```
INFO Reading configuration /config/bms.yaml...
INFO Starting log output from seplos-bms/debug
INFO Connected to MQTT broker!
[20:05:47][D][uart_debug:158]: >>> "~200246470000FDA7\r"
[20:05:48][D][uart_debug:158]: >>> "~200246920000FDA7\r"
[20:05:50][D][uart_debug:158]: >>> "~200246930000FDA6\r"
[20:05:52][D][uart_debug:158]: >>> "~20024642E002FFFD09\r"
[20:06:00][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[20:06:00][D][uart_debug:158]: <<< "~2000460010960001100CD70CE90CF40CD60CEF0CE50CE10CDC0CE90CF00CE80CEF0CEA0CDA0CDE0CD8060BA60BA00B970BA60BA50BA2FD5C14A0344E0A426803134650004603E8149F000"
[20:06:00][D][uart_debug:158]: <<< "0000000000000DC6C\r"
[20:06:02][D][uart_debug:158]: >>> "~2000464F0000FD9A\r"
[20:06:02][D][uart_debug:158]: <<< "~200046000000FDB4\r"
[20:06:04][D][uart_debug:158]: >>> "~200046510000FDAE\r"
[20:06:04][D][uart_debug:158]: <<< "~20004600C040313130312D5350313520020743414E50726F746F636F6C3A536F666172202020F046\r"
[20:06:06][D][uart_debug:158]: >>> "~200246470000FDA7\r"
[20:06:08][D][uart_debug:158]: >>> "~200246920000FDA7\r"
[20:06:10][D][uart_debug:158]: >>> "~200246930000FDA6\r"
[20:06:12][D][uart_debug:158]: >>> "~20024642E002FFFD09\r"
[20:06:20][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[20:06:20][D][uart_debug:158]: <<< "~2000460010960001100CD80CE80CF40CD70CEE0CE50CE10CDD0CE90CF00CE80CEF0CEB0CDA0CDE0CD9060BA60BA00B970BA60BA50BA2FD7214A0344A0A426803134650004603E8149F000"
[20:06:20][D][uart_debug:158]: <<< "0000000000000DC7C\r"
[20:06:22][D][uart_debug:158]: >>> "~2000464F0000FD9A\r"
[20:06:22][D][uart_debug:158]: <<< "~200046000000FDB4\r"
[20:06:24][D][uart_debug:158]: >>> "~200046510000FDAE\r"
[20:06:24][D][uart_debug:158]: <<< "~20004600C040313130312D5350313520020743414E50726F746F636F6C3A536F666172202020F046\r"
[20:06:26][D][uart_debug:158]: >>> "~200246470000FDA7\r"
[20:06:28][D][uart_debug:158]: >>> "~200246920000FDA7\r"
[20:06:30][D][uart_debug:158]: >>> "~200246930000FDA6\r"
[20:06:32][D][uart_debug:158]: >>> "~20024642E002FFFD09\r"
[20:06:40][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[20:06:40][D][uart_debug:158]: <<< "~2000460010960001100CD70CE80CF40CD70CEF0CE50CE10CDC0CE80CF00CE80CEF0CEA0CDA0CDE0CD9060BA60BA00B970BA60BA60BA2FD7C14A034460A426803134650004603E8149F000"
[20:06:40][D][uart_debug:158]: <<< "0000000000000DC78\r"
[20:06:42][D][uart_debug:158]: >>> "~2000464F0000FD9A\r"
[20:06:42][D][uart_debug:158]: <<< "~200046000000FDB4\r"
[20:06:44][D][uart_debug:158]: >>> "~200046510000FDAE\r"
[20:06:44][D][uart_debug:158]: <<< "~20004600C040313130312D5350313520020743414E50726F746F636F6C3A536F666172202020F046\r"
[20:06:46][D][uart_debug:158]: >>> "~200246470000FDA7\r"
[20:06:48][D][uart_debug:158]: >>> "~200246920000FDA7\r"
[20:06:50][D][uart_debug:158]: >>> "~200246930000FDA6\r"
[20:06:52][D][uart_debug:158]: >>> "~20024642E002FFFD09\r"
[20:07:00][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[20:07:00][D][uart_debug:158]: <<< "~2000460010960001100CD70CE80CF30CD60CEE0CE50CE00CDC0CE80CEF0CE70CEE0CE90CD90CDD0CD8060BA60BA00B970BA60BA60BA2FD60149F34420A426803124650004603E8149F000"
[20:07:00][D][uart_debug:158]: <<< "0000000000000DC86\r"
[20:07:02][D][uart_debug:158]: >>> "~2000464F0000FD9A\r"
[20:07:02][D][uart_debug:158]: <<< "~200046000000FDB4\r"
[20:07:04][D][uart_debug:158]: >>> "~200046510000FDAE\r"
[20:07:04][D][uart_debug:158]: <<< "~20004600C040313130312D5350313520020743414E50726F746F636F6C3A536F666172202020F046\r"
[20:07:05][I][ota:105]: Boot seems successful, resetting boot loop counter.
[20:07:06][D][uart_debug:158]: >>> "~200246470000FDA7\r"
[20:07:08][D][uart_debug:158]: >>> "~200246920000FDA7\r"
[20:07:10][D][uart_debug:158]: >>> "~200246930000FDA6\r"
[20:07:12][D][uart_debug:158]: >>> "~20024642E002FFFD09\r"
[20:07:20][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[20:07:20][D][uart_debug:158]: <<< "~2000460010960001100CD70CE80CF40CD60CEE0CE50CE00CDC0CE80CF00CE70CEE0CE90CD90CDD0CD8060BA60BA00B970BA60BA70BA2FD4F149F343F0A426803124650004603E8149E000"
[20:07:20][D][uart_debug:158]: <<< "0000000000000DC73\r"
[20:07:22][D][uart_debug:158]: >>> "~2000464F0000FD9A\r"
[20:07:22][D][uart_debug:158]: <<< "~200046000000FDB4\r"
[20:07:24][D][uart_debug:158]: >>> "~200046510000FDAE\r"
[20:07:24][D][uart_debug:158]: <<< "~20004600C040313130312D5350313520020743414E50726F746F636F6C3A536F666172202020F046\r"
[20:07:26][D][uart_debug:158]: >>> "~200246470000FDA7\r"
[20:07:28][D][uart_debug:158]: >>> "~200246920000FDA7\r"
[20:07:30][D][uart_debug:158]: >>> "~200246930000FDA6\r"
[20:07:32][D][uart_debug:158]: >>> "~20024642E002FFFD09\r"
[20:07:40][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[20:07:40][D][uart_debug:158]: <<< "~2000460010960001100CD70CE70CF30CD60CEE0CE40CE00CDB0CE80CEF0CE70CEE0CEA0CD90CDD0CD8060BA60BA00B970BA60BA70BA2FD49149F343B0A426803124650004603E8149E000"
[20:07:40][D][uart_debug:158]: <<< "0000000000000DC6B\r"
[20:07:42][D][uart_debug:158]: >>> "~2000464F0000FD9A\r"
[... 27 lines cut ...]
[20:08:30][D][uart_debug:158]: >>> "~200246930000FDA6\r"
[20:08:32][D][uart_debug:158]: >>> "~20024642E002FFFD09\r"
[20:08:40][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[20:08:40][D][uart_debug:158]: <<< "~2000460010960001100CD70CE70CF30CD60CEE0CE40CE00CDB0CE70CF00CE60CEE0CE90CD90CDD0CD8060BA60BA00B970BA60BA70BA2FD6A149F34300A426803114650004603E8149E000"
[... cut ...]
[20:09:02][D][uart_debug:158]: >>> "~2000464F0000FD9A\r"
[20:09:02][D][uart_debug:158]: <<< "~200046000000FDB4\r"
[20:09:04][D][uart_debug:158]: >>> "~200046510000FDAE\r"
[20:09:04][D][uart_debug:158]: <<< "~20004600C040313130312D5350313520020743414E50726F746F636F6C3A536F666172202020F046\r"
[20:09:06][D][uart_debug:158]: >>> "~200246470000FDA7\r"
[20:09:08][D][uart_debug:158]: >>> "~200246920000FDA7\r"
[20:09:10][D][uart_debug:158]: >>> "~200246930000FDA6\r"
[20:09:12][D][uart_debug:158]: >>> "~20024642E002FFFD09\r"
```

syssi, 2022-04-28:
Good job! I will try to decode the responses to get a better understanding what's returned and what's missing. The next step will be a small custom component polling the BMS and publishing the retrieved data. I will ping you later the week for further testing.

syssi, 2022-05-01:
```
>>> "~20004642E00200FD37\r"
<<< "~2000460010960001100CD70CE90CF40CD60CEF0CE50CE10CDC0CE90CF00CE80CEF0CEA0CDA0CDE0CD8060BA60BA00B970BA60BA50BA2FD5C14A0344E0A426803134650004603E8149F0000000000000000DC6C\r"

>>> "~20004642E00200FD37\r"
<<< "~2000460010960001100CD80CE80CF40CD70CEE0CE50CE10CDD0CE90CF00CE80CEF0CEB0CDA0CDE0CD9060BA60BA00B970BA60BA50BA2FD7214A0344A0A426803134650004603E8149F0000000000000000DC7C\r"

>>> "~20004642E00200FD37\r"
<<< "~2000460010960001100CD70CE80CF40CD70CEF0CE50CE10CDC0CE80CF00CE80CEF0CEA0CDA0CDE0CD9060BA60BA00B970BA60BA60BA2FD7C14A034460A426803134650004603E8149F0000000000000000DC78\r"

>>> "~20004642E00200FD37\r"
<<< "~2000460010960001100CD70CE80CF30CD60CEE0CE50CE00CDC0CE80CEF0CE70CEE0CE90CD90CDD0CD8060BA60BA00B970BA60BA60BA2FD60149F34420A426803124650004603E8149F0000000000000000DC86\r"

>>> "~20004642E00200FD37\r"
<<< "~2000460010960001100CD70CE80CF40CD60CEE0CE50CE00CDC0CE80CF00CE70CEE0CE90CD90CDD0CD8060BA60BA00B970BA60BA70BA2FD4F149F343F0A426803124650004603E8149E0000000000000000DC73\r"

>>> "~20004642E00200FD37\r"
<<< "~2000460010960001100CD70CE70CF30CD60CEE0CE40CE00CDB0CE80CEF0CE70CEE0CEA0CD90CDD0CD8060BA60BA00B970BA60BA70BA2FD49149F343B0A426803124650004603E8149E0000000000000000DC6B\r"

# Get manufacturer info
>>> "~200046510000FDAE\r"
<<< "~20004600C040313130312D5350313520020743414E50726F746F636F6C3A536F666172202020F046\r"

>>> "~200046510000FDAE\r"
<<< "~20004600C040313130312D5350313520020743414E50726F746F636F6C3A536F666172202020F046\r"

>>> "~200046510000FDAE\r"
<<< "~20004600C040313130312D5350313520020743414E50726F746F636F6C3A536F666172202020F046\r"

>>> "~200046510000FDAE\r"
<<< "~20004600C040313130312D5350313520020743414E50726F746F636F6C3A536F666172202020F046\r"

>>> "~200046510000FDAE\r"
<<< "~20004600C040313130312D5350313520020743414E50726F746F636F6C3A536F666172202020F046\r"
```

syssi, 2022-05-01:
```
# Request Telemetry

Value         Len Description        ASCII

7E.           1   SOI
32.30.        2   Protocol version   20
30.30.        2   Address            00
34.36.        2   Device (CID1)      46
34.32.        2   Function (CID2)    42
45.30.30.32.  4   Length             E002
30.30.        n   Data               00
46.44.33.37.  4   Checksum           FD37
0D            1   EOI

# Response Telemetry

Value         Len Description              ASCII

7E.           1   SOI
32.30.        2   Protocol version         20
30.30.        2   Address                  00
34.36.        2   Device (CID1)            46
30.30.        2   Function (CID2)          00
31.30.39.36.  4   Length                   1096
              n   Data

30.30.        2   Data flag                00
30.31.        2   Command group            01
31.30.        2   Number of cells          1.0 * 16
30.43.44.37.  4   Voltage of cell 1        0CD7 (3287 * 0.001 = 3.287 V)
30.43.45.39.  4   Voltage of cell 2        0CE9 (3305 * 0.001 = 3.305 V)
30.43.46.34.  4   Voltage of cell 3
30.43.44.36.  4   Voltage of cell 4
30.43.45.46.  4   Voltage of cell 5
30.43.45.35.  4   Voltage of cell 6
30.43.45.31.  4   Voltage of cell 7
30.43.44.43.  4   Voltage of cell 8
30.43.45.39.  4   Voltage of cell 9        0CE9 (3305 * 0.001 = 3.305 V)
30.43.46.30.  4   Voltage of cell 10
30.43.45.38.  4   Voltage of cell 11
30.43.45.46.  4   Voltage of cell 12
30.43.45.41.  4   Voltage of cell 13
30.43.44.41.  4   Voltage of cell 14
30.43.44.45.  4   Voltage of cell 15
30.43.44.38.  4   Voltage of cell 16
30.36.        2   Number of temperatures N=6  06
30.42.41.36.  4   Cell temperature 1       0BA6 (2982 * 0.01 °C)
30.42.41.30.  4   Cell temperature 2       0BA0 (2976 * 0.01 °C)
30.42.39.37.  4   Cell temperature 3       0B97 (2967 * 0.01 °C)
30.42.41.36.  4   Cell temperature 4       0BA6 (2982 * 0.01 °C)
30.42.41.35.  4   Cell temperature 5       0BA5 (2981 * 0.01 °C)
30.42.41.32.  4   Cell temperature 6       0BA2 (2978 * 0.01 °C)
46.44.35.43.  4   Charge/discharge current FD5C (64860 * 0.01 = 64.860 A??)
31.34.41.30.  4   Total battery voltage    14A0 (5280 * 0.01 = 52.80 V)
33.34.34.45.  4   Residual capacity        344E (13390 * 0.01 = 13.390 Ah)
30.41.        2   Custom number P=10       0A

34.32.36.38.  4   Battery capacity         4268 (17000 * 0.01 = 170 Ah)
30.33.31.33.  4   State of charge          0313 (787 * 0.01 = 7.87 %?)
34.36.35.30.  4   Rated capacity           4650 (18000 * 0.01 = 180 Ah)
30.30.34.36.  4   Number of cycles         0046 (70)
30.33.45.38.  4   SOH                      03E8 (1000)
31.34.39.46.  4   Port Voltage             149F (5279)
30.30.30.30.  4   Reserved
30.30.30.30.  4   Reserved
30.30.30.30.  4   Reserved
30.30.30.30.  4   Reserved

44.43.36.43.  4   Checksum
0D.           1   EOI
```

syssi, 2022-05-01:
```
# Request vendor info

Value         Len Description        ASCII

7E.           1   SOI
32.30.        2   Protocol version   20
30.30.        2   Address            00
34.36.        2   Device (CID1)      46
35.31.        2   Function (CID2)    51
30.30.30.30.  4   Length             0000
46.44.41.45.  4   Checksum           FDAE
0D.

# Response vendor info

Value         Len Description          ASCII

7E.           1   SOI
32.30.        2   Protocol version     20
30.30.        2   Address              00
34.36.        2   Device (CID1)        46
30.30.        2   Function (CID2)      00
43.30.34.30.  4   Length               C040
33.31.33.31.33.30.33.31.32.44.35.33.35.30.33.31.33.35.32.30.   20  Device Name    313130312D5350313520 = 1101-SP15
30.32.30.37.  4   Version              0207
34.33.34.31.34.45.35.30.37.32.36.46.37.34.36.46.36.33.36.46.36.43.33.41.35.33.36.46.36.36.36.31.37.32.32.30.32.30.32.30.       Manufacturer Name   43414E50726F746F636F6C3A536F666172202020 = "CANProtocol:Sofar   "
46.30.34.36.  4   Checksum
0D.
```

NosIreland, 2022-05-08:
`46.44.35.43.  4   Charge/discharge current FD5C (64860 * 0.01 = 64.860 A??)` the conversion of this one is wrong as it is a signed integer
`30.33.31.33.  4   State of charge          0313 (787 * 0.01 = 7.87 %?)` this shoudl be 78.7%

I got his running now and pulling stats into grafana. 
[image]
It would be nice for ESP to do the conversion and publish the actual values to mqtt instead of HEX string.

syssi, 2022-05-08:
Good job! I've implemented a CRC check yesterday. Converting the bytes to ASCII-Hex is a bit painful but I hope I will finish the implementation next weekend.

syssi, 2022-06-18:
I've pushed a first draft of the external component: https://github.com/syssi/esphome-seplos-bms/blob/main/esp8266-example.yaml

The component requests the telemetry frame periodically and decodes the contents to the log (for now):

```
[18:53:57][I][seplos_bms:024]: Telemetry frame received
[18:53:57][D][seplos_bms:041]: Number of cells: 16
[18:53:57][D][seplos_bms:047]: Cell voltage 1: 3.287 V
[18:53:57][D][seplos_bms:047]: Cell voltage 2: 3.304 V
[18:53:57][D][seplos_bms:047]: Cell voltage 3: 3.316 V
[18:53:57][D][seplos_bms:047]: Cell voltage 4: 3.287 V
[18:53:57][D][seplos_bms:047]: Cell voltage 5: 3.311 V
[18:53:57][D][seplos_bms:047]: Cell voltage 6: 3.301 V
[18:53:57][D][seplos_bms:047]: Cell voltage 7: 3.297 V
[18:53:57][D][seplos_bms:047]: Cell voltage 8: 3.292 V
[18:53:57][D][seplos_bms:047]: Cell voltage 9: 3.304 V
[18:53:57][D][seplos_bms:047]: Cell voltage 10: 3.312 V
[18:53:57][D][seplos_bms:047]: Cell voltage 11: 3.304 V
[18:53:57][D][seplos_bms:047]: Cell voltage 12: 3.311 V
[18:53:57][D][seplos_bms:047]: Cell voltage 13: 3.306 V
[18:53:57][D][seplos_bms:047]: Cell voltage 14: 3.290 V
[18:53:57][D][seplos_bms:047]: Cell voltage 15: 3.294 V
[18:53:57][D][seplos_bms:047]: Cell voltage 16: 3.289 V
[18:53:57][D][seplos_bms:051]: Number of temperature sensors: 6
[18:53:57][D][seplos_bms:058]: Temperature sensor 0: 29.82 °C
[18:53:57][D][seplos_bms:058]: Temperature sensor 1: 29.76 °C
[18:53:57][D][seplos_bms:058]: Temperature sensor 2: 29.67 °C
[18:53:58][D][seplos_bms:058]: Temperature sensor 3: 29.82 °C
[18:53:58][D][seplos_bms:062]: Environment temperature: 29.82 °C
[18:53:58][D][seplos_bms:065]: Mosfet temperature: 29.78 °C
[18:53:58][D][seplos_bms:068]: Current: -6.44 A
[18:53:58][D][seplos_bms:071]: Total battery voltage: 52.80 V
[18:53:58][D][seplos_bms:074]: Residual capacity: 133.82 Ah
[18:53:58][D][seplos_bms:078]: Battery capacity: 170.00 Ah
[18:53:58][D][seplos_bms:081]: State of charge: 78.7 %
[18:53:58][D][seplos_bms:084]: Rated capacity: 180.00 %
[18:53:58][D][seplos_bms:087]: Rated capacity: 70 %
[18:53:58][D][seplos_bms:090]: State of health: 100.0 %
[18:53:58][D][seplos_bms:093]: Port voltage: 52.79 V
```

I would be happy if you give it a try!

syssi, 2022-06-19:
All measurements of the telemetry frame are supported now. Please give it a try!

closed 2022-06-19 by syssi