syssi/esphome-seplos-bms #2: Testing

faizan-elite, 2022-06-08:
Hi Syssi, 
Great work on the project thus far. I tested your code on the Seplos BMS (15s) model 1101-ZH26 running Version 2.02. Connected to the WeMos via RS485. 

I have attached my logs to help you advance your project. Look forward to upcoming changes and decoding on ESP8266 (if possible). 

If you would like some assistance in testing and advancing this project further you may reach out.
[logs_wemos-mini_upload.txt](https://github.com/syssi/esphome-seplos-bms/files/8859171/logs_wemos-mini_upload.txt)

syssi, 2022-06-08:
Thanks for your support! I will ping you next week for some testing!

faizan-elite, 2022-06-08:
Sure. FYI my BMS has the following ports: 
RS 232 (2400 Baud) 
2 x RS 485 (9600 Baud) 

So far i have been able to decode the responses from the telemetry and communication commands via NodeRed. (Still working on decoding some of those 20bit warning responses).

syssi, 2022-06-18:
I've pushed a first draft of the external component: https://github.com/syssi/esphome-seplos-bms/blob/main/esp8266-example.yaml

The component requests the telemetry frame periodically and decodes the contents to the log (for now):

```
[18:53:57][I][seplos_bms:024]: Telemetry frame received
[18:53:57][D][seplos_bms:041]: Number of cells: 16
[18:53:57][D][seplos_bms:047]: Cell voltage 1: 3.287 V
[18:53:57][D][seplos_bms:047]: Cell voltage 2: 3.304 V
[18:53:57][D][seplos_bms:047]: Cell voltage 3: 3.316 V
[18:53:57][D][seplos_bms:047]: Cell voltage 4: 3.287 V
[18:53:57][D][seplos_bms:047]: Cell voltage 5: 3.311 V
[18:53:57][D][seplos_bms:047]: Cell voltage 6: 3.301 V
[18:53:57][D][seplos_bms:047]: Cell voltage 7: 3.297 V
[18:53:57][D][seplos_bms:047]: Cell voltage 8: 3.292 V
[18:53:57][D][seplos_bms:047]: Cell voltage 9: 3.304 V
[18:53:57][D][seplos_bms:047]: Cell voltage 10: 3.312 V
[18:53:57][D][seplos_bms:047]: Cell voltage 11: 3.304 V
[18:53:57][D][seplos_bms:047]: Cell voltage 12: 3.311 V
[18:53:57][D][seplos_bms:047]: Cell voltage 13: 3.306 V
[18:53:57][D][seplos_bms:047]: Cell voltage 14: 3.290 V
[18:53:57][D][seplos_bms:047]: Cell voltage 15: 3.294 V
[18:53:57][D][seplos_bms:047]: Cell voltage 16: 3.289 V
[18:53:57][D][seplos_bms:051]: Number of temperature sensors: 6
[18:53:57][D][seplos_bms:058]: Temperature sensor 0: 29.82 °C
[18:53:57][D][seplos_bms:058]: Temperature sensor 1: 29.76 °C
[18:53:57][D][seplos_bms:058]: Temperature sensor 2: 29.67 °C
[18:53:58][D][seplos_bms:058]: Temperature sensor 3: 29.82 °C
[18:53:58][D][seplos_bms:062]: Environment temperature: 29.82 °C
[18:53:58][D][seplos_bms:065]: Mosfet temperature: 29.78 °C
[18:53:58][D][seplos_bms:068]: Current: -6.44 A
[18:53:58][D][seplos_bms:071]: Total battery voltage: 52.80 V
[18:53:58][D][seplos_bms:074]: Residual capacity: 133.82 Ah
[18:53:58][D][seplos_bms:078]: Battery capacity: 170.00 Ah
[18:53:58][D][seplos_bms:081]: State of charge: 78.7 %
[18:53:58][D][seplos_bms:084]: Rated capacity: 180.00 %
[18:53:58][D][seplos_bms:087]: Rated capacity: 70 %
[18:53:58][D][seplos_bms:090]: State of health: 100.0 %
[18:53:58][D][seplos_bms:093]: Port voltage: 52.79 V
```

I would be happy if you give it a try!

faizan-elite, 2022-06-19:
Thanks for the update. 

I have tested the new draft. Seems to be getting errors. I have attached the log and mqtt messages.

[logs_wemos-mini_logs.txt](https://github.com/syssi/esphome-seplos-bms/files/8935580/logs_wemos-mini_logs.txt)
[image]

syssi, 2022-06-19:
Thanks for testing! Your log containing some real responses helps a lot.

syssi, 2022-06-19:
You battery has 15 cells, right?

faizan-elite, 2022-06-19:
Yes its 15 cells.

syssi, 2022-06-19:
I've pushed a fix to handle/decode shorter telemetry responses of a 15 cell system. Please give it a try. You can ignore the MQTT topics for now because all important output is written to the log. No sensors are implemented yet.

faizan-elite, 2022-06-19:
Thanks Working Now. Attached logs.

[logs_wemos-mini_upload.txt](https://github.com/syssi/esphome-seplos-bms/files/8935736/logs_wemos-mini_upload.txt)

faizan-elite, 2022-06-19:
attached a more detailed log. seems to be getting the invalid header error in between readings. 

[logs_wemos-mini_logs.txt](https://github.com/syssi/esphome-seplos-bms/files/8935765/logs_wemos-mini_logs.txt)

but overall its working well.

syssi, 2022-06-19:
The invalid header errors are fine because some bytes (-> the start of frame byte) gets missed every now and then. This can be improved by not using software serial. Let's ignore this "issue" for now because it's not dangerous.

syssi, 2022-06-19:
As next step I will expose the values as sensors. So you can access the data via `mqtt` or `api`.

faizan-elite, 2022-06-19:
Thanks That will be great for anyone wanting to get data from their seplos bms.

It is currently how i have been able to setup on my end using nodered to update MQTT topics and send notifications for battery warnings as they occur.

[image]

[image]

[image]

my next step now is to pull mqtt topics and construct CAN messages in Pylon protocol to send to inverter. Since my batt has the RS232 instead of the CAN port it is unable to communicate with the inverter.

syssi, 2022-06-19:
I've started another project here: https://github.com/syssi/esphome-virtual-can-bms

This component helps to pick some measurements of a connected BMS and construct CAN bus message to inform a Sunny Boy f.e.

faizan-elite, 2022-06-19:
thanks. will take a look.

Best Regards
--
Faizan Owais

syssi, 2022-06-19:
I've implemented all sensors. Please add the [sensors to your configuration yaml ](https://github.com/syssi/esphome-seplos-bms/blob/main/esp8266-example.yaml) and give it another try! :-)

syssi, 2022-06-19:
Could you provide a photo of your Seplos BMS mainboard? I would like to know the exact model number and PCB version.

faizan-elite, 2022-06-19:
Thanks for the update. Will test it and provide you feedback tomorrow. 

My pack is actually assembled and sold by a local company (not branded) however being under warranty for another 3 years getting inside is not an option at this time. Took me sometime to figure out it was a seplos bms after much research figuring out how to get the battery and inverter communication to work. 

From the manufacturer info command i was able to get the following: 
Model : 1101-ZH26
Protocol Version 2.02

Same infö as above also shows up in the seplos battery monitor software.

faizan-elite, 2022-06-19:
Was able to test the update. Will have more logs when the battery is in use during the day to check the power figures. The sensors are working. This is awesome work on your part !

note: Json object in MQTT not able to see the value of any sensor as part of the object. Maybe I'm missing something?

[logs_wemos-mini_upload.txt](https://github.com/syssi/esphome-seplos-bms/files/8936415/logs_wemos-mini_upload.txt)

[image]

syssi, 2022-06-19:
Try to have a look at the full MQTT traffic of the node. A ESPHome node exposes one sensor per topic. The message contains the plain measurement (no JSON object).

faizan-elite, 2022-06-20:
Thanks.

The topic I was referring to maybe created by home assistant. The topic I have setup in yaml shows the sensor sub topic with values. All good here.

syssi, 2022-06-20:
You told me you are already requesting and parsing the CID2 `0x44` frame. Could you provide a raw request and response payload I can work with?

faizan-elite, 2022-06-20:
Is this the warnings / alarm CID Command >>> "~20004644E00200FD35\r" ?

On 20 Jun 2022, 12:39 PM +0500, syssi/esphome-seplos-bms ***@***.***>, wrote:
>
> 0x44

syssi, 2022-06-20:
Yes!

```
~20004644E00200FD35\r
       ^^ CID2
```

faizan-elite, 2022-06-20:
Attached decoding the communication response. Its a bit messy but should be able to understand. 

let me know if you need any help here / confusion on anything.

Also attached my NodeRed Flow if that can help you. It takes the mqtt responses (first version of your code basic uart) and decodes all the values for the telemetry data & warnings. 

[Seplos BMS Decode Communication Response.docx](https://github.com/syssi/esphome-seplos-bms/files/8938683/Seplos.BMS.Decode.Communication.Response.docx)

[Seplos BMS Flow NodeRed.json.zip](https://github.com/syssi/esphome-seplos-bms/files/8938702/Seplos.BMS.Flow.NodeRed.json.zip)

syssi, 2022-06-20:
I've found what I'm looking for at your first post / log:

```
>>> "~20004644E00200FD35\r"
<<< "~20004600A06000010F000000000000000000000000000000060000000000000000140000000000000300000200000000000000000002EB74\r"
```

Thanks for sharing your node red implementation. This will speed up things. :-)

faizan-elite, 2022-06-20:
Sure anyway i can help.. My nodered is setup for 15s so you might need to adjust things on your end accordingly for 16s.

syssi, 2022-06-20:
Could you do me another favor? Please flash this configuration to your device

```
substitutions:
  name: seplos-bms
  tx_pin: GPIO4
  rx_pin: GPIO5

esphome:
  name: ${name}
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
ota:
logger:

uart:
  id: uart0
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: true
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

interval:
  - interval: 20s
    then:
      # Get system parameters (CID2 `0x47`)
      - uart.write: "~200046470000FDA9\r"
      - delay: 2s

      # Get protocol version (CID2 `0x4F`)
      - uart.write: "~2000464F0000FD9A\r"
      - delay: 2s

      # Get manufacturer info (CID2 `0x51`)
      - uart.write: "~200046510000FDAE\r"
      - delay: 2s

      # Get management info (pylontech only?)
      - uart.write: "~200046920000FDA9\r"
      - delay: 2s

      # Get module serial number (pylontech only?)
      - uart.write: "~200046930000FDA8\r"
      - delay: 2s
```

and provide the log. I'm looking for response examples of the commands above and I would to know your BMS is able to respond to "custom pylontech commands".

javierhouse, 2022-12-21:
Hi Syssi, 

Im reading you project because I need read the information of my bms seplos. I would like learn something about this protocol, because is new for me. I try your last code in my ESP32 and I can see the out message in the log, but I don´t receive any response from the bms.

The only difference is the baud_rate parameter, I just try with 19200 and 9600 to, and have the same result.

`22:01:17][C][uart.arduino_esp32:108]: UART Bus 1:
[22:01:17][C][uart.arduino_esp32:109]:   TX Pin: GPIO17
[22:01:17][C][uart.arduino_esp32:110]:   RX Pin: GPIO16
[22:01:17][C][uart.arduino_esp32:112]:   RX Buffer Size: 384
[22:01:17][C][uart.arduino_esp32:114]:   Baud Rate: 19200 baud
[22:01:17][C][uart.arduino_esp32:115]:   Data Bits: 8
[22:01:17][C][uart.arduino_esp32:116]:   Parity: NONE
[22:01:17][C][uart.arduino_esp32:117]:   Stop bits: 1
[22:01:17][C][captive_portal:088]: Captive Portal:
[22:01:17][C][web_server:125]: Web Server:
[22:01:17][C][web_server:126]:   Address: battery.local:80
[22:01:17][C][mdns:103]: mDNS:
[22:01:17][C][mdns:104]:   Hostname: battery
[22:01:17][C][ota:093]: Over-The-Air Updates:
[22:01:17][C][ota:094]:   Address: battery.local:3232
[22:01:17][C][ota:097]:   Using Password.
[22:01:17][C][api:138]: API Server:
[22:01:17][C][api:139]:   Address: battery.local:6053
[22:01:17][C][api:143]:   Using noise encryption: NO
[22:01:25][D][uart_debug:158]: >>> "~200046470000FDA9\r"
[22:01:27][D][uart_debug:158]: >>> "~2000464F0000FD9A\r"
[22:01:29][D][uart_debug:158]: >>> "~200046510000FDAE\r"
[22:01:31][D][uart_debug:158]: >>> "~200046920000FDA9\r"`
 
but really I don´t know were is the problem.

thanks in advance.

syssi, 2022-12-21:
Could you provide a photo of your wiring? I would like to verify the RX/TX lines are connected to GPIO16/17. Which RS485 converter module do you use? May be flow control is required? Does your RS485 module have RX/TX leds? Does the TX led blink periodically? If the RX led links instead you should switch the RX/TX lines. What about the dip switches of your Seplos BMS? Did you turn off all switches?

javierhouse, 2022-12-21:
Syssl, first of all, thank you a lot.

1. Wiring (tx-->brawn rx-->grey)
[image]
[image]

2. Which RS485 converter module do you use? 
I`m using this model: [https://www.amazon.es/ARCELI-Adaptador-Puerto-M%C3%B3dulo-convertidor/dp/B07DN115BZ](url)

3. Does your RS485 module have RX/TX leds? 
Yes, and if I`m not wrong, both are blinking every 2/5 seconds

4. If the RX led links instead you should switch the RX/TX lines.
I`m sorry, I don`t understand what do you mean with this.

5. What about the dip switches of your Seplos BMS? Did you turn off all switches? 
Yes all are "off" actually I only have one battery.

syssi, 2022-12-22:
3. Please unplug the BMS. The TX LED should blink periodically. The RX LED must stay off.
4. I tried to say you have to swap the TX/RX lines if the TX pin of the ESP is connected to the RX pin of the RS485 module . It looks like you have connected the RS485 module properly.

javierhouse, 2022-12-22:
3. Yes, If I unplug the A & B cable from BMS to the RS845 only the TX Led blink.

closed 2023-01-24 by syssi