syssi/esphome-seplos-bms #6: Request and decode telesignalization frame

syssi, 2022-06-19:


syssi, 2022-08-16:
https://github.com/celsworth/lxp-pylon-utils/blob/master/monitor.rb#L109-L188

chmutoff, 2023-08-21:
Hi! I managed to decode the alarms, but I can't send two messages to modbus.  I've tried to send 2 commands in the update method in seplos_bms.cpp but this seems to break down the first command as i get no data in the response. Could you advice how to process two commands? I guess the code need to be changed to send one command, wait for response, process it and then send a second command and process it.

syssi, 2023-08-22:
Could you share some more details? In best case some could. So it's easier for me to provide a good solution.

chmutoff, 2023-08-22:
Yes, what i'm trying to do is to send 0x42 packet and then 0x44. In the file seplos_bms.cpp i've tried the following:

`
void SeplosBms::update() { this->send(0x42, this->pack_); this->send(0x44, this->pack_); }
`

After that, i'm getting the following response from the bms:
`
[10:43:04][W][seplos_bms:021]: Unhandled data received (data_len: 0x00): 20.00.46.03.00.00 (6)
`
This looks like the BMS received one single malformed packet.

I think that the update method should be something like:
```
void SeplosBms::update() {
   this->send(0x42, this->pack_); 
   this>read(); // Get the response and process it
   this->send(0x44, this->pack_); 
   this->read(); // Get the response and process it
}
```

syssi, 2023-08-22:
Please add a `delay` between the two calls just for poking around:

```
#include "esphome/core/hal.h"

this->send(0x42, this->pack_); 
delay(150);  // NOLINT
this->send(0x44, this->pack_);
```

All responses will be reported as the mentioned warning because there is no decoder available at the moment for this kind of response. 

```
[10:43:04][W][seplos_bms:021]: Unhandled data received (data_len: 0x00): 20.00.46.03.00.00 (6)
```

chmutoff, 2023-08-22:
The delay option is working! Right now the differentiation between telemetry and non telemetry frames is done by this condition
```
data.size() >= 44 && data[8] >= 8 && data[8] <= 16
```
Which is basically looking at size of the response and the number of cells, but the telecommand frames also have this data, so the `on_telemetry_data_` method is triggered by both frames. There is no identification of the response frame. All the frames have the same header, except the lengh field, but that is not very precise to differentiate between frame types. For now, I think that reading and processing the response after sending it is the best approach.

syssi, 2023-08-22:
> For now, I think that reading and processing the response after sending it is the best approach.

Yes. The implementation must be improved on the long run here. We have to remember the last transmitted request and expect the dedicated response to avoid guessing the frame type by length.

tlamoureux24, 2024-06-09:
Hello, i'm not a coder but found something maybe could help (please delete my post if useless)
https://github.com/Louisvdw/dbus-serialbattery/blob/master/etc/dbus-serialbattery/bms/seplosv3.py

Have a good day ! :)

syssi, 2024-06-09:
https://github.com/Louisvdw/dbus-serialbattery/blob/master/etc/dbus-serialbattery/bms/seplosv3.py#L221-L266

tlamoureux24, 2024-06-10:
Hello, i supposed you point me to the part of code which is interesting but sorry i'm not a coder and i don't know what to do with this .....
I simply share here what i found while searching another way to get all datas from the bms.

syssi, 2024-06-10:
That's fine. The link to the explicit method is just a pointer for myself.