syssi/esphome-seplos-bms #8: Implement additional commands

syssi, 2022-06-19:
| CID2 | Description |
| ---- | ----------- |
| 0x45 | Telecontrol command   |
| 0x47 | Acquisition of teleregulation information   |
| 0x49 | Setting of teleregulation information   |
| 0x4F | Acquisition of the communication protocol version number   |
| 0x51 | Acquisition of device vendor information   |
| 0x4B | Acquisition of historical data   |
| 0x4D | Acquisition time   |
| 0x4E | Synchronization time   |
| 0xA0 | Production calibration   |
| 0xA1 | Production setting   |
| 0xA2 | Regular recording   |

syssi, 2022-08-14:
```
# "History record" tab
# "Get first" button
~2000464BC0040000FCC7

# "SN" -> "read" button
~200046A20000FDA1

# Disable discharging mosfet
~20004645A00600001FFC5D

# Disable charging mosfet
~20004645A00600011FFC5C

# Enable current limit control
~20004645A006000210FC71

# Enable temperature control
~20004645A006000310FC70

# Shutdown system
~20004645A006000400FC70

# Restore factory defaults
~20004645A006000500FC6F

```

syssi, 2022-08-16:
```
~2000 46 45 A0 06 00 00 1F FC5D # Disable discharging mosfet
~2000 46 45 A0 06 00 01 1F FC5C # Disable charging mosfet
~2000 46 45 A0 06 00 02 10 FC71 # Enable current limit control
~2000 46 45 A0 06 00 03 10 FC70 # Enable temperature control
~2000 46 45 A0 06 00 04 00 FC70 # Shutdown system
~2000 46 45 A0 06 00 05 00 FC6F # Restore factory defaults
```

midadami, 2023-04-30:
Hi, any way I can help in order to achieve changing parameters through this implementation?
Im looking to change parameters such as monomer voltage protection, max current and so on. 
I have a SEPLOS 100A CAN connected to inverter and with RS485 connected to HA through esp8266. Can change cables temporarely to connect to Battery Monitor if needed.
Thanks.

syssi, 2023-04-30:
Could you use the battery monitor / windows application to capture some commands? There is a debug view to monitor the raw traffic. Could you provide such a capture?

midadami, 2023-04-30:
Changing parameter 50, Charging overcurrent warning to 25A from 35A. If you need something else just ask for it and I will see if I can deliver. 
Thanks
[changing parameter.xls](https://github.com/syssi/esphome-seplos-bms/files/11361826/changing.parameter.xls)

midadami, 2023-04-30:
Same as before but Changing parameter 50, Charging overcurrent warning to 35A from 25A.
[changing parameter to 35.xls](https://github.com/syssi/esphome-seplos-bms/files/11361840/changing.parameter.to.35.xls)

midadami, 2023-05-01:
Another one a bit different, desactivating function.
[Desactivate_Activate monomer high voltage alarm.xls](https://github.com/syssi/esphome-seplos-bms/files/11366970/Desactivate_Activate.monomer.high.voltage.alarm.xls)

syssi, 2023-05-01:
Cool! I will review the captures as soon as possible and will provide a first draft of the implementation.

syssi, 2023-05-02:
```
# changing parameter.xls
>>> telemetrydata ~20004642E00200FD37$
<<< ~2000460010960000100D3A0D3C0D3C0D3B0D3B0D3C0D3C0D3A0D3B0D390D3D0D390D3D0D3D0D3A0D3C060B780B950B980B960BC90B9E0000152C76C00A76C003E876C0000603E814F00001FE6B036402D8DCB8$
>>> remotecommunicationdata ~20004644E00200FD35$
<<< ~20004600806200001002020202020202020202020202020202060000000000000002140021000000020100001000000000000000000010EB0E$
>>> telemetrydata ~20004642E00200FD37$
<<< ~2000460010960000100D3A0D3C0D3C0D3B0D3B0D3C0D3C0D3A0D3B0D390D3D0D390D3E0D3D0D3A0D3C060B780B950B980B960BC90B9E0000152C76C00A76C003E876C0000603E814F0FFFDFE6C036402D8DC61$
>>> remotecommunicationdata ~20004644E00200FD35$
<<< ~20004600806200001002020202020202020202020202020202060000000000000002140021000000020100001000000000000000000010EB0E$
>>> telemetrydata ~20004642E00200FD37$
<<< ~2000460010960000100D3A0D3C0D3C0D3B0D3B0D3C0D3C0D3A0D3B0D390D3D0D3A0D3E0D3D0D3A0D3C060B780B950B980B960BC90B9E0000152C76C00A76C003E876C0000603E814F00002FE6C036402D8DCAD$
>>> remotecommunicationdata ~20004644E00200FD35$
<<< ~20004600806200001002020202020202020202020202020202060000000000000002140021000000020100001000000000000000000010EB0E$
>>> telemetrydata ~20004642E00200FD37$
<<< ~2000460010960000100D3A0D3C0D3C0D3B0D3B0D3C0D3C0D3A0D3B0D390D3D0D3A0D3E0D3D0D3A0D3C060B780B950B980B960BC90B9E0000152C76C00A76C003E876C0000603E814F00003FE6C036402D8DCAC$
>>> remotecommunicationdata ~20004644E00200FD35$
<<< ~20004600806200001002020202020202020202020202020202060000000000000002140021000000020100001000000000000000000010EB0E$
>>> telemetrydata ~20004642E00200FD37$
<<< ~2000460010960000100D3A0D3C0D3C0D3B0D3B0D3C0D3C0D3A0D3B0D390D3D0D3A0D3E0D3D0D3A0D3C060B780B950B980B960BC90B9E0000152C76C00A76C003E876C0000603E814F00007FE6C036402D8DCA8$
>>> remotecommunicationdata ~20004644E00200FD35$
<<< ~20004600806200001002020202020202020202020202020202060000000000000002140021000000020100001000000000000000000010EB0E$
>>> telemetrydata ~20004642E00200FD37$
<<< ~2000460010960000100D3A0D3C0D3B0D3B0D3B0D3C0D3C0D3A0D3B0D390D3D0D3A0D3E0D3D0D3A0D3C060B780B950B970B960BC90B9E0000152C76C00A76C003E876C0000603E814F0000BFE6B036402D8DCA0$
>>> remotecommunicationdata ~20004644E00200FD35$
<<< ~20004600806200001002020202020202020202020202020202060000000000000002140021000000020100001000000000000000000010EB0E$
>>> telemetrydata ~20004642E00200FD37$
<<< ~2000460010960000100D3A0D3C0D3B0D3A0D3B0D3C0D3C0D3A0D3B0D390D3D0D3A0D3E0D3D0D3A0D3B060B780B950B980B960BC90B9E0000152C76C00A76C003E876C0000603E814F0000AFE6D036402D8DCA0$
>>> remotecommunicationdata ~20004644E00200FD35$
<<< ~20004600806200001002020202020202020202020202020202060000000000000002140021000000020100001000000000000000000010EB0E$
>>> telemetrydata ~20004642E00200FD37$
<<< ~2000460010960000100D3A0D3C0D3B0D3A0D3B0D3C0D3C0D3A0D3B0D390D3D0D390D3D0D3D0D390D3B060B780B950B980B960BC90B9E0000152B76C00A76C003E876C0000603E814F0000FFE6B036402D8DCAF$
>>> remotecommunicationdata ~20004644E00200FD35$
<<< ~20004600806200001002020202020202020202020202020202060000000000000002140021000000020100001000000000000000000010EB0E$
>>> telemetrydata ~20004642E00200FD37$
<<< ~2000460010960000100D3A0D3C0D3B0D3A0D3B0D3C0D3C0D3A0D3B0D390D3D0D390D3D0D3D0D390D3B060B780B950B980B960BC90B9E0000152B76C00A76C003E876C0000603E814F0000CFE6C036402D8DCB1$
>>> remotecommunicationdata ~20004644E00200FD35$
<<< ~20004600806200001002020202020202020202020202020202060000000000000002140021000000020100001000000000000000000010EB0E$
>>> telemetrydata ~20004642E00200FD37$
<<< ~2000460010960000100D3A0D3C0D3B0D3A0D3B0D3C0D3C0D3A0D3B0D390D3D0D390D3D0D3D0D390D3B060B780B950B970B960BC90B9E0000152B76C00A76C003E876C0000603E814F0000CFE6B036402D8DCB3$
>>> remotecommunicationdata ~20004644E00200FD35$
<<< ~20004600806200001002020202020202020202020202020202060000000000000002140021000000020100001000000000000000000010EB0E$
>>> telemetrydata ~20004642E00200FD37$
<<< ~2000460010960000100D3A0D3C0D3B0D3A0D3B0D3C0D3C0D3A0D3B0D390D3D0D390D3D0D3D0D390D3B060B780B950B980B960BC90B9E0000152B76C00A76C003E876C0000603E814F00009FE6C036402D8DCBB$
>>> remotecommunicationdata ~20004644E00200FD35$
<<< ~20004600806200001002020202020202020202020202020202060000000000000002140021000000020100001000000000000000000010EB0E$
SetPack#0allparameters ~200046A18152003C0D520CF80C1C0C5D0D8E0CF80BB80C5D0D4805DC155014C0136013C8159014C012C013C8189C17D40C130BEB0AE70B050C3B0BEB0AAB0AC90C130BEB0AB50AC90C3B0C090AA10AB50AAB0B0F0BF50BB90ADD0AF10C090BF50AAB0ABF0CD10C9F0D030C9F09C40898DCD8E4A82710D8F09E5807D076C076C01B0F0A64630A100A0A1E3C0505010A0A1EF030050160506409000D0008FFFFFF3FBF9FAC9E313130312D5A41313020B268$
<<< ~200046000000FDB4$
GetManufacturerInformation ~200046510000FDAE$
<<< ~20004600C040313130312D5A41313020100443414E3A504E475F4459455F4C7578705F544242F02F$
>>> telemetrydata ~20004642E00200FD37$
<<< ~2000460010960000100D3A0D3C0D3B0D3A0D3B0D3C0D3C0D3A0D3B0D390D3D0D390D3D0D3D0D390D3B060B780B950B980B960BC90B9E0000152B76C00A76C003E876C0000603E814F00007FE6D036402D8DCBC$
>>> remotecommunicationdata ~20004644E00200FD35$
<<< ~20004600806200001002020202020202020202020202020202060000000000000002140021000000020100001000000000000000000010EB0E$
>>> allparameters ~20004647E00200FD32$
<<< ~200046008152003C0D520CF80C1C0C5D0D8E0CF80BB80C5D0D4805DC155014C0136013C8159014C012C013C8189C17D40C130BEB0AE70B050C3B0BEB0AAB0AC90C130BEB0AB50AC90C3B0C090AA10AB50AAB0B0F0BF50BB90ADD0AF10C090BF50AAB0ABF0CD10C9F0D030C9F0DAC0898DCD8E4A82710D8F09E5807D076C076C01B0F0A64630A100A0A1E3C0505010A0A1EF0300501605064090 [... cut ...]
<<< ~20004600806200001000000000000000000000000000000000060000000000000000140000000000020300000100000000000000000001EB31$
>>> allparameters ~20004647E00200FD32$
<<< ~200046008152003C0D520CF80C1C0C5D0D8E0CF80BB80C5D0D4805DC155014C0136013C8159014C012C013C8189C17D40C130BEB0AE70B050C3B0BEB0AAB0AC90C130BEB0AB50AC90C3B0C090AA10AB50AAB0B0F0BF50BB90ADD0AF10C090BF50AAB0ABF0CD10C9F0D030C9F09C40898DCD8E4A82710D8F09E5807D076C076C01B0F0A64630A100A0A1E3C0505010A0A1EF030050160506409000D0008FFFFFF3FBF9FAC9E313130312D5A41313020B27A$
```

midadami, 2023-05-20:
Any update on this funcionality. If you need more info to help you ask without a problem.
Im guessing you are trying to identify every parameter on the sequence which might be a nightmare. 
From my point of view it will suffice if we could enter the whole sequence of characters by hand into a named variable and then use it through a button on esphome (HA). Obviously this approach would require the user to obtain the sequence beforehand from his seplos but it will avoid having to program all the logic behind value ranges on parameters and errors due to wrong value on parameter.

ouinouin, 2025-01-21:
Hi, interrested by this topic, i can do some testing if necessary , i m specially interrested by the enable disable charge/discharge mosfet functionality if existing somewhere on those seplos BMS .

ouinouin, 2025-01-21:
for reference : on the seplos V3 , the commands to turn on/off the charge /discharge mosfet are : 

funnily the payload is either 55AA (0101010101010101) or AA55 (1010101010101010) to say zero or one.... 
```
2025/01/21 13:57:11:511:Execute Pack#0【Discharing FETs Off】CloseCommand(Address:0x1510)
00 10 15 10 00 01 02 55 AA 52 BE 
BMS response OK
00 10 15 10 00 01 05 D1 

2025/01/21 13:58:49:731:Execute Pack#0【Discharing FETs On】OpenCommand(Address:0x1510)
00 10 15 10 00 01 02 AA 55 53 0E 
BMS response OK
00 10 15 10 00 01 05 D1 

2025/01/21 13:57:11:511:Execute Pack#0【Discharing FETs Off】CloseCommand(Address:0x1510)
00 10 15 10 00 01 02 55 AA 52 BE 
BMS response OK
00 10 15 10 00 01 05 D1 

2025/01/21 14:02:22:058:Execute Pack#0【Charing FETs Off】CloseCommand(Address:0x1511)
00 10 15 11 00 01 02 55 AA 53 6F 
BMS response OK
00 10 15 11 00 01 54 11 

2025/01/21 14:05:07:575:Execute Pack#0【Charing FETs On】OpenCommand(Address:0x1515)
00 10 15 15 00 01 02 AA 55 53 5B 
BMS response OK
00 10 15 15 00 01 15 D0 
```