syssi/esphome-seplos-bms #20: Add mosfet control

cristi2005, 2022-08-08:
Hi,

Can you add mosfet controll button in this implementations ?

Thank you,
Cristi

syssi, 2022-08-08:
If you provide some testing I'm happy to be able to add the feature.

cristi2005, 2022-08-08:
Sure, you tell me what tests you need me to do and I do them. Thanks

syssi, 2022-08-10:
Are you able to talk to your Seplos device using the Battery Monitor software of the manufacturer? Could you provide screenshots of the different tabs? Does the software provide switches to control the Mosfets? I'm trying to find out some control frames. The protocol description doesn't help because the important parts how to send commands to the BMS are missing.

cristi2005, 2022-08-11:
When I try to stop the input or output of the mosfets, it gives me the error in the attached picture. I think the commands are blocked from the software :(

[image]

syssi, 2022-08-11:
This control switch is at the calibrate tab? I could try to implement the command but it's useless for your battery because it looks like your BMS rejects the request.

cristi2005, 2022-08-11:
Yes it is in calibrate tab. For the moment I have RS485 variant with comunication with the inverter but I ordered CAN bus variant and with will come in next week and I play with him and I see if can swich on or of the mosfets.

syssi, 2022-08-14:
I've extracted the payload of the different buttons:

```
# Disable discharging mosfet
~20004645A00600001FFC5D

# Disable charging mosfet
~20004645A00600011FFC5C

# Enable current limit control
~20004645A006000210FC71

# Enable temperature control
~20004645A006000310FC70

# Shutdown system
~20004645A006000400FC70

# Restore factory defaults
~20004645A006000500FC6F
```

See also https://github.com/syssi/esphome-seplos-bms/issues/8

syssi, 2023-01-24:
Let's close this issue and use #8 for further discussions!

closed 2023-01-24 by syssi