syssi/esphome-seplos-bms #21: Installation tutorial

atze09, 2022-08-09:
Î already flashed the ESP via ESPHome and watched many youtube tutorial videos.
But how can i upload the Setting under "installation" to my ESP?
i am struggling here, can anyone help me or maybe link me to a suitable tutorial?
Via Python program and docker, or is there an easier way?
Sorry for all the noob questions.
Thank you.

syssi, 2022-08-09:
Do you have access to a linux machine?

atze09, 2022-08-09:
> Do you have access to a linux machine?

Windows and Linux both are available.

syssi, 2022-08-09:
Please watch this short terminal session recording:

[image]

In words:

1. Make sure python3 is installed
2. Make sure the python package manager is available
3. Use the package manager (pip3) to install ESPHome
4. Use the installed CLI tool to build the example configuration and additional file called `secrets.yaml` is required.
5. Create the `secrets.yaml` containing your WiFi network details
6. Execute `esphome run example.yaml` to build and flash the setup

syssi, 2022-08-09:
Please try to replicate these steps and ping me if something doesn't work as expected / recorded.

atze09, 2022-08-09:
> Please watch this short terminal session recording:
>
> [image] [ [image] ](https://user-images.githubusercontent.com/2735933/183646786-b5471183-9ccf-491f-9b32-4649abcd624c.gif) [ [...]

Awesome thank you very much for your great support.

Now two questions are left.
Do i need additional settings for the "esp8266-example-multiple-battery-banks.yaml"?
Because i have 2 seplos battery banks?

And is this RS485 to TTL also supported?
[https://www.amazon.de/DollaTek-MAX485-RS485-RS-485-Entwicklungsboard/dp/B07DK4QG6H/ref=sr_1_4?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=1G2C9SV8GI721&keywords=rs485+ttl&qid=1660051166&sprefix=rs485+ttl%2Caps%2C95&sr=8-4](url)

syssi, 2022-08-09:
You can use the `esp8266-example-multiple-battery-banks.yaml` if your battery banks without changes if your battery banks addresses are setup properly:

https://github.com/syssi/esphome-seplos-bms/blob/main/esp8266-example-multiple-battery-banks.yaml#L40
https://github.com/syssi/esphome-seplos-bms/blob/main/esp8266-example-multiple-battery-banks.yaml#L43
https://github.com/syssi/esphome-seplos-bms/blob/main/esp8266-example-multiple-battery-banks.yaml#L46

If there are other addresses assigned please change there values before executing `esphome run config.yaml`.

I cannot recommend the mentioned RS485 modules. They require flow control. I prefer the modules without flow control. They look like this: https://raw.githubusercontent.com/syssi/esphome-jnge-mppt-controller/main/images/circuit_thumbnail.jpg

atze09, 2022-08-11:
Awesome i ordered the RS485 modules.
One last question:
How is the communication with Home assistant in your JKBMS Reposirory it is done via mqtt and how is this handled here in the Seplos Rep?

> You can use the `esp8266-example-multiple-battery-banks.yaml` with your battery banks without changes if your battery banks addresses are setup properly:
> 
> https://github.com/syssi/esphome-seplos-bms/blob/main/esp8266-example-multiple-battery-banks.yaml#L40 https://github.com/syssi/esphome-seplos-bms/blob/main/esp8266-example-multiple-battery-banks.yaml#L43 https://github.com/syssi/esphome-seplos-bms/blob/main/esp8266-example-multiple-battery-banks.yaml#L46
> 
> If there are other addresses assigned please change there values before executing `esphome run config.yaml`.
> 
> I cannot recommend the mentioned RS485 modules. They require flow control. I prefer the modules without flow control. They look like this: https://raw.githubusercontent.com/syssi/esphome-jnge-mppt-controller/main/images/circuit_thumbnail.jpg

atze09, 2022-08-11:
Sensors are created automatically but how do they interact and why did you add mqtt at JK Repo?

closed 2022-08-11 by atze09

syssi, 2022-08-11:
If you use Home Assistant please use the `api` component to connect your node. If you use another Home Automation Software there is no native integration and you should use a generic approach like `mqtt`.