syssi/esphome-seplos-bms #23: new code does not seem to work

NosIreland, 2022-08-30:
I have not checked this for a while as my previous code was working as per https://github.com/syssi/esphome-seplos-bms/issues/1.
Now I tried  a new one https://github.com/syssi/esphome-seplos-bms/blob/main/esp8266-example.yaml but it does not work. My hardware has not changed so it is fine. Here are the things that do not look right to me:
- the GPIOs are diferent should they not be   tx_pin: GPIO1  rx_pin: GPIO3?
- baudrate dropped from 19200 to 9600?

If I use the settings provided I get:
```
[20:45:11][D][uart_debug:158]: >>> "~20004642E00200FD37\r"

[20:45:13][D][uart_debug:158]: >>> "~20004642E00200FD37\r"

[20:45:15][D][uart_debug:158]: >>> "~20004642E00200FD37\r"

```

If I set GPIOs and baud rate the way it was I get
```
[20:41:15][D][uart_debug:158]: <<< "go7\xBF/\xF7}\xBD\xB5u7\xB5\xB5\xFD\xFDuuo\xDF\xDF\xBA\xFC"
[20:41:15][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[20:41:15][W][seplos_modbus:086]: Invalid header.
[20:41:15][W][seplos_modbus:086]: Invalid header.
[20:41:15][W][seplos_modbus:086]: Invalid header.
[20:41:15][W][seplos_modbus:086]: Invalid header.
[20:41:15][W][seplos_modbus:086]: Invalid header.
```

Any ideas?

syssi, 2022-08-30:
Hmmm. The OP at #2 is able to talk to the BMS using 9600 bauds.

syssi, 2022-08-31:
Did you disable the logger output on GPIO1/GPIO3 by `baudrate: 0`?

syssi, 2022-08-31:
Let's go a few steps back in time and please try this test configuration again:

```
substitutions:
  name: seplos-bms
  tx_pin: GPIO1
  rx_pin: GPIO3

esphome:
  name: ${name}
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
ota:
logger:
  baud_rate: 0

uart:
  id: uart0
  baud_rate: 19200
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: true
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

interval:
  - interval: 20s
    then:
      # Get pack #0 telemetry data (CID2 `0x42`)
      - uart.write: "~20004642E00200FD37\r"
      - delay: 2s
```

Do you see any responses?

NosIreland, 2022-08-31:
Yes, I do see responce 
```
[10:23:21][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[10:23:21][D][uart_debug:158]: <<< "~2000460010960001100D270D2C0D2A0D2F0D270D2A0D300D2C0D320D310D320D2B0D260D360D270D26060BC40BBD0BBA0BBB0BD20BCE09FA151330CD0A426802DE4650006103E81520000"
[10:23:21][D][uart_debug:158]: <<< "0000000000000DD5B\r"
[10:23:41][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[10:23:41][D][uart_debug:158]: <<< "~2000460010960001100D280D2D0D2C0D310D280D2B0D320D2E0D340D330D340D2D0D270D370D280D28060BC40BBD0BBA0BBB0BD20BCE09F7151530DB0A426802DF4650006103E81522000"
[10:23:41][D][uart_debug:158]: <<< "0000000000000DD5E\r"
```

This code is not much diferent from what I've been suign so far:
```
substitutions:
  name: seplos-bms-1
  tx_pin: GPIO1
  rx_pin: GPIO3

esphome:
  name: ${name}
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server:
  port: 80

# api:
ota:

logger:
  baud_rate: 0

uart:
  id: uart0
  baud_rate: 19200
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: RX
    after:
      bytes: 200
    dummy_receiver: true
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

interval:
  - interval: 10s
    then:
      - uart.write: "~20004642E00200FD37\r"
```
It has worked fine for me. I'm just wondering what is different in new code.

syssi, 2022-08-31:
I will provide some more instructions later the day to move to the new version. I'm pretty sure we are able to identify to root cause / difference between the solutions.

syssi, 2022-09-03:
Could you use this config example as next step:

```
substitutions:
  name: seplos-bms
  external_components_source: github://syssi/esphome-seplos-bms@increase-rx-timeout
  tx_pin: GPIO1
  rx_pin: GPIO3

esphome:
  name: ${name}
  platform: ESP8266
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
ota:
logger:
  baud_rate: 0

uart:
  id: uart0
  baud_rate: 19200
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  rx_buffer_size: 384
  debug:
    direction: BOTH
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

seplos_modbus:
  id: modbus0
  uart_id: uart0

seplos_bms:
  id: bms0
  seplos_modbus_id: modbus0

sensor:
  - platform: seplos_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    temperature_1:
      name: "${name} temperature 1"
    temperature_2:
      name: "${name} temperature 2"
    temperature_3:
      name: "${name} temperature 3"
    temperature_4:
      name: "${name} temperature 4"
    temperature_5:
      name: "${name} environment temperature"
    temperature_6:
      name: "${name} mosfet temperature"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    residual_capacity:
      name: "${name} residual capacity"
    battery_capacity:
      name: "${name} battery capacity"
    rated_capacity:
      name: "${name} rated capacity"
    state_of_charge:
      name: "${name} state of charge"
    charging_cycles:
      name: "${name} charging cycles"
    state_of_health:
      name: "${name} state of health"
    port_voltage:
      name: "${name} port voltage"
```

This example contains three changes:

1. The increased RX timeout by using the feature branch `github://syssi/esphome-seplos-bms@increase-rx-timeout`
2. Baudrate 19200
3. GPIO1/GPIO3
4. Logger `baud_rate: 0` to avoid collisions on GPIO1/GPIO3

NosIreland, 2022-09-06:
Thanks for update and sorry for delay.
This does look better. Is there a place to set delay between checks as it is doing it constantly now?
Also I do see memory allocation errors poping up:
```
[09:31:04][D][sensor:124]: 'seplos-bms temperature 4': Sending state 29.62000 °C with 0 decimals of accuracy
[09:31:04][D][sensor:124]: 'seplos-bms environment temperature': Sending state 30.05000 °C with 0 decimals of accuracy
[09:31:04][D][sensor:124]: 'seplos-bms mosfet temperature': Sending state 29.80000 °C with 0 decimals of accuracy
[09:31:04][D][sensor:124]: 'seplos-bms current': Sending state 2.64000 A with 2 decimals of accuracy
[09:31:04][D][sensor:124]: 'seplos-bms total voltage': Sending state 52.92000 V with 2 decimals of accuracy
[09:31:04][E][json:034]: Could not allocate memory for JSON document! Requested 504 bytes, largest free heap block: 504 bytes
[09:31:04][D][sensor:124]: 'seplos-bms power': Sending state 139.70879 W with 2 decimals of accuracy
[09:31:04][D][sensor:124]: 'seplos-bms charging power': Sending state 139.70879 W with 2 decimals of accuracy
[09:31:04][D][sensor:124]: 'seplos-bms discharging power': Sending state 0.00000 W with 2 decimals of accuracy
[09:31:04][E][json:034]: Could not allocate memory for JSON document! Requested 504 bytes, largest free heap block: 504 bytes
[09:31:04][D][sensor:124]: 'seplos-bms residual capacity': Sending state 81.52000 Ah with 2 decimals of accuracy
[09:31:04][D][sensor:124]: 'seplos-bms battery capacity': Sending state 170.00000 Ah with 2 decimals of accuracy
```

syssi, 2022-09-06:
You can change the polling interval by:

```
seplos_bms:
  id: bms0
  seplos_modbus_id: modbus0
  update_interval: 60s
```

The JSON encoder/decoder memory issue is strange. This custom component doesn't use JSON.

NosIreland, 2022-09-07:
After some further testing I can see that the temperature is not decoded properly as it alwasy showing 30 degrees on all 6 sensors and that is not the case.

NosIreland, 2022-09-07:
The memory allocation errors are gone since update interval was added. I have 10s atht emoment and I has been handling it well.

closed 2022-09-07 by syssi

syssi, 2022-09-07:
I've merged some better default into the `main` branch now. The example configuration should work for you again if you update the `tx_pin, rx_pin` and the `baud_rate` to align with your setup.

syssi, 2022-09-07:
Thanks for your feedback! Feel free to create additional issues if you miss something!

syssi, 2022-09-12:
Regarding

```
[09:31:04][E][json:034]: Could not allocate memory for JSON document! Requested 504 bytes, largest free heap block: 504 bytes
```

Could you make sure to use the most recent version of ESPHome? It looks like you did hit this issue:

https://github.com/esphome/issues/issues/3187

PR #24 (2022-09-07): Introduce configurable rx_timeout
Fixes: #23