syssi/esphome-seplos-bms #25: missing  baud_rate: 0 in example

NosIreland, 2022-09-08:
Add `baud_rate: 0` to example config under logger as otherwise it does not work

syssi, 2022-09-08:
`baud_rate: 0` isn't required if GPIO4/GPIO5 is used. I don't wont to disable the serial output of the logger per default to simplify the debugging if no WLAN connection is available/possible.

NosIreland, 2022-09-08:
> `baud_rate: 0` isn't required if GPIO4/GPIO5 is used. I don't wont to disable the serial output of the logger per default to simplify the debugging if no WLAN connection is available/possible.

It does not work with GPIO1/3, I get:
`[12:33:57][W][uart.arduino_esp8266:127]:   You're using the same serial port for logging and the UART component. Please disable logging over the serial port by setting logger->baud_rate to 0.`
and also
```
[12:34:13][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:13][W][seplos_modbus:086]: Invalid header.
[12:34:14][W][seplos_modbus:086]: Invalid header.
```

syssi, 2022-09-08:
I try to rephrase my statement / explain the background. It looks like you didn't understand my point:

At the `esp8266-example.yaml` I'm trying to encourage people in not using GPIO1/GPIO3 because the `logger` component uses these pins per default. If you use a Wemos D1 mini the GPIO1/3 is connected to an Serial-to-USB converter. If you open a terminal or if you use `eshome logs esp8266-example.yaml` you can see the serial log output of the ESPHome node. This is super helpful if your node is unable to join the local WiFi network or crashs permanently. If we disable the serial output of the `logger` per default by `baud_rate: 0` it's hard to debug WiFi issues or boot loops.

In your case it's a **must** to disable the serial output of the logger because the log output and the BMS communication cannot coexist disturb each other.

closed 2022-09-08 by syssi