syssi/esphome-seplos-bms #26: Temperature readings are wrong

NosIreland, 2022-09-08:
The temperature for 6 sensors are note decoded correctly and is alwasy showing 30 degrees.
If I take output from UARTdebug
`2000460010960001100CFF0CFE0D030D0C0CFD0D000D0A0D010CFD0D090D000CFD0CFD0D0A0CFD0D02060BCB0BC40BBF0BC00BEB0BDAFF8214CF31C50A426802ED4650006603E814D3000`
The temperature sensor data is `060BCB0BC40BBF0BC00BEB0BDA`
Decodinf it gives me:
06 = 6 Sensors
0BCB 
(3019-2731)/10 = 28.8
0BC4 
(3012-2731)/10 = 28.1
0BBF 
(3007-2737)/10 = 27
0BC0 
(3008-2737)/10 = 27.1
0BEB 
(3051-2737)/10 = 31.4
0BDA
(3034-2737)/10 = 29.7

The values that I get are correct and this is what BMS is showing me. This means the decoding is not correct in the code.

syssi, 2022-09-08:
Did you find out the `2731` be try and error? Can we be sure the `2731` applies to most of the Seplos BMS models?

NosIreland, 2022-09-08:
> Did you find out the `2731` be try and error? Can we be sure the `2731` applies to most of the Seplos BMS models?

It is in Seplos manual https://raw.githubusercontent.com/syssi/esphome-seplos-bms/main/docs/SEPLOS%20BMS%20Communication%20Protocol_V2.0.pdf page 9
[image]

syssi, 2022-09-08:
I've prepared a new feature branch. Could you give it a try?

```
substitutions:
  name: seplos-bms
  external_components_source: github://syssi/esphome-seplos-bms@fix-temperature-sensors
```

NosIreland, 2022-09-08:
It does work but I will need to leave it for 24 hours to see how it works and if it produces expected pattern.

syssi, 2022-09-08:
If it doesn't crash your setup (because of a under-temperature protection -> shutdown on high load): I would be happy if you could cool down a tempature probe to minus degrees to get an idea minus values are implemented correctly.

syssi, 2022-09-08:
Thanks for your support and feedback to improve this custom component. Please create additional issues if you miss something or the component doesn't work as expected.

PR #27 (2022-09-08): Fix temperature offsets
Fixes: #26
closed 2022-09-08 by syssi