syssi/esphome-seplos-bms #28: Support for 1101-SP16

atze09, 2022-10-04:
Hello i have two 1101-S16 units and would also like to use your ESPHOME Splos BMS.
Could you also implement?
i would help with testing.
Thanks.
Regards,
Dominik

syssi, 2022-10-04:
I assume it's already supported but untested. Please give it a try an ping me if it doesn't work out of the box.

atze09, 2022-10-04:
> I assume it's already supported but untested. Please give it a try an ping me if it doesn't work out of the box.

I tired but no entity shows up in HA.
Do i need to put any specific RS485 protocol setting in the BMS?

syssi, 2022-10-04:
How did you connect the ESP to the BMS? Does both BMS share the same RS485 bus? It's important to select proper addresses using the dip switches. The same addresses must be used at the configuration YAML:

https://github.com/syssi/esphome-seplos-bms/blob/main/esp8266-example-multiple-battery-banks.yaml#L39-L46

atze09, 2022-10-04:
> How did you connect the ESP to the BMS? Does both BMS share the same RS485 bus? It's important to select proper addresses using the dip switches. The same addresses must be used at the configuration [...]

Currently i am trying with one single BMS and all dip switches are off.
I will connect the second BMS in a few weeks.
But for the first trial i will try with one single BMS connected this way:
[image]

to the RS485 TTL to TTOl module

atze09, 2022-10-04:
Where can i display the response of all sensors enabled?

syssi, 2022-10-04:
This is the ESPHome log. You can connect to your ESPHome node and view the logs by:

```
esphome log esp8266-example.yaml
```

Please enable the debug output of the `uart` component: https://github.com/syssi/esphome-seplos-bms#debugging

You should see the outgoing and incoming traffic now. Please double check the baudrate setting of the `uart` component. The BMS models uses different baudrates.

atze09, 2022-10-04:
> This is the ESPHome log. You can connect to your ESPHome node and view the logs by:
>
> ```
> esphome log esp8266-example.yaml
> ```
>
> Please enable the debug output of the `uart` component: [...]

Can i also obtain this wireless? 
i don't have the chance to have a linux machine next to my battery pack.hm

syssi, 2022-10-04:
Yes. You can obtain the log via the `api` component or if you use `mqtt` it will be retrieved from the MQTT broker. Just execute the command above. It will provide a list depending on your configuration YAML.

atze09, 2022-10-04:
If i make a plain install in my ESPHome i can get it with api key.
Do i only need to add:
external_components:
  - source: github://syssi/esphome-seplos-bms@main
To it and then install, is that all? and how can i shoose more battery banks then?
Sorry for noob question.

Or do i need to install it via linux machine.
i am struggling with the first installation.

syssi, 2022-10-04:
If this is your first ESPHome node you should use the `esp8266-example.yaml` as starting point: https://github.com/syssi/esphome-seplos-bms/blob/main/esp8266-example.yaml

Please make sure the RS485 module is connected to GPIO4/GPIO5. To make it more explicit I try to provide a step by step guide:

1. Use a linux host (if available) and make sure python3 + pip (the python package manage is installed). You could also use a windows machine. If you have the choice please pick a x86 based linux host.
2. Install the `esphome` package by `pip3 install esphome`. The command line interface tool `esphome` should be available now.
3. Try to execute the command line tool: `.local/bin/esphome`

```
$ ~/.local/bin/esphome 
usage: esphome [-h] [-v] [-q] [-s key value] command ...
esphome: error: the following arguments are required: command
```

4. Create a working directory for your esphome YAML configurations: `mkdir ~/esphome`
5. Goto the working directory (`cd ~/esphome`) and download the example configuration: `wget https://raw.githubusercontent.com/syssi/esphome-seplos-bms/main/esp8266-example.yaml`
6. Create a new file called `secrets.yaml` and provide your WiFi credentials:

```
echo "wifi_ssid: WIFI_SSID" > secrets.yaml
echo "wifi_password: WIFI_PASSWORD" >> secrets.yaml
```

7. Rename the `esp8266-example.yaml` to something meaningful: `mv esp8266-example.yaml seplos-bms.yaml`
8. Compile and flash the YAML configuration to your ESP: `esphome run seplos-bms.yaml`
9. After flashing the `esphome` tool tries to retrieve the log automatically. If you want to access the ESPHome node without re-flashing you should use `esphome logs seplos-bms.yaml`.

atze09, 2022-10-04:
Thank you very very much.
i got it, here is a part of the output log:
`[16:27:11][C][seplos_bms:150]:   Unit of Measurement: 'V'
[16:27:11][C][seplos_bms:150]:   Accuracy Decimals: 3
[16:27:11][C][seplos_bms:151]: Cell Voltage 4 'seplos-bms cell voltage 4'
[16:27:11][C][seplos_bms:151]:   Device Class: 'voltage'
[16:27:11][C][seplos_bms:151]:   State Class: 'measurement'
[16:27:11][C][seplos_bms:151]:   Unit of Measurement: 'V'
[16:27:11][C][seplos_bms:151]:   Accuracy Decimals: 3
[16:27:11][C][seplos_bms:152]: Cell Voltage 5 'seplos-bms cell voltage 5'
[16:27:11][C][seplos_bms:152]:   Device Class: 'voltage'
[16:27:11][C][seplos_bms:152]:   State Class: 'measurement'
[16:27:11][C][seplos_bms:152]:   Unit of Measurement: 'V'
[16:27:11][C][seplos_bms:152]:   Accuracy Decimals: 3
[16:27:11][C][seplos_bms:153]: Cell Voltage 6 'seplos-bms cell voltage 6'
[16:27:11][C][seplos_bms:153]:   Device Class: 'voltage'
[16:27:11][C][seplos_bms:153]:   State Class: 'measurement'
[16:27:11][C][seplos_bms:153]:   Unit of Measurement: 'V'
[16:27:11][C][seplos_bms:153]:   Accuracy Decimals: 3
[16:27:11][C][seplos_bms:154]: Cell Voltage 7 'seplos-bms cell voltage 7'
[16:27:11][C][seplos_bms:154]:   Device Class: 'voltage'
[16:27:11][C][seplos_bms:154]:   State Class: 'measurement'
[16:27:11][C][seplos_bms:154]:   Unit of Measurement: 'V'
[16:27:11][C][seplos_bms:154]:   Accuracy Decimals: 3
[16:27:11][C][seplos_bms:155]: Cell Voltage 8 'seplos-bms cell voltage 8'
[16:27:11][C][seplos_bms:155]:   Device Class: 'voltage'
[16:27:11][C][seplos_bms:155]:   State Class: 'measurement'
[16:27:11][C][seplos_bms:155]:   Unit of Measurement: 'V'
[16:27:11][C][seplos_bms:155]:   Accuracy Decimals: 3
[16:27:11][C][seplos_bms:156]: Cell Voltage 9 'seplos-bms cell voltage 9'
[16:27:11][C][seplos_bms:156]:   Device Class: 'voltage'
[16:27:11][C][seplos_bms:156]:   State Class: 'measurement'
[16:27:11][C][seplos_bms:156]:   Unit of Measurement: 'V'
[16:27:11][C][seplos_bms:156]:   Accuracy Decimals: 3
[16:27:11][C][seplos_bms:157]: Cell Voltage 10 'seplos-bms cell voltage 10'
[16:27:11][C][seplos_bms:157]:   Device Class: 'voltage'
[16:27:11][C][seplos_bms:157]:   State Class: 'measurement'
[16:27:11][C][seplos_bms:157]:   Unit of Measurement: 'V'
[16:27:13][D][seplos_modbus:163]: Send frame: ~20004642E00200FD37
[16:27:13][D][uart_debug:114]: >>> 7E:32:30:30:30:34:36:34:32:45:30:30:32:30:30:46:44:33:37:0D
[16:27:23][D][seplos_modbus:163]: Send frame: ~20004642E00200FD37
[16:27:23][D][uart_debug:114]: >>> 7E:32:30:30:30:34:36:34:32:45:30:30:32:30:30:46:44:33:37:0D
[16:27:33][D][seplos_modbus:163]: Send frame: ~20004642E00200FD37
[16:27:33][D][uart_debug:114]: >>> 7E:32:30:30:30:34:36:34:32:45:30:30:32:30:30:46:44:33:37:0D
[16:27:43][D][seplos_modbus:163]: Send frame: ~20004642E00200FD37
[16:27:43][D][uart_debug:114]: >>> 7E:32:30:30:30:34:36:34:32:45:30:30:32:30:30:46:44:33:37:0D
[16:27:53][D][seplos_modbus:163]: Send frame: ~20004642E00200FD37
[16:27:53][D][uart_debug:114]: >>> 7E:32:30:30:30:34:36:34:32:45:30:30:32:30:30:46:44:33:37:0D
[16:28:03][D][seplos_modbus:163]: Send frame: ~20004642E00200FD37
[16:28:03][D][uart_debug:114]: >>> 7E:32:30:30:30:34:36:34:32:45:30:30:32:30:30:46:44:33:37:0D
[16:28:13][D][seplos_modbus:163]: Send frame: ~20004642E00200FD37
[16:28:13][D][uart_debug:114]: >>> 7E:32:30:30:30:34:36:34:32:45:30:30:32:30:30:46:44:33:37:0D
[16:28:23][D][seplos_modbus:163]: Send frame: ~20004642E00200FD37
[16:28:23][D][uart_debug:114]: >>> 7E:32:30:30:30:34:36:34:32:45:30:30:32:30:30:46:44:33:37:0D
[16:28:33][D][seplos_modbus:163]: Send frame: ~20004642E00200FD37
[16:28:33][D][uart_debug:114]: >>> 7E:32:30:30:30:34:36:34:32:45:30:30:32:30:30:46:44:33:37:0D
[16:28:43][D][seplos_modbus:163]: Send frame: ~20004642E00200FD37
[16:28:43][D][uart_debug:114]: >>> 7E:32:30:30:30:34:36:34:32:45:30:30:32:30:30:46:44:33:37:0D
`

atze09, 2022-10-04:
And here the first part, i set the Baud to 19200, because it is the ine BMS description:

16:27:10][I][app:102]: ESPHome version 2022.6.3 compiled on Oct  4 2022, 16:15:59
[16:27:10][C][wifi:491]: WiFi:
[16:27:10][C][wifi:353]:   Local MAC: 58:BF:25:D7:2E:93
[16:27:10][C][wifi:354]:   SSID: 'DS17'
[16:27:10][C][wifi:355]:   IP Address: 192.168.17.76
[16:27:10][C][wifi:356]:   BSSID: A4:2B:B0:20:88:8D
[16:27:10][C][wifi:358]:   Hostname: 'seplos-bms'
[16:27:10][C][wifi:360]:   Signal strength: -80 dB ▂▄▆█
[16:27:10][C][wifi:364]:   Channel: 1
[16:27:10][C][wifi:365]:   Subnet: 255.255.255.0
[16:27:10][C][wifi:366]:   Gateway: 192.168.17.1
[16:27:10][C][wifi:367]:   DNS1: 192.168.17.2
[16:27:10][C][wifi:368]:   DNS2: 0.0.0.0
[16:27:10][C][logger:275]: Logger:
[16:27:10][C][logger:276]:   Level: DEBUG
[16:27:10][C][logger:277]:   Log Baud Rate: 115200
[16:27:10][C][logger:278]:   Hardware UART: UART0
[16:27:10][C][uart.arduino_esp8266:102]: UART Bus:
[16:27:10][C][uart.arduino_esp8266:103]:   TX Pin: GPIO4
[16:27:10][C][uart.arduino_esp8266:104]:   RX Pin: GPIO5
[16:27:10][C][uart.arduino_esp8266:106]:   RX Buffer Size: 384
[16:27:10][C][uart.arduino_esp8266:108]:   Baud Rate: 19200 baud
[16:27:10][C][uart.arduino_esp8266:109]:   Data Bits: 8
[16:27:10][C][uart.arduino_esp8266:110]:   Parity: NONE
[16:27:10][C][uart.arduino_esp8266:111]:   Stop bits: 1
[16:27:10][C][uart.arduino_esp8266:115]:   Using software serial
[16:27:10][C][seplos_modbus:129]: SeplosModbus:
[16:27:10][C][seplos_modbus:131]:  RX timeout: 150 ms
[16:27:10][C][seplos_bms:142]: SeplosBms:

syssi, 2022-10-04:
This log message indicates outgoing traffic:

```
[16:28:43][D][uart_debug:114]: >>> 7E:32:30:30:30:34:36:34:32:45:30:30:32:30:30:46:44:33:37:0D
```

If your BMS does respond you should see incoming traffic:

```
[16:28:43][D][uart_debug:114]: <<< ...
``` 

Could you provide a photo of your RS485 module? Is your RS485 module equipped with a TX and RX led? Does the TX led blink periodically?

atze09, 2022-10-06:
I did some research and found out that i should use GPIO 5 and 4 Pin as written in the code.
Before i used TX RX.
Is this in the description, maybe you could higlight it.
Thanks for your awesome work, now i only need to implement it into mi home assistant with any API key.

syssi, 2022-10-06:
I would like to add your BMS model as supported. Do you agree?

atze09, 2022-10-06:
yes of course

commit c0120bfb (2022-10-06): Add 1101-SP16 as supported

Closes: #28