syssi/esphome-seplos-bms #30: 1101-SP16 support multiple batteries

atze09, 2022-10-10:
Hello,
After i got it working with my single unit 1101-SP16 with Baud 19200, i wanted to try the multiple battery example, because my second BMS will arrive these days.
I changed the DIp to the following:
[image]
And extended the code to the multiple example, but the sensors in HA are not working.

Here is the debug log.
INFO Reading configuration /config/esphome/seplos-bms-1.yaml...
INFO Updating https://github.com/syssi/esphome-seplos-bms.git@main
INFO Starting log output from seplos-bms-1.local using esphome API
INFO Successfully connected to seplos-bms-1.local
[13:52:57][I][app:102]: ESPHome version 2022.9.4 compiled on Oct 10 2022, 13:42:49
[13:52:57][C][wifi:502]: WiFi:
[13:52:57][C][wifi:360]:   Local MAC: 48:3F:DA:87:75:CB
[13:52:57][C][wifi:361]:   SSID: [redacted]
[13:52:57][C][wifi:362]:   IP Address: 192.168.17.178
[13:52:57][C][wifi:363]:   BSSID: [redacted]
[13:52:57][C][wifi:365]:   Hostname: 'seplos-bms-1'
[13:52:57][C][wifi:367]:   Signal strength: -65 dB ▂▄▆█
[13:52:57][C][wifi:371]:   Channel: 1
[13:52:57][C][wifi:372]:   Subnet: 255.255.255.0
[13:52:57][C][wifi:373]:   Gateway: 192.168.17.1
[13:52:57][C][wifi:374]:   DNS1: 192.168.17.2
[13:52:57][C][wifi:375]:   DNS2: 0.0.0.0
[13:52:57][C][logger:275]: Logger:
[13:52:57][C][logger:276]:   Level: DEBUG
[13:52:57][C][logger:277]:   Log Baud Rate: 115200
[13:52:57][C][logger:278]:   Hardware UART: UART0
[13:52:57][C][uart.arduino_esp8266:102]: UART Bus:
[13:52:57][C][uart.arduino_esp8266:103]:   TX Pin: GPIO4
[13:52:57][C][uart.arduino_esp8266:104]:   RX Pin: GPIO5
[13:52:57][C][uart.arduino_esp8266:106]:   RX Buffer Size: 384
[13:52:57][C][uart.arduino_esp8266:108]:   Baud Rate: 19200 baud
[13:52:57][C][uart.arduino_esp8266:109]:   Data Bits: 8
[13:52:57][C][uart.arduino_esp8266:110]:   Parity: NONE
[13:52:57][C][uart.arduino_esp8266:111]:   Stop bits: 1
[13:52:57][C][uart.arduino_esp8266:115]:   Using software serial
[13:52:57][C][seplos_modbus:129]: SeplosModbus:
[13:52:57][C][seplos_modbus:131]:  RX timeout: 150 ms
[13:52:57][C][seplos_bms:142]: SeplosBms:
[13:52:57][C][seplos_bms:143]: Minimum Cell Voltage 'seplos-bms-1 bank 0 min cell voltage'
[13:52:57][C][seplos_bms:143]:   Device Class: 'voltage'
[13:52:57][C][seplos_bms:143]:   State Class: 'measurement'
[13:52:57][C][seplos_bms:143]:   Unit of Measurement: 'V'
[13:52:57][C][seplos_bms:143]:   Accuracy Decimals: 3
[13:52:57][C][seplos_bms:144]: Maximum Cell Voltage 'seplos-bms-1 bank 0 max cell voltage'
[13:52:57][C][seplos_bms:144]:   Device Class: 'voltage'
[13:52:57][C][seplos_bms:144]:   State Class: 'measurement'
[13:52:57][C][seplos_bms:144]:   Unit of Measurement: 'V'
[13:52:57][C][seplos_bms:144]:   Accuracy Decimals: 3
[13:52:57][C][seplos_bms:147]: Delta Cell Voltage 'seplos-bms-1 bank 0 delta cell voltage'
[13:52:57][C][seplos_bms:147]:   Device Class: 'voltage'
[13:52:57][C][seplos_bms:147]:   State Class: 'measurement'
[13:52:57][C][seplos_bms:147]:   Unit of Measurement: 'V'
[13:52:57][C][seplos_bms:147]:   Accuracy Decimals: 3
[13:52:57][C][seplos_bms:170]: Total Voltage 'seplos-bms-1 bank 0 total voltage'
[13:52:57][C][seplos_bms:170]:   Device Class: 'voltage'
[13:52:57][C][seplos_bms:170]:   State Class: 'measurement'
[13:52:57][C][seplos_bms:170]:   Unit of Measurement: 'V'
[13:52:57][C][seplos_bms:170]:   Accuracy Decimals: 2
[13:52:57][C][seplos_bms:171]: Current 'seplos-bms-1 bank 0 current'
[13:52:57][C][seplos_bms:171]:   Device Class: 'current'
[13:52:57][C][seplos_bms:171]:   State Class: 'measurement'
[13:52:57][C][seplos_bms:171]:   Unit of Measurement: 'A'
[13:52:57][C][seplos_bms:171]:   Accuracy Decimals: 2
[13:52:57][C][seplos_bms:171]:   Icon: 'mdi:current-dc'
[13:52:57][C][seplos_bms:172]: Power 'seplos-bms-1 bank 0 power'
[13:52:57][C][seplos_bms:172]:   Device Class: 'power'
[13:52:57][C][seplos_bms:172]:   State Class: 'measurement'
[13:52:57][C][seplos_bms:172]:   Unit of Measurement: 'W'
[13:52:57][C][seplos_bms:172]:   Accuracy Decimals: 2
[13:52:57][C][seplos_bms:176]: State of charge 'seplos-bms-1 bank 0 state of charge'
[13:52:57][C][seplos_bms:176]:   State Class: 'measurement'
[13:52:57][C][seplos_bms:176]:   Unit of Measurement: '%'
[13:52:57][C][seplos_bms:176]:   Accuracy Decimals: 1
[13:52:57][C][seplos_bms:176]:   Icon: 'mdi:battery-50'
[13:52:57][C][seplos_bms:142]: SeplosBms:
[13:52:57][C][seplos_bms:142]: SeplosBms:
[13:52:57][C][captive_portal:088]: Captive Portal:
[13:52:57][C][mdns:100]: mDNS:
[13:52:57][C][mdns:101]:   Hostname: seplos-bms-1
[13:52:57][C][ota:089]: Over-The-Air Updates:
[13:52:57][C][ota:090]:   Address: seplos-bms-1.local:8266
[13:52:57][C][ota:093]:   Using Password.
[13:52:57][C][api:138]: API Server:
[13:52:57][C][api:139]:   Address: seplos-bms-1.local:6053
[13:52:57][C][api:141]:   Using noise encryption: YES

[13:52:59][D][uart_debug:114]: >>> 7E:32:30:30:30:34:36:34:32:45:30:30:32:30:30:46:44:33:37:0D

[13:52:59][D][uart_debug:114]: >>> 7E:32:30:30:32:34:36:34:32:45:30:30:32:30:32:46:44:33:33:0D

[13:53:03][D][uart_debug:114]: >>> 7E:32:30:30:31:34:36:34:32:45:30:30:32:30:31:46:44:33:35:0D
[13:53:03][D][uart_debug:114]: <<< 7E:32:30:30:31:34:36:30:30:31:30:39:36:30:30:30:31:31:30:30:44:30:44:30:44:30:44:30:44:30:43:30:44:30:44:30:44:30:44:30:44:30:45:30:44:30:45:30:44:30:45:30:44:30:45:30:44:30:44:30:44:30:45:30:44:30:45:30:44:30:46:30:44:30:46:30:44:30:45:30:44:30:45:30:36:30:42:44:32:30:42:44:34:30:42:43:43:30:42:44:30:30:42:39:45:30:42:41:42:30:30:30:30:31:34:45:32:36:39:45:34:30:41:37:30:44:41:30:33:41:41:37:30:44:41:30:30:36:42:30:33:45:38:31:34:44:46:30:30:30
[13:53:03][I][seplos_bms:031]: Telemetry frame received
[13:53:03][D][uart_debug:114]: <<< 30:30:30:30:30:30:30:30:30:30:30:30:30:44:43:41:32:0D

[13:53:09][D][uart_debug:114]: >>> 7E:32:30:30:30:34:36:34:32:45:30:30:32:30:30:46:44:33:37:0D

[13:53:09][D][uart_debug:114]: >>> 7E:32:30:30:32:34:36:34:32:45:30:30:32:30:32:46:44:33:33:0D

[13:53:13][D][uart_debug:114]: >>> 7E:32:30:30:31:34:36:34:32:45:30:30:32:30:31:46:44:33:35:0D
[13:53:13][D][uart_debug:114]: <<< 7E:32:30:30:31:34:36:30:30:31:30:39:36:30:30:30:31:31:30:30:44:30:44:30:44:30:44:30:44:30:43:30:44:30:44:30:44:30:44:30:44:30:45:30:44:30:45:30:44:30:45:30:44:30:45:30:44:30:44:30:44:30:45:30:44:30:45:30:44:30:46:30:44:30:45:30:44:30:44:30:44:30:44:30:36:30:42:44:32:30:42:44:34:30:42:43:43:30:42:44:30:30:42:39:45:30:42:41:41:30:30:30:30:31:34:45:32:36:39:45:34:30:41:37:30:44:41:30:33:41:41:37:30:44:41:30:30:36:42:30:33:45:38:31:34:44:46:30:30:30
[13:53:13][I][seplos_bms:031]: Telemetry frame received

syssi, 2022-10-10:
Could you provide your configuration YAML? I would like to check the addresses.

atze09, 2022-10-10:
Or is this only working when i have my second BMS physically installed?
Of course here is the yaml.
substitutions:
  name: seplos-bms-1
  battery_bank0: "${name} bank 0"
  battery_bank1: "${name} bank 1"
  battery_bank2: "${name} bank 2"
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin: GPIO4
  rx_pin: GPIO5

esphome:
  name: ${name}

external_components:
  - source: ${external_components_source}
    refresh: 0s

esp8266:
  board: d1_mini

# Enable logging
logger:
  level: DEBUG
# Enable Home Assistant API
api:
  encryption:
    key: "API Key"

ota:
  password: "pass"

wifi:
  ssid: ssid
  password: pass
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Seplos-Bms-1 Fallback Hotspot"
    password: "Pass"

captive_portal:

uart:
  id: uart0
  # Please set the default baudrate of your Seplos BMS model here. It's sometimes 19200 baud instead of 9600.
  baud_rate: 19200
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  rx_buffer_size: 384
  debug:
    direction: BOTH
#    after:
#      delimiter: "\r"
#    sequence:
#      - lambda: UARTDebug::log_string(direction, bytes);

seplos_modbus:
  id: modbus0
  uart_id: uart0
  rx_timeout: 150ms

#seplos_bms:
#  id: bms0
#  seplos_modbus_id: modbus0
#  update_interval: 10s

seplos_bms:
  - id: battery_bank0
    address: 0x00
    seplos_modbus_id: modbus0
    update_interval: 10s
  - id: battery_bank1
    address: 0x01
    seplos_modbus_id: modbus0
    update_interval: 10s
  - id: battery_bank2
    address: 0x02
    seplos_modbus_id: modbus0
    update_interval: 10s

sensor:
  - platform: seplos_bms
    seplos_bms_id: battery_bank0
    min_cell_voltage:
      name: "${battery_bank0} min cell voltage"
    max_cell_voltage:
      name: "${battery_bank0} max cell voltage"
    delta_cell_voltage:
      name: "${battery_bank0} delta cell voltage"
    average_cell_voltage:
      name: "${battery_bank0} average cell voltage"
    total_voltage:
      name: "${battery_bank0} total voltage"
    current:
      name: "${battery_bank0} current"
    power:
      name: "${battery_bank0} power"
    state_of_charge:
      name: "${battery_bank0} state of charge"

NosIreland, 2022-10-12:
I do not see how 2 seplos or more would work in parrallel. Even with native seplos software you cannot monitor pack00. This is excpected because pack00 becomes a master in modbus and there can only be 1 master.

syssi, 2022-10-17:
@NosIreland Did I understand you correctly: If we want to monitor multiple batteries / addresses we have to start with `0x01`? In general it should be possible to poll multiple batteries on the same RS485 bus. I've replicated the behaviour of the windows application here.

syssi, 2022-10-19:
I assume the `esp8266-example-multiple-battery-banks.yaml` isn't correct. It should be:

```
seplos_bms:
  - id: battery_bank0
    address: 0x01
    seplos_modbus_id: modbus0
    update_interval: 10s
  - id: battery_bank1
    address: 0x02
    seplos_modbus_id: modbus0
    update_interval: 10s
  - id: battery_bank2
    address: 0x03
    seplos_modbus_id: modbus0
    update_interval: 10s
```

Is my understanding correct: If you talk to multiple Seplos via RS485 the enumeration of the devices must start with `0x01`?

syssi, 2022-10-19:
I tried to improve the examples: https://github.com/syssi/esphome-seplos-bms/commit/81221c766ed2fbe0fc683b561a8387400bcaa59f

atze09, 2022-10-19:
> I tried to improve the examples: [81221c7](https://github.com/syssi/esphome-seplos-bms/commit/81221c766ed2fbe0fc683b561a8387400bcaa59f)

Looks good thank you.
My setup looks a bit different, but i don't know if it is possible to solve it this way.
I connected my master battery bank via can bus to my victron system, that's why i need the following DIP states.
[image]

i don't know if i can get all the data in parallel from the RS485 master.

Your example file currently works for the master battery bank.
In the logs i saw many "Unhandled data received".
[logs_seplos-bms-1_upload.txt](https://github.com/syssi/esphome-seplos-bms/files/9823553/logs_seplos-bms-1_upload.txt)
Here the log.
Maybe you can integrate both multiple versions (RS485 and CAN). or maybe i need to try another RS485 protocol setting in the BMS

atze09, 2022-10-21:
Update from my trials:
If i configure  the BMS with the DIP status:
Pack 1: Dips all off
Pack 2: DIP 1 on, all others off
Your code works with battery bank 0 0x00 and battery bank 1 0x01, as you had before.
But then the CAN only sees 1 Pack.

I am in contact with Seplos about this issue, they stated that multiple packs are only possible for either CAN or RS485.
I don't like this kind of feedback, but this is the status.

@syssi but could you help to handle the unhandlet data, maybe this issue could be solved without Seplos.
Thanks

syssi, 2022-10-21:
Could you also try:

Pack 1: DIP 1 on, all others off  (address 0x01 in YAML)
Pack 2: DIP 2 on, all otehrs off  (address 0x02 in YAML)

Does this setup work too? Because the manual states `addr 0x00` is reserved/dedicated for single pack setups only.

syssi, 2022-10-21:
@atze09 Could you contact me via discord (`syssi#5175`)?

atze09, 2022-10-21:
> syssi#5175

Yes i have added you. 
The answer from the Seplos engineer to my log:
This is the data from the master BMS, but it is not the data of the master BMS itself, but the data that has been aggregated and processed.

atze09, 2022-10-21:
> Could you also try:
> 
> Pack 1: DIP 1 on, all others off (address 0x01 in YAML) Pack 2: DIP 2 on, all otehrs off (address 0x02 in YAML)
> 
> Does this setup work too? Because the manual states `addr 0x00` is reserved/dedicated for single pack setups only.

This works also, but then i don't have any CAN bus output

NosIreland, 2022-10-21:
> @NosIreland Did I understand you correctly: If we want to monitor multiple batteries / addresses we have to start with `0x01`? In general it should be possible to poll multiple batteries on the same RS485 bus. I've replicated the behaviour of the windows application here.
The problem is that when dipswitches are changed and batteries are daisy chained one of the seplos BMS becomes master at 0x00 address and this makes it imposible to monitor. We can only see batteries 0x01 and above but not 0x00

atze09, 2022-10-21:
> The problem is that when dipswitches are changed and batteries are daisy chained one of the seplos BMS becomes master at 0x00 address and this makes it imposible to monitor. We can only see [...]

But I could.monitor 0x00 and 0x01

NosIreland, 2022-10-22:
> 
But the batteries were working individually and in that case you cannot connect them properly to inverter as the data is not agregated.

atze09, 2022-10-23:
The feedback of seplos engineer is:
Multiple battery data of current BMS version can only be obtained by CAN Bus or RS485 not both in parallel.
If CAN is connected master battery data can not be provided by RS485 only second batteries data.
I will close this issue here.
Just saw Seplos launches new revision 16 of their BMS, maybe it is possible with new revision.
@syssi my workaround will be an NEEY balancer und your esphome jkbms Module to get the cell voltage data of master BMS

closed 2022-10-23 by atze09