syssi/esphome-seplos-bms #31: Seplosbms CRC check error

miguel300477, 2022-10-15:
hi,

I have been trying to get data but i keep getting this error:

[W][seplos_modbus:101]: SeplosBms CRC Check failed! DD5D != DCFD

from time to time i get the bms data ok, but most of the time is that error.
I played with the rx timeout  and i get data, but alot of that error.
[logs_seplosbms_logs (2).txt](https://github.com/syssi/esphome-seplos-bms/files/9792498/logs_seplosbms_logs.2.txt)

I have a seplos BMS from millenials , 150A 16cells.
this is my config:

`substitutions:
  name: seplos-bms
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin: GPIO4
  rx_pin: GPIO5

esphome:
  name: ${name}

esp8266:
  board: esp01_1m

external_components:
  - source: ${external_components_source}
    refresh: 0s

# Enable logging
logger:
  
# Enable Home Assistant API
api:
  encryption:
    key: "LElVpf+yOX/T7gPTYWuijvEsBFLPyGlGuRXZkKLq5js="

ota:
  password: "fbce9898f7a22f8bbc01ce2114efe287"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Seplosbms Fallback Hotspot"
    password: "OiVjReZiaGkq"

captive_portal:

uart:
  id: uart0
  baud_rate: 19200
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: true
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

seplos_modbus:
  id: modbus0
  uart_id: uart0
  rx_timeout: 1100ms

seplos_bms:
  id: bms0
  seplos_modbus_id: modbus0
  update_interval: 10s

sensor:
  - platform: seplos_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    temperature_1:
      name: "${name} temperature 1"
    temperature_2:
      name: "${name} temperature 2"
    temperature_3:
      name: "${name} temperature 3"
    temperature_4:
      name: "${name} temperature 4"
    temperature_5:
      name: "${name} environment temperature"
    temperature_6:
      name: "${name} mosfet temperature"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    residual_capacity:
      name: "${name} residual capacity"
    battery_capacity:
      name: "${name} battery capacity"
    rated_capacity:
      name: "${name} rated capacity"
    state_of_charge:
      name: "${name} state of charge"
    charging_cycles:
      name: "${name} charging cycles"
    state_of_health:
      name: "${name} state of health"
    port_voltage:
      name: "${name} port voltage"`
can you help me?

miguel300477, 2022-10-19:
Hi, i have sorted it out.

I was missing the RX buffer.

closed 2022-10-19 by miguel300477

syssi, 2022-10-19:
Good job! I will add a note to the example.

miguel300477, 2022-10-19:
Yes,

I added this from multiple battery config.

 rx_buffer_size: 384

I have single battery but without that it would not work. I got some data but mostly CRC check error.
With rx buffer size works perfect with default rx timeout.

Geat job.