syssi/esphome-seplos-bms #37: esp32 example file missing and max485 support

JacquesdeBruyn, 2022-12-08:
Hi, I'm new to esphome and home assistant. I can't seem to get the 8266 code working on my 32. I have modified where gpio pins differ and added the flow control pin under uart but haven't gotten any readings out yet. I just want to get readings out to home assistant to mange loads. Just voltage, soc and amps/watts in out needed mostly. I can post my modified code if you need it. it does flash no problem but I don't get any return in the logs. I have a usb to 485 and the pc software for this bms

syssi, 2022-12-08:
Could you provide your configuration YAML?

JacquesdeBruyn, 2022-12-08:
Here is what I have cobbled together.

substitutions:
  name: seplos-bms
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin: GPIO1
  rx_pin: GPIO3

esphome:
  name: test1

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

external_components:
  - source: ${external_components_source}
    refresh: 0s

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "G6tBZkXCx152iuOzVdDatSC6VCr7tfLeaoscQBkMulE="

ota:
  password: "7856dff1fbb6da712c947d4ad2c4d8e2"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test1 Fallback Hotspot"
    password: "zN1Djnt6O6Bo"

uart:
  id: uart0
  # Please set the default baudrate of your Seplos BMS model here. It's sometimes 19200 baud instead of 9600.
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  # The increased RX buffer size is important because
  # the full BMS response must fit into the buffer
  rx_buffer_size: 384

seplos_modbus:
  flow_control_pin: 4
  id: modbus0
  uart_id: uart0
  rx_timeout: 150ms

seplos_bms:
  id: bms0
  # Dip switch configuration of a single pack setup / address 0x0
  #  8    7    6    5    4    3    2    1
  # off, off, off, off, off, off, off, off
  address: 0x00
  seplos_modbus_id: modbus0
  update_interval: 10s

sensor:
  - platform: seplos_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    temperature_1:
      name: "${name} temperature 1"
    temperature_2:
      name: "${name} temperature 2"
    temperature_3:
      name: "${name} temperature 3"
    temperature_4:
      name: "${name} temperature 4"
    temperature_5:
      name: "${name} environment temperature"
    temperature_6:
      name: "${name} mosfet temperature"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    residual_capacity:
      name: "${name} residual capacity"
    battery_capacity:
      name: "${name} battery capacity"
    rated_capacity:
      name: "${name} rated capacity"
    state_of_charge:
      name: "${name} state of charge"
    charging_cycles:
      name: "${name} charging cycles"
    state_of_health:
      name: "${name} state of health"
    port_voltage:
      name: "${name} port voltage"

syssi, 2022-12-08:
Is the flow control pin required by your RS485 module? If the pin is connected to `GPIO4` please use this configuration:

```
seplos_modbus:
  flow_control_pin: GPIO4
```

The implementation sends a request to the BMS periodically. Does the `TX` led of your RS485 module blink periodically?

JacquesdeBruyn, 2022-12-08:
I had a couple issues, so far the baud rate is in fact 19200(confirmed with pc software) I've also changed a couple pins to fix logging errors. its' responding with Invalid header now. BMS is 1101-SP06 by the looks of it.

Changed flow to gpio18 and and rx/tx to 16/17

syssi, 2022-12-08:
You could enable the `debug` feature of the `uart` component by:

```
uart:
  id: uart0
  baud_rate: 19200
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  rx_buffer_size: 384
  debug:
    dummy_receiver: false
    direction: BOTH
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

And provide some logs / traffic captures.

JacquesdeBruyn, 2022-12-08:
Here is a bit of the return. I noticed the log baud rate shows as 115200.

 [logs_test1_upload.txt](https://github.com/syssi/esphome-seplos-bms/files/10185074/logs_test1_upload.txt)

JacquesdeBruyn, 2022-12-08:
Correction bms is 1101-SP05

It is working now. I'll comment the code in here tomorrow.

JacquesdeBruyn, 2022-12-08:
I should thank you for putting in all the hard work. I'm pretty sure there are plenty of us that appreciate it.

syssi, 2022-12-13:
@JacquesdeBruyn Please don't forget to leave a comment what you have done to make it work. Thanks in advance!

JacquesdeBruyn, 2022-12-14:
Hi, the holiday rush hit here. so before I get busy again here is my running code. Also be weary of bad RJ45 connectors. That was a fun one to find.

Anyways, Code as follows:
substitutions:
  name: seplos-bms
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin: GPIO17
  rx_pin: GPIO16

esphome:
  name: test1

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

external_components:
  - source: ${external_components_source}
    refresh: 0s

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "G6tBZkXCx152iuOzVdDatSC6VCr7tfLeaoscQBkMulE="

ota:
  password: "7856dff1fbb6da712c947d4ad2c4d8e2"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test1 Fallback Hotspot"
    password: "zN1Djnt6O6Bo"

uart:
  id: uart0
  # Please set the default baudrate of your Seplos BMS model here. It's sometimes 19200 baud instead of 9600.
  baud_rate: 19200
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  # The increased RX buffer size is important because
  # the full BMS response must fit into the buffer
  rx_buffer_size: 384

seplos_modbus:
  flow_control_pin: 18
  id: modbus0
  uart_id: uart0
  rx_timeout: 150ms

seplos_bms:
  id: bms0
  # Dip switch configuration of a single pack setup / address 0x0
  #  8    7    6    5    4    3    2    1
  # off, off, off, off, off, off, off, off
  address: 0x00
  seplos_modbus_id: modbus0
  update_interval: 1s

sensor:
  - platform: seplos_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    temperature_1:
      name: "${name} temperature 1"
    temperature_2:
      name: "${name} temperature 2"
    temperature_3:
      name: "${name} temperature 3"
    temperature_4:
      name: "${name} temperature 4"
    temperature_5:
      name: "${name} environment temperature"
    temperature_6:
      name: "${name} mosfet temperature"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    residual_capacity:
      name: "${name} residual capacity"
    battery_capacity:
      name: "${name} battery capacity"
    rated_capacity:
      name: "${name} rated capacity"
    state_of_charge:
      name: "${name} state of charge"
    charging_cycles:
      name: "${name} charging cycles"
    state_of_health:
      name: "${name} state of health"
    port_voltage:
      name: "${name} port voltage"

syssi, 2022-12-14:
Let's close the issue because it's resolved.

closed 2022-12-14 by syssi