syssi/esphome-seplos-bms #41: no connect to seplos BMS

linuxdep, 2023-02-26:
Hi,
i have installed to ESP32 with TTL to RS485 Board and i can't conect the BMS.
This is the only information on log, debug ist enabled.
```
INFO Starting log output from seplos-bms01.local using esphome API
WARNING Can't connect to ESPHome API for seplos-bms01.local: Handshake timed out
WARNING seplos-bms01.local: Connection error occurred: Connection lost
```
i have changed TX and RX, changed A and B, always the same result, no connect. With original software from seplos it works fine with 19200 baud...
how can i do that its work???

syssi, 2023-02-27:
The log cannot be retrieved because the ESP cannot be reached properly via network. Could you attach the ESP to your computer and retrieve the log messages via the serial interface just for testing and getting an idea what's going on?

linuxdep, 2023-02-27:
OK, I found it. Had used an old network cable, which I had sorted out, built a new one, now it works. If cables no longer work, you should cut off the plug right away...
If I see values in the log, I just have to process it further.
Thank you very much, also for your great work here.

closed 2023-02-27 by linuxdep