syssi/esphome-seplos-bms #45: CAN socket on latest Seplos revision has rs485

archi, 2023-04-03:
Thanks for the project! And the Neey balancer one! (And for the generous FOSS work overall <3).

I have the latest revision of the Seplos 200A 48V 16S BMS (if the seller didn't lie - I haven't disassembled the battery far enough to verify). That one supports RS485 on the CAN socket. Pins are 1&2 (A: green, B: orange), did not test 7/8. Ground is disconnected.

Maybe you want to reflect this in your docs? Should I open a PR?

(I'm not sure if that's advertised anywhere, but I just realized that it shouldn't work when comparing my setup with that of a friend; though the Seplos advertises only half the charging current and I'm afraid that's because it thinks the esp is another charger).

syssi, 2023-04-04:
Thanks for your feedback! I'm would be happy about your contribution. I don't think the ESP connected via RS485(!) is able to interver the current measurement.

syssi, 2023-04-04:
Could you provide a photo of your BMS or the accessible connectors?

syssi, 2023-04-07:
@archi Friendly reminder. Could you provide a photo of your BMS / the accessible connectors?

archi, 2023-04-07:
Hi @syssi, the battery is assembled in one of these metal boxes (like seplos mason). A photo of the exterior is probably rather uninteresting, and I'd like to avoid having to disassemble the front assembly.

Since this is part of a bigger group order, I can attach some images of what should be the same device:

[image]
[image]

External connectors are the usual three rj45 sockets (that's my actual battery):
[image]

andreas-bulling, 2023-08-11:
I have the same case/battery - does the "normal" software version and connection through the RS485-TTL converter not work or why does this issue exist?

andreas-bulling, 2023-08-11:
Oh, @archi - and which Neey balancer project (this I also have in my battery - I somehow feel we might have joined the same group order on the Akkudoktor forum... *g)

archi, 2023-08-11:
@andreas-bulling yes, we are :)

I don't understand the intention of your question ;-) I have no idea if the original software works or not, since I don't use it (I don't have a Windows laptop, and my desktop is not exactly portable). I just use a different set of pins for the esphome.

andreas-bulling, 2023-08-11:
Sorry, I realise that my question is not clear. With "normal software version" I didn't mean the Windows software but the code and corresponding instructions here in this repository. Particularly the RS485/TTL hardware converter that is required. Did you find a different way of connecting the Seplos battery to ESPHome? Does it work equally well?

Also, if you don't use the Windows software - how do you configure the BMS and Neey? I've just assembled my battery today and don't see either of them when scanning for Bluetooth devices...

archi, 2023-08-11:
I'd like to point out: This isn't a tech support issue, this is an issue regarding improving the docs. You should/could have asked the same questions in the forum or WA group of the group buy; and I guess that has been discussed there several times already. With that being said...

The essential info is in the opening post: I use pins 1&2 (A: green, B: orange). Gnd not connected. The TTL converter is just some cheap part ([this one](https://www.amazon.de/gp/product/B09Z2GTMJ8/)). The code version is whatever was in `main` during early April. I'd guess b700842.
The BMS ESP (can't recall if 8266 or 32) fails after some time (hours, maybe a day) and needs manual rebooting. But I don't require the data from that, since I get reliable cell level voltage from the Neey via Bluetooth with an ESP32. And because I'm currently still busy with renovations, so I didn't bother investigating/opening an issue.

For BT connection to the Neey & BMS you need some luck with your phone: Works well with my Pixel 4a and 4a 5G, but not at all with my Huawei tablet. Mind: The app versions matter. The Neey could be installed from the Play Store, but it's not listed anymore. [It's mirrored here, but don't ask me if that link is safe.](https://apkpure.com/neey-electronics/com.neey.app) The BMS needs some apk. [I've saved it to my Google Drive, but no idea if there is newer/better version out.](https://drive.google.com/drive/folders/1-BtnDrIkkIRSzY9cJJ_HayW9NNFx4I-_?usp=sharing).

syssi, 2023-08-12:
> The BMS ESP (can't recall if 8266 or 32) fails after some time (hours, maybe a day) and needs manual rebooting.

If this is still an issue I would be happy about an bug report some day. I'm not aware of any stability issues.

nkinnan, 2024-09-16:
I have run into the same reboot issue on my EG4 based on what appears to be the same hardware but different firmware (commands are slightly different).  The solution in my case was to increase the timeout between requests.  The BMS seems to get overwhelmed and lock up if you send it too many commands when it hasn't responded yet.  It may or may not help with this different firmware.  Timeout of 2 seconds helped for me, you may consider increasing it to as much as say 5 in case of no response.

broekieman, 2025-03-23:
Am I right in my connection rj45 wire 1 en 2 to usb>uart/ttl A and B?