syssi/esphome-seplos-bms #47: Seplos 200 a 16s CAN

juerse, 2023-04-04:
Hi, I have a seplos 200a 16s with CAN1101 firmware v16.04.06_PN01_220924_9600 which communicates with using the 9600 band. The dip switches are 10000000. What are the settings for correct operation? Thank you

syssi, 2023-04-04:
Did you try the example configuration already? Could you turn all dip switches off?

juerse, 2023-04-05:
I have try with this

substitutions:
  name: esp8266-seplos
  device_description: "Monitor a Seplos BMS via RS485"
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin: GPIO5
  rx_pin: GPIO4

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-seplos-bms"
    version: 1.1.0

esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  networks:
  - ssid: xxxxxxxxxxxxxxxxxxxx
    password: xxxxxxxxxxxxxx
  - ssid: xxxxxxxxxxxxxxxxxxxx
    password: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

api:
ota:
logger:
  level: DEBUG

#web_server:
#  port: 80
uart:
  id: uart0
  # Please set the default baudrate of your Seplos BMS model here. It's sometimes 19200 baud instead of 9600.
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  # The increased RX buffer size is important because
  # the full BMS response must fit into the buffer
  rx_buffer_size: 384

seplos_modbus:
  id: modbus0
  uart_id: uart0
  rx_timeout: 150ms

seplos_bms:
  id: bms0
  # Dip switch configuration of a single pack setup / address 0x0
  #  8    7    6    5    4    3    2    1
  # off, off, off, off, off, off, off, off
  address: 0x01
  # Known protocol versions: 0x20 (Seplos), 0x26 (Boqiang)
  protocol_version: 0x20
  seplos_modbus_id: modbus0
  update_interval: 10s

sensor:
  - platform: seplos_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
[... 764 lines cut ...]
[12:25:59][D][sensor:126]: 'esp8266-seplos state of charge': Sending state 69.80000 % with 1 decimals of accuracy
[12:25:59][D][sensor:126]: 'esp8266-seplos rated capacity': Sending state 280.00000 Ah with 2 decimals of accuracy
[12:25:59][D][sensor:126]: 'esp8266-seplos charging cycles': Sending state 22.00000  with 0 decimals of accuracy
[12:25:59][D][sensor:126]: 'esp8266-seplos state of health': Sending state 100.00000 % with 1 decimals of accuracy
[12:25:59][D][sensor:126]: 'esp8266-seplos port voltage': Sending state 54.54000 V with 2 decimals of accuracy
[12:25:59][W][seplos_modbus:086]: Invalid header
[12:26:00][W][seplos_modbus:086]: Invalid header
[12:26:00][W][seplos_bms:021]: Unhandled data received (data_len: 0x02): 20.01.46.42.E0.02.01 (7)
[12:26:00][I][seplos_bms:030]: Telemetry frame (81 bytes) received
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 1': Sending state 3.39600 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 2': Sending state 3.40100 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 3': Sending state 3.40900 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 4': Sending state 3.40400 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 5': Sending state 3.39900 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 6': Sending state 3.41300 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 7': Sending state 3.41000 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 8': Sending state 3.40200 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 9': Sending state 3.40200 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 10': Sending state 3.40600 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 11': Sending state 3.40400 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 12': Sending state 3.40600 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 13': Sending state 3.40400 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 14': Sending state 3.41000 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 15': Sending state 3.40900 V with 3 decimals of accuracy
[12:26:00][D][sensor:126]: 'esp8266-seplos cell voltage 16': Sending state 3.40100 V with 3 decimals of accuracy

syssi, 2023-04-05:
It looks like you are very close to your goal! If I understand your current setup correctly Solar Assistants sends a

```
>>> 20.01.46.42.E0.02.01
```

periodically to the BMS. The BMS responds properly with a telemetry frame which can be decoded properly. Please enable the `debug` output of the `uart` component. See https://github.com/syssi/esphome-seplos-bms#debugging

Please provide a new log (ESP attached via RS485 to the BMS) capture. This time you should see the outgoing RAW traffic of this implementation. Is there a difference to the request payload of Solar Assistant?

juerse, 2023-04-05:
I have use this configuration

substitutions:
  name: esp8266-seplos
  device_description: "Monitor a Seplos BMS via RS485"
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin: GPIO5
  rx_pin: GPIO4

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-seplos-bms"
    version: 1.1.0

esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  networks:
  - ssid: xxxxxxxxxxxxx
    password: xxxxxxxxxxxxxx  
  - ssid: xxxxxxxxxxxxxxx
    password: xxxxxxxxxxxxx

api:
ota:
logger:

#web_server:
#  port: 80
uart:
  id: uart0
  # Please set the default baudrate of your Seplos BMS model here. It's sometimes 19200 baud instead of 9600.
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  # The increased RX buffer size is important because
  # the full BMS response must fit into the buffer
  rx_buffer_size: 384
  debug:
    dummy_receiver: false
    direction: BOTH
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
      
seplos_modbus:
  id: modbus0
  uart_id: uart0
  rx_timeout: 150ms

seplos_bms:
  id: bms0
  # Dip switch configuration of a single pack setup / address 0x0
  #  8    7    6    5    4    3    2    1
  # off, off, off, off, off, off, off, off
  address: 0x01
  # Known protocol versions: 0x20 (Seplos), 0x26 (Boqiang)
  protocol_version: 0x20
  seplos_modbus_id: modbus0
  update_interval: 10s

and this is a log a port with solar assistant and esp
[... 875 lines cut ...]
[22:52:43][W][seplos_modbus:086]: Invalid header
[22:52:43][W][seplos_modbus:086]: Invalid header
[22:52:43][W][seplos_modbus:086]: Invalid header
[22:52:43][W][seplos_modbus:086]: Invalid header
[22:52:43][W][seplos_modbus:086]: Invalid header
[22:52:43][W][seplos_modbus:086]: Invalid header
[22:52:43][D][uart_debug:158]: <<< "000000B830B1CDB8B\r"
[22:52:43][W][seplos_modbus:086]: Invalid header
[22:52:44][D][uart_debug:158]: <<< "~20014642E00201FD35\r"
[22:52:45][W][seplos_bms:021]: Unhandled data received (data_len: 0x02): 20.01.46.42.E0.02.01 (7)
[22:52:45][W][seplos_modbus:086]: Invalid header
[22:52:45][W][seplos_modbus:086]: Invalid header
[22:52:45][W][seplos_modbus:086]: Invalid header
[22:52:45][D][uart_debug:158]: <<< "~2001460010960001100CF00CEC0CE40CEC0CEF0CE90CEA0CED0CEE0CEA0CEB0CE50CED0CE90CEC0CEF060B8B0B8B0B8D0B8C0B830B5BFC6514AC54E30A6D6003086D60001603E814AD002"
[22:52:45][W][seplos_modbus:086]: Invalid header
[22:52:45][W][seplos_modbus:086]: Invalid header
[22:52:45][W][seplos_modbus:086]: Invalid header
[22:52:45][W][seplos_modbus:086]: Invalid header
[22:52:45][W][seplos_modbus:086]: Invalid header
[22:52:45][W][seplos_modbus:086]: Invalid header
[22:52:45][W][seplos_modbus:086]: Invalid header
[22:52:45][W][seplos_modbus:086]: Invalid header
[22:52:45][W][seplos_modbus:086]: Invalid header
[22:52:45][D][uart_debug:158]: <<< "300000B830B1CDB9D\r"
[22:52:45][W][seplos_modbus:086]: Invalid header

syssi, 2023-04-06:
Could you provide a photo of your RS485 converter?

juerse, 2023-04-06:
This is https://www.amazon.it/dp/B082D5PXJ4

syssi, 2023-04-06:
This is the RS485 converter you are using between Solar Assistant and the BMS. How did you attach the ESP?

juerse, 2023-04-06:
First Port esp second solar assistant , 2 Port are parallel port

syssi, 2023-04-06:
Could you show me the RS485 converter module between your ESP and BMS?

juerse, 2023-04-06:
it is as I showed in the picture. direct esp connected on the port and
direct connected converter on the other port

syssi, 2023-04-06:
Which picture are you talking about? I did see the Amazon link to a USB-to-RS485 adapter. Do I miss something?

juerse, 2023-04-06:
[image]
[image]

syssi, 2023-04-06:
Please change the retrieval of the external component from the `main` branch to this feature branch:

```
substitutions:
  external_components_source: github://syssi/esphome-seplos-bms@debug-decoder
```

Please flash your ESP another time and provide a new log. The "Invalid header" response indicates the rejected byte now. Furthermore another warning is raised if the RX buffer gets cleared due to a timeout (big gaps of silence between two bytes).

We are trying to solve two issues here:

1. What's the noise on the wire triggering the "Invalid header" warnings
2. Why is your ESP unable to request a telemetry frame at the BMS

syssi, 2023-04-06:
Could you provide a close up of your RJ45 connector? Which pins do you use exactly? Is GND connected or not?

juerse, 2023-04-06:
i have connect 1-2-3 pin , gnd 3

INFO Reading configuration /config/esphome/esp8266-seplos.yaml...
INFO Updating https://github.com/syssi/esphome-seplos-bms.git@debug-decoder
INFO Generating C++ source...
INFO Compiling app...
Processing esp8266-seplos (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/api/api_connection.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/api/api_server.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/api/list_entities.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/api/proto.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/api/user_services.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/esp8266/core.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/esp8266/gpio.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/esp8266/preferences.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/logger/logger.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/md5/md5.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/network/util.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/sensor/automation.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/sensor/filter.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/sensor/sensor.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/seplos_bms/seplos_bms.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/seplos_modbus/seplos_modbus.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/socket/socket.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/text_sensor/filter.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/text_sensor/text_sensor.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/uart/uart.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/uart/uart_component.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/uart/uart_component_esp32_arduino.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/uart/uart_component_esp8266.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/uart/uart_component_esp_idf.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/uart/uart_component_rp2040.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/uart/uart_debugger.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/wifi/wifi_component.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/wifi/wifi_component_esp32_arduino.cpp.o
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
[... cut ...]
[13:01:00][D][uart_debug:158]: <<< "~2001460010960001100D610D640D6D0D670D610D6E0D6C0D670D630D660D680D6E0D6A0D6E0D6D0D64060B720B710B760B740B840B6E0B691574685F0A6D6003BA6D60001603E8157A002"
[13:01:00][D][uart_debug:158]: <<< "200000BC40B40DD11\r"
[13:01:00][I][seplos_bms:030]: Telemetry frame (81 bytes) received
[13:01:00][W][seplos_modbus:019]: Buffer cleared
[13:01:00][W][seplos_modbus:019]: Buffer cleared
[13:01:00][W][seplos_modbus:019]: Buffer cleared
[13:01:00][W][seplos_modbus:019]: Buffer cleared
[13:01:00][W][seplos_modbus:019]: Buffer cleared
[13:01:01][W][seplos_modbus:019]: Buffer cleared
[13:01:01][D][uart_debug:158]: <<< "~20014642E00201FD35\r"

syssi, 2023-04-06:
Please decrease the `rx_timeout` to `50ms`

```
seplos_modbus:
  id: modbus0
  uart_id: uart0
  rx_timeout: 50ms
```

and provide another log. Does it reduce the number of "Invalid header" warnings?

juerse, 2023-04-06:
this is the last log with the following configuration and connected solar assistant and esp on the parallel port. If i remove solar assistant cable i have't this log

substitutions:
  name: esp8266-seplos
  device_description: "Monitor a Seplos BMS via RS485"
  external_components_source: github://syssi/esphome-seplos-bms@debug-decoder
  tx_pin: GPIO5
  rx_pin: GPIO4

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-seplos-bms"
    version: 1.1.0

esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  networks:
  - ssid: xxxxxxxxxxxxxxx
    password: xxxxxxxxxxxxxxxxxxxxxxxx
  - ssid: xxxxxxxxxxxxxxxxxxxxxxxxx
    password: xxxxxxxxxxxxxxxxxxxxxxxxx
api:
ota:
logger:

#web_server:
#  port: 80
uart:
  id: uart0
  # Please set the default baudrate of your Seplos BMS model here. It's sometimes 19200 baud instead of 9600.
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  # The increased RX buffer size is important because
  # the full BMS response must fit into the buffer
  rx_buffer_size: 384
  debug:
    dummy_receiver: false
    direction: BOTH
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
      
seplos_modbus:
  id: modbus0
  uart_id: uart0
  rx_timeout: 50ms

seplos_bms:
  id: bms0
  # Dip switch configuration of a single pack setup / address 0x0
  #  8    7    6    5    4    3    2    1
  # off, off, off, off, off, off, off, off
  address: 0x01
  # Known protocol versions: 0x20 (Seplos), 0x26 (Boqiang)
  protocol_version: 0x20
  seplos_modbus_id: modbus0
  update_interval: 10s

INFO Reading configuration /config/esphome/esp8266-seplos.yaml...
INFO Updating https://github.com/syssi/esphome-seplos-bms.git@debug-decoder
[... 1657 lines cut ...]
[21:39:39][W][seplos_modbus:031]: Buffer cleared due to reset: 30.30
[21:39:39][W][seplos_modbus:090]: Invalid header: 0x30
[21:39:39][W][seplos_modbus:031]: Buffer cleared due to reset: 30.31
[21:39:39][W][seplos_modbus:090]: Invalid header: 0x36
[21:39:39][W][seplos_modbus:031]: Buffer cleared due to reset: 36.30
[21:39:39][W][seplos_modbus:090]: Invalid header: 0x33
[21:39:39][W][seplos_modbus:031]: Buffer cleared due to reset: 33.45
[21:39:39][W][seplos_modbus:090]: Invalid header: 0x38
[21:39:39][W][seplos_modbus:031]: Buffer cleared due to reset: 38.31
[21:39:39][W][seplos_modbus:090]: Invalid header: 0x34
[21:39:39][W][seplos_modbus:031]: Buffer cleared due to reset: 34.41
[21:39:39][W][seplos_modbus:090]: Invalid header: 0x45
[21:39:39][W][seplos_modbus:031]: Buffer cleared due to reset: 45.30
[21:39:39][D][uart_debug:158]: <<< "~2001460010960001100CF10CEE0CE80CED0CEF0CEA0CEB0CEE0CEE0CEB0CED0CE80CEE0CEB0CEC0CF0060B7D0B7D0B7F0B7E0B9B0B74FC3114AE5DCF0A6D6003596D60001603E814AE002"
[21:39:39][W][seplos_modbus:090]: Invalid header: 0x30
[21:39:39][W][seplos_modbus:031]: Buffer cleared due to reset: 30.32
[21:39:39][W][seplos_modbus:090]: Invalid header: 0x37
[21:39:39][W][seplos_modbus:031]: Buffer cleared due to reset: 37.30
[21:39:39][W][seplos_modbus:090]: Invalid header: 0x30
[21:39:39][W][seplos_modbus:031]: Buffer cleared due to reset: 30.30
[21:39:39][W][seplos_modbus:090]: Invalid header: 0x30
[21:39:39][W][seplos_modbus:031]: Buffer cleared due to reset: 30.30
[21:39:39][W][seplos_modbus:090]: Invalid header: 0x42
[21:39:39][W][seplos_modbus:031]: Buffer cleared due to reset: 42.43
[21:39:39][W][seplos_modbus:090]: Invalid header: 0x43

syssi, 2023-04-06:
Thanks for the logs! They will help to get an idea what's going on but we need some more tests unfortunately. Please increase the `rx_timeout` to `250ms` now and provide another log. Thanks in advance!

juerse, 2023-04-06:
INFO Reading configuration /config/esphome/esp8266-seplos.yaml...
INFO Updating https://github.com/syssi/esphome-seplos-bms.git@debug-decoder
INFO Generating C++ source...
INFO Compiling app...
Processing esp8266-seplos (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Compiling /data/esp8266-seplos/.pioenvs/esp8266-seplos/src/main.cpp.o
Linking /data/esp8266-seplos/.pioenvs/esp8266-seplos/firmware.elf
RAM:   [====      ]  38.1% (used 31212 bytes from 81920 bytes)
Flash: [====      ]  36.6% (used 382609 bytes from 1044464 bytes)
Building /data/esp8266-seplos/.pioenvs/esp8266-seplos/firmware.bin
esp8266_copy_factory_bin(["/data/esp8266-seplos/.pioenvs/esp8266-seplos/firmware.bin"], ["/data/esp8266-seplos/.pioenvs/esp8266-seplos/firmware.elf"])
========================= [SUCCESS] Took 10.37 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of esp8266-seplos.local
INFO  -> 192.168.178.132
INFO Uploading /data/esp8266-seplos/.pioenvs/esp8266-seplos/firmware.bin (386768 bytes)
INFO Compressed to 269044 bytes
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from esp8266-seplos.local using esphome API
WARNING Can't connect to ESPHome API for esp8266-seplos.local: Error resolving IP address: [Errno -5] No address associated with hostname
INFO Trying to reconnect to esp8266-seplos.local in the background
INFO Successfully connected to esp8266-seplos.local
[22:28:30][I][app:102]: ESPHome version 2023.3.2 compiled on Apr  6 2023, 22:28:01
[22:28:30][I][app:104]: Project syssi.esphome-seplos-bms version 1.1.0
[22:28:30][C][wifi:504]: WiFi:
[22:28:30][C][wifi:362]:   Local MAC: xxxxxxxxxxxxxxxxx
[22:28:30][C][wifi:363]:   SSID: 'xxxxxxxxxxxxxxxxx'[redacted]
[22:28:30][C][wifi:364]:   IP Address: 192.168.178.132
[22:28:30][C][wifi:365]:   BSSID: xxxxxxxxxxxxxxxxxxx[redacted]
[22:28:30][C][wifi:367]:   Hostname: 'esp8266-seplos'
[22:28:30][C][wifi:369]:   Signal strength: -75 dB ▂▄▆█
[22:28:30][C][wifi:373]:   Channel: 1
[22:28:30][C][wifi:374]:   Subnet: 255.255.255.0
[22:28:30][C][wifi:375]:   Gateway: 192.168.178.1
[22:28:30][C][wifi:376]:   DNS1: 8.8.8.8
[22:28:30][C][wifi:377]:   DNS2: 8.8.4.4
[22:28:30][C][logger:293]: Logger:
[22:28:30][C][logger:294]:   Level: DEBUG
[22:28:30][C][logger:295]:   Log Baud Rate: 115200
[22:28:30][C][logger:296]:   Hardware UART: UART0
[22:28:30][C][uart.arduino_esp8266:102]: UART Bus:
[22:28:30][C][uart.arduino_esp8266:103]:   TX Pin: GPIO5
[22:28:30][C][uart.arduino_esp8266:104]:   RX Pin: GPIO4
[22:28:30][C][uart.arduino_esp8266:106]:   RX Buffer Size: 384
[22:28:30][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[22:28:30][C][uart.arduino_esp8266:109]:   Data Bits: 8
[22:28:30][C][uart.arduino_esp8266:110]:   Parity: NONE
[22:28:30][C][uart.arduino_esp8266:111]:   Stop bits: 1
[22:28:30][C][uart.arduino_esp8266:115]:   Using software serial
[22:28:30][W][seplos_modbus:019]: Buffer cleared due to timeout: 
[22:28:30][C][seplos_modbus:133]: SeplosModbus:
[22:28:30][C][seplos_modbus:135]:  RX timeout: 250 ms
[22:28:30][C][seplos_bms:156]: SeplosBms:
[22:28:30][C][mdns:108]: mDNS:
[22:28:30][C][mdns:109]:   Hostname: esp8266-seplos
[22:28:30][C][ota:093]: Over-The-Air Updates:
[22:28:30][C][ota:094]:   Address: esp8266-seplos.local:8266
[22:28:30][C][api:138]: API Server:
[22:28:30][C][api:139]:   Address: esp8266-seplos.local:6053
[22:28:30][C][api:143]:   Using noise encryption: NO
[... 1624 lines cut ...]
[22:31:01][D][uart_debug:158]: <<< "~20014642E00201FD35\r"
[22:31:01][W][seplos_bms:021]: Unhandled data received (data_len: 0x02): 20.01.46.42.E0.02.01 (7)
[22:31:01][W][seplos_modbus:031]: Buffer cleared due to reset: 7E.32.30.30.31.34.36.34.32.45.30.30.32.30.31.46.44.33.35.0D (20)
[22:31:01][D][uart_debug:158]: <<< "~2001460010960001100CF10CEC0CE30CEC0CEE0CE70CE80CED0CED0CEA0CEC0CE70CED0CE80CEC0CEF060B7D0B7D0B7F0B7E0B950B6DFC0414AB5A610A6D60033A6D60001603E814AC002"
[22:31:01][D][uart_debug:158]: <<< "D00000BCC0B5ADB71\r"
[22:31:01][I][seplos_bms:030]: Telemetry frame (81 bytes) received
[22:31:01][W][seplos_modbus:031]: Buffer cleared due to reset: 7E.32.30.30.31.34.36.30.30.31.30.39.36.30.30.30.31.31.30.30.43.46.31.30.43.45.43.30.43.45.33.30.43.45.43.30.43.45.45.30.43.45.37.30.43.45.38.30.43.45.44.30.43.45.44.30.43.45.41.30.43.45.43.30.43.45.37.30.43.45.44.30.43.45.38.30.43.45.43.30.43.45.46.30.36.30.42.37.44.30.42.37.44.30.42.37.46.30.42.37.45.30.42.39.35.30.42.36.44.46.43.30.34.31.34.41.42.35.41.36.31.30.41.36.44.36.30.30.33.33.41.36.44.36.30.30.30.31.36.30
[22:31:01][W][seplos_modbus:019]: Buffer cleared due to timeout: 
[22:31:02][W][seplos_modbus:019]: Buffer cleared due to timeout: 
[22:31:02][W][seplos_modbus:019]: Buffer cleared due to timeout: 
[22:31:02][W][seplos_modbus:019]: Buffer cleared due to timeout: 
[22:31:02][D][uart_debug:158]: <<< "~20014642E00201FD35\r"
[22:31:02][W][seplos_bms:021]: Unhandled data received (data_len: 0x02): 20.01.46.42.E0.02.01 (7)
[... cut ...]
[22:31:02][W][seplos_modbus:019]: Buffer cleared due to timeout: 
[22:31:03][W][seplos_modbus:019]: Buffer cleared due to timeout: 
[22:31:03][W][seplos_modbus:019]: Buffer cleared due to timeout: 
[22:31:03][W][seplos_modbus:019]: Buffer cleared due to timeout: 
[22:31:04][D][uart_debug:158]: <<< "~20014642E00201FD35\r"
[22:31:04][W][seplos_bms:021]: Unhandled data received (data_len: 0x02): 20.01.46.42.E0.02.01 (7)
[22:31:04][W][seplos_modbus:031]: Buffer cleared due to reset: 7E.32.30.30.31.34.36.34.32.45.30.30.32.30.31.46.44.33.35.0D (20)

syssi, 2023-04-06:
Perfect! It looks like we have reduced the "Invalid header" issues by increasing the `rx_timeout`. To could go back to the `main` branch again:

```
substitutions:
  external_components_source: github://syssi/esphome-seplos-bms@main
```

Please unplug the Solar Assistant connection. Does the BMS still doesn't respond if the ESP requests the telemetry frame? Please keep in mind the update interval is set to `10s` at the moment. The update interval of Solar Assistant is `1s` ATM.

juerse, 2023-04-07:
INFO Reading configuration /config/esphome/esp8266-seplos.yaml...
INFO Updating https://github.com/syssi/esphome-seplos-bms.git@main
INFO Starting log output from esp8266-seplos.local using esphome API
INFO Successfully connected to esp8266-seplos.local
[13:00:28][I][app:102]: ESPHome version 2023.3.2 compiled on Apr  7 2023, 12:55:25
[13:00:28][I][app:104]: Project syssi.esphome-seplos-bms version 1.1.0
[13:00:28][C][wifi:504]: WiFi:
[13:00:28][C][wifi:362]:   Local MAC: xxxxxxxxxxx
[13:00:28][C][wifi:363]:   SSID: 'xxxxxxxxxxxxxxxx'[redacted]
[13:00:28][C][wifi:364]:   IP Address: 192.168.178.132
[13:00:28][C][wifi:365]:   BSSID:xxxxxxxxxxxxxxxx[redacted]
[13:00:28][C][wifi:367]:   Hostname: 'esp8266-seplos'
[13:00:28][C][wifi:369]:   Signal strength: -72 dB ▂▄▆█
[13:00:28][C][wifi:373]:   Channel: 1
[13:00:28][C][wifi:374]:   Subnet: 255.255.255.0
[13:00:28][C][wifi:375]:   Gateway: 192.168.178.1
[13:00:28][C][wifi:376]:   DNS1: 8.8.8.8
[13:00:28][C][wifi:377]:   DNS2: 8.8.4.4
[13:00:28][C][logger:293]: Logger:
[13:00:28][C][logger:294]:   Level: DEBUG
[13:00:28][C][logger:295]:   Log Baud Rate: 115200
[13:00:28][C][logger:296]:   Hardware UART: UART0
[13:00:28][C][uart.arduino_esp8266:102]: UART Bus:
[13:00:28][C][uart.arduino_esp8266:103]:   TX Pin: GPIO5
[13:00:28][C][uart.arduino_esp8266:104]:   RX Pin: GPIO4
[13:00:28][C][uart.arduino_esp8266:106]:   RX Buffer Size: 384
[13:00:28][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[13:00:28][C][uart.arduino_esp8266:109]:   Data Bits: 8
[13:00:28][C][uart.arduino_esp8266:110]:   Parity: NONE
[13:00:28][C][uart.arduino_esp8266:111]:   Stop bits: 1
[13:00:28][C][uart.arduino_esp8266:115]:   Using software serial
[13:00:28][C][seplos_modbus:129]: SeplosModbus:
[13:00:28][C][seplos_modbus:131]:   RX timeout: 250 ms
[13:00:28][C][seplos_bms:156]: SeplosBms:
[13:00:28][C][mdns:108]: mDNS:
[13:00:28][C][mdns:109]:   Hostname: esp8266-seplos
[13:00:28][C][ota:093]: Over-The-Air Updates:
[13:00:28][C][ota:094]:   Address: esp8266-seplos.local:8266
[13:00:28][C][api:138]: API Server:
[13:00:28][C][api:139]:   Address: esp8266-seplos.local:6053
[13:00:28][C][api:143]:   Using noise encryption: NO

[13:00:34][D][uart_debug:158]: >>> "~20014642E00201FD35\r"

[13:00:44][D][uart_debug:158]: >>> "~20014642E00201FD35\r"

[13:00:55][D][uart_debug:158]: >>> "~20014642E00201FD35\r"

[13:01:04][D][uart_debug:158]: >>> "~20014642E00201FD35\r"

[13:01:14][D][uart_debug:158]: >>> "~20014642E00201FD35\r"

[13:01:24][D][uart_debug:158]: >>> "~20014642E00201FD35\r"

[13:01:34][D][uart_debug:158]: >>> "~20014642E00201FD35\r"

[13:01:44][D][uart_debug:158]: >>> "~20014642E00201FD35\r"

[13:01:55][D][uart_debug:158]: >>> "~20014642E00201FD35\r"

[13:02:04][D][uart_debug:158]: >>> "~20014642E00201FD35\r"

[13:02:14][D][uart_debug:158]: >>> "~20014642E00201FD35\r"
[13:02:15][I][ota:113]: Boot seems successful, resetting boot loop counter.

[13:02:24][D][uart_debug:158]: >>> "~20014642E00201FD35\r"

[13:02:34][D][uart_debug:158]: >>> "~20014642E00201FD35\r"

syssi, 2023-04-07:
Please check the TX line between the BMS and the ESP. The log indicates exactly the same request payload as the Solar Assistant request. If the root cause isn't the connection between the ESP and the BMS you should replace the RS485 converter. Does the TX led of the inverter blink periodically (once every 10 seconds)?

juerse, 2023-04-07:
the tx led flashes every 10s while the rx led does not flash if I do not
connect the solar assistant

syssi, 2023-04-07:
The flashing other TX led indicates we are using the correct GPIO of the ESP and the component transmits the periodic requests properly. Please check the connection between the RS485 converter and the BMS. I'm aware of a lot of faulty RS485 converters. Please replace the module just for testing.

juerse, 2023-04-07:
Ok i have this
ANGEEK MAX485 - Modulo convertitore da RS-485 TTL a RS485 per kit fai da te
Arduino (5 pezzi) https://amzn.eu/d/i8GZkLc

what is the right way to connect it?

syssi, 2023-04-07:
You don't have a second module of the current board?

juerse, 2023-04-07:
yes, but i doubt it will work...

syssi, 2023-04-07:
It will! ;-) I recommend the RS485 modules with automatic flow control over the one with the flow control pin.

juerse, 2023-04-11:
yes it was just that damned rs485 adapter. thank you!!!!!!!
Is it also possible to change the seplos settings directly from the esp?

syssi, 2023-04-11:
There are no controls implemented yet because it lacks on testers. If you are able to talk to your BMS using the windows application you could provide some traffic captures which are required to add some controls.

This is the corresponding issue: https://github.com/syssi/esphome-seplos-bms/issues/8

juerse, 2023-04-11:
if i can help i will but i don't understand what i have to do

syssi, 2023-04-16:
Please try `rx_timeout: 250ms` to avoid the `Invalid header` errors.

juerse, 2023-04-16:
it does not change

syssi, 2023-04-17:
Could you change this line:

```
external_components_source: github://syssi/esphome-seplos-bms@main
```

to

```
external_components_source: github://syssi/esphome-seplos-bms@debug-decoder
```

and flash your ESP again? Please provide another log. May be we can improve the communication a bit.

juerse, 2023-04-17:
INFO Reading configuration /config/esphome/esp8266-seplos1.yaml...
INFO Updating https://github.com/syssi/esphome-seplos-bms.git@debug-decoder
INFO Generating C++ source...
INFO Compiling app...
Processing esp8266-seplos1 (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Compiling /data/esp8266-seplos1/.pioenvs/esp8266-seplos1/src/esphome/components/seplos_modbus/seplos_modbus.cpp.o
Compiling /data/esp8266-seplos1/.pioenvs/esp8266-seplos1/src/main.cpp.o
Linking /data/esp8266-seplos1/.pioenvs/esp8266-seplos1/firmware.elf
RAM:   [====      ]  39.9% (used 32672 bytes from 81920 bytes)
Flash: [====      ]  37.4% (used 390933 bytes from 1044464 bytes)
Building /data/esp8266-seplos1/.pioenvs/esp8266-seplos1/firmware.bin
esp8266_copy_factory_bin(["/data/esp8266-seplos1/.pioenvs/esp8266-seplos1/firmware.bin"], ["/data/esp8266-seplos1/.pioenvs/esp8266-seplos1/firmware.elf"])
========================= [SUCCESS] Took 12.63 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of esp8266-seplos1.local
INFO  -> 192.168.178.161
INFO Uploading /data/esp8266-seplos1/.pioenvs/esp8266-seplos1/firmware.bin (395088 bytes)
INFO Compressed to 273046 bytes
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from esp8266-seplos1.local using esphome API
WARNING Can't connect to ESPHome API for esp8266-seplos1.local: Error resolving IP address: [Errno -5] No address associated with hostname
INFO Trying to reconnect to esp8266-seplos1.local in the background
INFO Successfully connected to esp8266-seplos1.local
[19:27:03][I][app:102]: ESPHome version 2023.3.2 compiled on Apr 17 2023, 19:26:30
[19:27:03][I][app:104]: Project syssi.esphome-seplos-bms version 1.1.0
[19:27:03][C][wifi:504]: WiFi:
[19:27:03][C][wifi:362]:   Local MAC: xxxxxxxxxxxx
[19:27:03][C][wifi:363]:   SSID: xxxxxxxxxxxxx[redacted]
[19:27:03][C][wifi:364]:   IP Address: 192.168.178.161
[19:27:03][C][wifi:365]:   BSSID: xxxxxxxxxxxxxxxxxxxxx[redacted]
[19:27:03][C][wifi:367]:   Hostname: 'esp8266-seplos1'
[19:27:03][C][wifi:369]:   Signal strength: -76 dB ▂▄▆█
[19:27:03][C][wifi:373]:   Channel: 6
[19:27:03][C][wifi:374]:   Subnet: 255.255.255.0
[19:27:03][C][wifi:375]:   Gateway: 192.168.178.1
[19:27:03][C][wifi:376]:   DNS1: 8.8.8.8
[19:27:03][C][wifi:377]:   DNS2: 8.8.4.4
[19:27:03][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:03][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:03][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:03][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:03][C][logger:293]: Logger:
[19:27:03][C][logger:294]:   Level: DEBUG
[19:27:03][C][logger:295]:   Log Baud Rate: 115200
[19:27:03][C][logger:296]:   Hardware UART: UART0
[19:27:03][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:03][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:03][C][uart.arduino_esp8266:102]: UART Bus:
[19:27:03][C][uart.arduino_esp8266:103]:   TX Pin: GPIO5
[19:27:03][C][uart.arduino_esp8266:104]:   RX Pin: GPIO4
[19:27:03][C][uart.arduino_esp8266:106]:   RX Buffer Size: 384
[19:27:03][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[19:27:03][C][uart.arduino_esp8266:109]:   Data Bits: 8
[19:27:03][C][uart.arduino_esp8266:110]:   Parity: NONE
[19:27:03][C][uart.arduino_esp8266:111]:   Stop bits: 1
[19:27:03][C][uart.arduino_esp8266:115]:   Using software serial
[19:27:03][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:03][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:03][C][seplos_modbus:133]: SeplosModbus:
[19:27:03][C][seplos_modbus:135]:  RX timeout: 250 ms
[... 1305 lines cut ...]
[19:27:26][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:26][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:26][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:26][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:26][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:26][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:26][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:26][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:26][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:26][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:26][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:27][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:27][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:27][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:27][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:27][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:27][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:27][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:27][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:27][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:27][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:27][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:27][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:27][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:27][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF

syssi, 2023-04-17:
How do you power your ESP?

juerse, 2023-04-17:
[image]
[image]

juerse, 2023-04-19:
Can you help me?

syssi, 2023-04-19:
I will try to provide some ideas/thoughts:

```
[19:27:15][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:15][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:15][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:15][W][seplos_modbus:090]: Invalid header: 0xFF
[19:27:15][W][seplos_modbus:031]: Buffer cleared due to reset: FF.FF
[19:27:15][W][seplos_modbus:090]: Invalid header: 0xFF
```

These bytes (`0xff`) aren't valid UART-TTL traffic. IMO it's noise. This can happen if the RS485 module is faulty, the power supply noisy or because of bad grounding (`GND` connections).

Could you provide the back of your PCB? I would like to see the connections between the ESP and the RS485 module + the connection between the RS485 module and your BMS (do you use 2 or 3 wires here)?

juerse, 2023-05-07:
I'm desperate I tried to change more than 10 rs485 converters and many of these become incandescent when I connect them, others simply don't work. I checked the connections several times. Could I use these converters? https://www.amazon.it/gp/product/B07X541M2T/ How should I connect them?

juerse, 2023-05-10:
can you help me?

syssi, 2023-05-10:
Yes. You can use this converter type too but a flow control GPIO is required.

juerse, 2023-05-10:
Can you help me to connect?

syssi, 2023-05-10:
Please use this schematics: https://github.com/syssi/esphome-solax-x1-mini#rs485-ttl-module-with-flow-control-pin

syssi, 2023-05-12:
```
uart:
  id: uart_0
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}

seplos_modbus:
  - id: modbus0
    uart_id: uart_0
    flow_control_pin: GPIO0
```