syssi/esphome-seplos-bms #59: Question about fork to different bms

JacquesdeBruyn, 2023-07-10:
Hi, I'm curious if this is possible. I've gotten a battery with a PACE bms. From what I can tell in my limited knowledge on the subject the protocols appear to be fairly similar. This bms seems to be used in Hubble, Greenrich, SOK, Revov and the KATBATT that I now own. These use rs232 on a RJ12 connector. I'll attach the protocol sheet and a picture of bms front.
[72651.11625-PACE-RS232commuciation-protocal.PACE-RS232-TY16S.-20180705.pdf](https://github.com/syssi/esphome-seplos-bms/files/12000343/72651.11625-PACE-RS232commuciation-protocal.PACE-RS232-TY16S.-20180705.pdf)
[image]

syssi, 2023-07-13:
Did you try to use this implementation with your BMS already? The protocol looks pretty similar to the seplos. Let's give it a try! May be just some mine quirks are required.

JacquesdeBruyn, 2023-07-13:
I haven't, I understand that this bms uses the rs232 port which I still need to get a connector for. I'll put an update up when I have that sorted and tested. I have however tested this exact code on the rs485 with no response. That was just a simple plug in. I haven't been able to find dedicated pc software for this bms yet either.

syssi, 2023-07-13:
Okay, let's try to talk to your BMS step by step. As first step we need 3 items:

- ESP32 or ESP8266
- RS232 converter module (equipped with a MAX3232)
- RJ12 jack

Please ping me if you are ready. I will provide a YAML with a simple `uart` component in debug mode (printing all traffic to the log) which sends a request frame from the documentation periodically.

JacquesdeBruyn, 2023-07-13:
Ok, you'll have to give me a chance to source all of those. I have esp32s, but only rs485 modules. Place I usually order from only has a pcb with a bare max3232 for basic ttl to rs232. And then I need to find a couple rj12s. Was hoping this bms will work with 485

syssi, 2023-07-13:
We could also start with the RS485 route if you like.

syssi, 2023-07-13:
Please use this example configuration if you want to talk to your BMS using RS485 + Modbus: https://github.com/syssi/esphome-seplos-bms/blob/pace-bms/tests/esp8266-pace-rs485-modbus-test.yaml

You have to make sure to select `slave address 1` using the DIP switches.

If you want to talk to your BMS using RS232 please use this YAML:

https://github.com/syssi/esphome-seplos-bms/blob/pace-bms/tests/esp8266-pace-rs232-test.yaml

In both YAMLs the RX/TX pins of the `uart` are attached to GPIO5/GPIO4.

JacquesdeBruyn, 2023-07-13:
I'll test that when I get back home. Just got a couple Callouts to deal with today

JacquesdeBruyn, 2023-07-13:
Test done, got an response from 485 with bms set to master -DIP 1 on rest off.

JacquesdeBruyn, 2023-07-13:
Code as follows:

substitutions:
  name: pace-bms-tests
  tx_pin: GPIO17
  rx_pin: GPIO16

esphome:
  name: esphome-web-9d9124
  friendly_name: Test

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable Home Assistant API
api:
  reboot_timeout: 0s
  encryption:
    key: "VWCTZnKZwlLn5Z5vNwdvPk0+51EFx2radcwh9g0BbJY="

ota:

logger:
  level: DEBUG

uart:
  id: uart_0
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: false

modbus:
  - id: modbus0
    flow_control_pin: 18
    uart_id: uart_0
    send_wait_time: 200ms

modbus_controller:
  - id: bms0
    # Slave address 0x01
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s

sensor:
  # 0x0001 Voltage of pack   2 byte   R/UINT16  10mV
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} total voltage"
    address: 0x0001
    register_type: holding
    value_type: U_WORD
    accuracy_decimals: 0

  # 0x0002 State of charge    2 byte   R/UINT8   % (0-100%)
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} state of charge"
    address: 0x0002
    register_type: holding
    value_type: U_WORD
    accuracy_decimals: 0

JacquesdeBruyn, 2023-07-13:
Log response:

INFO ESPHome 2023.6.5
INFO Reading configuration /config/esphome/esphome-web-9d9124.yaml...
INFO Starting log output from esphome-web-9d9124.local using esphome API
INFO Successfully connected to esphome-web-9d9124.local
[19:21:04][I][app:102]: ESPHome version 2023.6.5 compiled on Jul 13 2023, 19:07:42
[19:21:04][C][wifi:543]: WiFi:
[19:21:04][C][wifi:379]:   Local MAC: EC:62:60:9D:91:24
[19:21:04][C][wifi:380]:   SSID: 'WatchingYouXD'[redacted]
[19:21:04][C][wifi:381]:   IP Address: 192.168.0.47
[19:21:04][C][wifi:383]:   BSSID: 78:44:76:D6:84:00[redacted]
[19:21:04][C][wifi:384]:   Hostname: 'esphome-web-9d9124'
[19:21:04][C][wifi:386]:   Signal strength: -48 dB ▂▄▆█
[19:21:04][C][wifi:390]:   Channel: 13
[19:21:04][C][wifi:391]:   Subnet: 255.255.255.0
[19:21:04][C][wifi:392]:   Gateway: 192.168.0.1
[19:21:04][C][wifi:393]:   DNS1: 192.168.0.1
[19:21:04][C][wifi:394]:   DNS2: 0.0.0.0
[19:21:04][C][logger:301]: Logger:
[19:21:04][C][logger:302]:   Level: DEBUG
[19:21:04][C][logger:303]:   Log Baud Rate: 115200
[19:21:04][C][logger:305]:   Hardware UART: UART0
[19:21:04][C][uart.arduino_esp32:124]: UART Bus 1:
[19:21:04][C][uart.arduino_esp32:125]:   TX Pin: GPIO17
[19:21:04][C][uart.arduino_esp32:126]:   RX Pin: GPIO16
[19:21:04][C][uart.arduino_esp32:128]:   RX Buffer Size: 256
[19:21:04][C][uart.arduino_esp32:130]:   Baud Rate: 9600 baud
[19:21:04][C][uart.arduino_esp32:131]:   Data Bits: 8
[19:21:04][C][uart.arduino_esp32:132]:   Parity: NONE
[19:21:04][C][uart.arduino_esp32:133]:   Stop bits: 1
[19:21:04][C][modbus:143]: Modbus:
[19:21:04][C][modbus:144]:   Flow Control Pin: GPIO18
[19:21:04][C][modbus:145]:   Send Wait Time: 200 ms
[19:21:04][C][modbus:146]:   CRC Disabled: NO
[19:21:04][C][modbus_controller:275]: ModbusController:
[19:21:04][C][modbus_controller:276]:   Address: 0x01
[19:21:04][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'pace-bms-tests total voltage'
[19:21:04][C][modbus_controller.sensor:010]: modbus_controller.sensor  State Class: ''
[19:21:04][C][modbus_controller.sensor:010]: modbus_controller.sensor  Unit of Measurement: ''
[19:21:04][C][modbus_controller.sensor:010]: modbus_controller.sensor  Accuracy Decimals: 0
[19:21:04][D][uart_debug:114]: <<< 7E:32:35:30:41:34:36:39:30:30:30:30:30:46:44:39:35:0D:7E:32:35:30:42:34:36:39:30:30:30:30:30:46:44:39:34:0D
[19:21:04][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'pace-bms-tests state of charge'
[19:21:04][C][modbus_controller.sensor:010]: modbus_controller.sensor  State Class: ''
[19:21:04][C][modbus_controller.sensor:010]: modbus_controller.sensor  Unit of Measurement: ''
[19:21:04][C][modbus_controller.sensor:010]: modbus_controller.sensor  Accuracy Decimals: 0
[19:21:04][C][mdns:112]: mDNS:
[19:21:04][C][mdns:113]:   Hostname: esphome-web-9d9124
[19:21:04][C][ota:093]: Over-The-Air Updates:
[19:21:04][C][ota:094]:   Address: esphome-web-9d9124.local:3232
[19:21:04][C][api:138]: API Server:
[19:21:04][C][api:139]:   Address: esphome-web-9d9124.local:6053
[19:21:04][C][api:141]:   Using noise encryption: YES
[19:21:04][D][uart_debug:114]: <<< 7E:32:35:30:43:34:36:39:30:30:30:30:30:46:44:39:33:0D:7E:32:35:30:44:34:36:39:30:30:30:30:30:46:44:39:32:0D
[19:21:05][D][uart_debug:114]: <<< 7E:32:35:30:45:34:36:39:30:30:30:30:30:46:44:39:31:0D
[19:21:05][D][uart_debug:114]: <<< 7E:32:35:30:46:34:36:39:30:30:30:30:30:46:44:39:30:0D
[19:21:05][D][uart_debug:114]: <<< 7E:32:35:30:32:34:36:39:30:30:30:30:30:46:44:41:34:0D
[19:21:05][D][uart_debug:114]: <<< 7E:32:35:30:33:34:36:39:30:30:30:30:30:46:44:41:33:0D
[19:21:05][D][uart_debug:114]: <<< 7E:32:35:30:34:34:36:39:30:30:30:30:30:46:44:41:32:0D
[19:21:05][D][uart_debug:114]: <<< 7E:32:35:30:35:34:36:39:30:30:30:30:30:46:44:41:31:0D
[19:21:06][D][uart_debug:114]: <<< 7E:32:35:30:36:34:36:39:30:30:30:30:30:46:44:41:30:0D
[19:21:06][D][uart_debug:114]: <<< 7E:32:35:30:37:34:36:39:30:30:30:30:30:46:44:39:46:0D
[19:21:06][D][uart_debug:114]: <<< 7E:32:35:30:38:34:36:39:30:30:30:30:30:46:44:39:45:0D
[19:21:06][D][uart_debug:114]: <<< 7E:32:35:30:39:34:36:39:30:30:30:30:30:46:44:39:44:0D
[19:21:06][D][uart_debug:114]: <<< 7E:32:35:30:41:34:36:39:30:30:30:30:30:46:44:39:35:0D
[19:21:06][D][uart_debug:114]: <<< 7E:32:35:30:42:34:36:39:30:30:30:30:30:46:44:39:34:0D
[19:21:06][D][uart_debug:114]: <<< 7E:32:35:30:43:34:36:39:30:30:30:30:30:46:44:39:33:0D
[19:21:07][D][uart_debug:114]: <<< 7E:32:35:30:44:34:36:39:30:30:30:30:30:46:44:39:32:0D
[19:21:07][D][uart_debug:114]: <<< 7E:32:35:30:45:34:36:39:30:30:30:30:30:46:44:39:31:0D
[19:21:07][D][uart_debug:114]: >>> 01:03:00:01:00:02:95:CB
[... 41 lines cut ...]
[19:21:12][D][uart_debug:114]: <<< 7E:32:35:30:36:34:36:39:30:30:30:30:30:46:44:41:30:0D
[19:21:12][D][uart_debug:114]: <<< 7E:32:35:30:37:34:36:39:30:30:30:30:30:46:44:39:46:0D
[19:21:12][D][uart_debug:114]: <<< 7E:32:35:30:38:34:36:39:30:30:30:30:30:46:44:39:45:0D
[19:21:13][D][uart_debug:114]: <<< 7E:32:35:30:39:34:36:39:30:30:30:30:30:46:44:39:44:0D
[19:21:13][D][uart_debug:114]: <<< 7E:32:35:30:41:34:36:39:30:30:30:30:30:46:44:39:35:0D
[19:21:13][D][uart_debug:114]: <<< 7E:32:35:30:42:34:36:39:30:30:30:30:30:46:44:39:34:0D
[19:21:13][D][uart_debug:114]: <<< 7E:32:35:30:43:34:36:39:30:30:30:30:30:46:44:39:33:0D
[19:21:13][D][uart_debug:114]: <<< 7E:32:35:30:44:34:36:39:30:30:30:30:30:46:44:39:32:0D
[19:21:13][D][uart_debug:114]: <<< 7E:32:35:30:45:34:36:39:30:30:30:30:30:46:44:39:31:0D
[... cut ...]
[19:21:15][D][uart_debug:114]: <<< 7E:32:35:30:42:34:36:39:30:30:30:30:30:46:44:39:34:0D
[19:21:15][D][uart_debug:114]: <<< 7E:32:35:30:43:34:36:39:30:30:30:30:30:46:44:39:33:0D
[19:21:15][D][uart_debug:114]: <<< 7E:32:35:30:44:34:36:39:30:30:30:30:30:46:44:39:32:0D
[19:21:16][D][uart_debug:114]: <<< 7E:32:35:30:45:34:36:39:30:30:30:30:30:46:44:39:31:0D
[19:21:16][D][uart_debug:114]: <<< 7E:32:35:30:46:34:36:39:30:30:30:30:30:46:44:39:30:0D
[19:21:16][D][uart_debug:114]: <<< 7E:32:35:30:32:34:36:39:30:30:30:30:30:46:44:41:34:0D

JacquesdeBruyn, 2023-07-13:
I used UART2 on my esp32 wroom. Seem to recall uart one giving issues last time with the seplos bms

syssi, 2023-07-13:
It looks like your BMS doesn't speak Modbus on the RS485 bus. If you send a Modbus request the BMS did respond sometimes with an ASCII response of the RS232 protocol.

Please try to use the RS232 test yaml on the RS485 bus. Just attach the RS485 to the `uart` GPIOs and give it a try.

JacquesdeBruyn, 2023-07-13:
Sure, any idea where to implement flow control in that yaml?

JacquesdeBruyn, 2023-07-13:
I plugged into the inverter port and got the proper response there:

[22:07:31][D][uart_debug:114]: >>> 01:03:00:01:00:02:95:CB
[22:07:32][D][modbus_controller.sensor:025]: Sensor new state: 5236.00
[22:07:32][D][sensor:094]: 'pace-bms-tests total voltage': Sending state 5236.00000  with 0 decimals of accuracy
[22:07:32][D][modbus_controller.sensor:025]: Sensor new state: 67.00
[22:07:32][D][sensor:094]: 'pace-bms-tests state of charge': Sending state 67.00000  with 0 decimals of accuracy
[22:07:32][D][uart_debug:114]: <<< 01:03:04:14:74:00:43:FE:28
[22:07:41][D][uart_debug:114]: >>> 01:03:00:01:00:02:95:CB
[22:07:42][D][modbus_controller.sensor:025]: Sensor new state: 5238.00
[22:07:42][D][sensor:094]: 'pace-bms-tests total voltage': Sending state 5238.00000  with 0 decimals of accuracy
[22:07:42][D][modbus_controller.sensor:025]: Sensor new state: 67.00
[22:07:42][D][sensor:094]: 'pace-bms-tests state of charge': Sending state 67.00000  with 0 decimals of accuracy
[22:07:42][D][uart_debug:114]: <<< 01:03:04:14:76:00:43:5F:E8
[22:07:51][D][uart_debug:114]: >>> 01:03:00:01:00:02:95:CB
[22:07:52][D][modbus_controller.sensor:025]: Sensor new state: 5236.00
[22:07:52][D][sensor:094]: 'pace-bms-tests total voltage': Sending state 5236.00000  with 0 decimals of accuracy
[22:07:52][D][modbus_controller.sensor:025]: Sensor new state: 67.00
[22:07:52][D][sensor:094]: 'pace-bms-tests state of charge': Sending state 67.00000  with 0 decimals of accuracy
[22:07:52][D][uart_debug:114]: <<< 01:03:04:14:74:00:43:FE:28

JacquesdeBruyn, 2023-07-13:
That would be the first rs485 next to the CAN port. Seplos uses the parallel ports. But issue resolved. Pinout is also the same as the seplos BMS

syssi, 2023-07-13:
Good job! Let's add some more registers soon.

syssi, 2023-07-13:
On which port did you receive these responses?

```
[19:21:13][D][uart_debug:114]: <<< 7E:32:35:30:41:34:36:39:30:30:30:30:30:46:44:39:35:0D
[19:21:13][D][uart_debug:114]: <<< 7E:32:35:30:42:34:36:39:30:30:30:30:30:46:44:39:34:0D
```

JacquesdeBruyn, 2023-07-13:
I'd be Interested to see if I can run both off one rs485 later. Voltage seems to have to move the period two digits forward. SOC seems fine. not sure why it pushes 0000 at the end. Anyway I'm going to catch a shower and call it a night. It's past 10 and have to run between my flat and the main house for every change.

JacquesdeBruyn, 2023-07-13:
> On which port did you receive these responses?
> 
> ```
> [19:21:13][D][uart_debug:114]: <<< 7E:32:35:30:41:34:36:39:30:30:30:30:30:46:44:39:35:0D
> [19:21:13][D][uart_debug:114]: <<< 7E:32:35:30:42:34:36:39:30:30:30:30:30:46:44:39:34:0D
> ```

This was on the double rs485 ports. The ones used for parallel packs

syssi, 2023-07-13:
The decimal place can be adjusted easily. The zeros at the end can be ignored and are shown at the log only. The passed measurement via MQTT or native API will be `68` because of the `0 decimals of accuracy` setting.

JacquesdeBruyn, 2023-07-13:
For clarity. I'll check in again in the morning.

[image]

FCLFCL82, 2023-07-14:
I am following this, I have a  [Gobel Power](https://github.com/fancyui/Gobel-Power-RN-BMS-RS485-ModBus) BMS which seems to be similar to the PACE.  I got similar responses using your RS485 code but on different register addresses:

`[15:21:08][D][uart_debug:114]: <<< 01:03:04:00:63:14:C3:45:7C
[15:21:17][D][uart_debug:114]: >>> 01:03:00:15:00:02:D5:CF
[15:21:17][D][modbus_controller.sensor:025]: Sensor new state: 99.00
[15:21:17][D][sensor:093]: 'gobelpower-bms-tests state of charge': Sending state 99.00000  with 0 decimals of accuracy
[15:21:17][D][modbus_controller.sensor:025]: Sensor new state: 5315.00
[15:21:17][D][sensor:093]: 'gobelpower-bms-tests total voltage': Sending state 5315.00000  with 0 decimals of accuracy
[15:21:18][D][uart_debug:114]: <<< 01:03:04:00:63:14:C3:45:7C`

syssi, 2023-07-14:
I've pushed a small change to the YAML to fix the decimals and units: https://github.com/syssi/esphome-seplos-bms/pull/60/commits/d37557a7d53e9c07420c94d22680e8e9049bf2d4

Please update your YAML and try again.

JacquesdeBruyn, 2023-07-14:
Unrelated question, the flags I think it is that tell hass this is a power related device. How do those work? I just noticed that the seplos one does not get shown as an option in the power section.

syssi, 2023-07-14:
> Unrelated question, the flags I think it is that tell hass this is a power related device. How do those work? I just noticed that the seplos one does not get shown as an option in the power section.

I don't understand your question. What are you looking for?