syssi/esphome-seplos-bms #62: 1101-SP88 Unable to get any comms.

matchermike1981, 2023-08-05:
Hi.

I am struggling to get this running on my Seplos BMS.

I am using a Wemos D1 mini: https://diyi0t.com/esp8266-wemos-d1-mini-tutorial/?utm_content=cmp-true
And a Fasizi 5pcs TTL To RS485 Adapter: https://www.amazon.co.uk/Fasizi-Adapter-Serial-Converter-Module/dp/B09Z2GTMJ8

I have installed the esp8266-example.yaml, changed my BAUD to 19200 (as tested with my laptop and seplos management software on the RS485 ports.

I am fairly sure it is GPIO1 and GPIO3 from reading the Wemos D1 info: https://mischianti.org/2019/09/19/wemos-d1-mini-esp8266-debug-on-secondary-uart-part-3/

I have enabled debug as you can see in my code. All i am able to get is :

`[19:49:58][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[19:50:08][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[19:50:18][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[19:50:28][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[19:50:38][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[19:50:48][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[19:50:58][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[19:51:08][D][uart_debug:158]: >>> "~20004642E00200FD37\r"`

An i get a flashing RXD light on the RS485 board, but nothing on TXD.

My seplos BMS is a 1101-SP88 SW16.4

Appreciate any advice please. Have to admit i am a n00b, and don't have a linux box. I am doing everything through ESPHOME on my HA instance.

Here is my board:

[image]
[image]

Here is my code'

`substitutions:
  name: seplos-bms
  device_description: "Monitor a Seplos BMS via RS485"
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin: GPIO1
  rx_pin: GPIO3

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-seplos-bms"
    version: 1.1.0

esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  baud_rate: 0

api:

uart:
  id: uart_0
  # Please set the default baudrate of your Seplos BMS model here. It's sometimes 19200 baud instead of 9600.
  baud_rate: 19200
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  # The increased RX buffer size is important because
  # the full BMS response must fit into the buffer
  rx_buffer_size: 384
  debug:
    dummy_receiver: false
    direction: BOTH
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

seplos_modbus:
  flow_control_pin: 4
  id: modbus0
  uart_id: uart_0
  rx_timeout: 150ms

seplos_bms:
  id: bms0
  # Dip switch configuration of a single pack setup / address 0x0
  #  8    7    6    5    4    3    2    1
  # off, off, off, off, off, off, off, off
  address: 0x00
  # Known protocol versions: 0x20 (Seplos), 0x26 (Boqiang)
  protocol_version: 0x20
  seplos_modbus_id: modbus0
  update_interval: 10s

sensor:
  - platform: seplos_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    temperature_1:
      name: "${name} temperature 1"
    temperature_2:
      name: "${name} temperature 2"
    temperature_3:
      name: "${name} temperature 3"
    temperature_4:
      name: "${name} temperature 4"
    temperature_5:
      name: "${name} environment temperature"
    temperature_6:
      name: "${name} mosfet temperature"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    residual_capacity:
      name: "${name} residual capacity"
    battery_capacity:
      name: "${name} battery capacity"
    rated_capacity:
      name: "${name} rated capacity"
    state_of_charge:
      name: "${name} state of charge"
    charging_cycles:
      name: "${name} charging cycles"
    state_of_health:
      name: "${name} state of health"
    port_voltage:
      name: "${name} port voltage"`

matchermike1981, 2023-08-05:
I swapped my connections onto pins D1 & D1 using GPIO 5 and GPIO4

I now get the following in my debug:

```
[20:39:16][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[20:39:16][W][seplos_modbus:090]: Invalid header: 0x00
[20:39:16][D][uart_debug:158]: <<< "\x00"

[20:39:26][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[20:39:26][W][seplos_modbus:090]: Invalid header: 0x00
[20:39:26][D][uart_debug:158]: <<< "\x00"

[20:39:36][D][uart_debug:158]: >>> "~20004642E00200FD37\r"
[20:39:36][W][seplos_modbus:090]: Invalid header: 0x00
[20:39:36][D][uart_debug:158]: <<< "\x00"
```

syssi, 2023-08-06:
Your setup looks pretty good already. I haven't checked the wiring yet. I will try to find some time this evening.

> I have installed the esp8266-example.yaml, changed my BAUD to 19200 (as tested with my laptop and seplos management software on the RS485 ports.

So you are able to talk to the BMS using the management software in general? If I remember correctly the application has a debug panel. You should see the raw traffic (request and responses) here. Please provide some request frames. Does they look like the requests of this implementation / the log messages above?

You have solved you first issue by swapping the RX/TX line already. If you have a second ESP and RS485 converter you could try to listen to the requests of the current setup. Just to make sure your components are healthy. There are a lot of broken RS485 board around.

closed 2023-08-11 by syssi