syssi/esphome-seplos-bms #65: Bluetooth

juerse, 2023-08-20:
Hi
Is it possible to use bluetooth integration?

syssi, 2023-08-20:
Not yet.

juerse, 2023-08-21:
when do you think it will be possible?
Thank you

syssi, 2023-08-21:
I don't provide estimates because it makes people angry if I miss the deadline. ;-) The implementation will take time.

juerse, 2023-08-21:
Ok thank you

andreas-bulling, 2023-09-01:
That'd be fantastic, although the wired connection also works really fine.

Even better would be if it was possible to change settings through ESPHome without having to use the (desktop) app.

saviothecnic, 2023-09-20:
When it will also be available in beta
I'll be happy to try

Dukat-Gul, 2024-09-24:
Hi @syssi just a quick status check. Is the ble / add bluetooth support branch considered a work in progress or is it something you are interrsted having others in the community perform testing?

syssi, 2024-09-24:
This is correct. At the moment there is no testing needed yet because the implementation isn't ready.

syssi, 2025-07-15:
The implementation is ready and merged into `main`. Please give it a try! :-)

saviothecnic, 2025-07-15:
WOW 
Great news!
Question for the Seplos V 2.0 BMS
Which example should I use?
I have 4 BMSs. Do I need to use 4 ESP32s, or can I use just one for all 4?

syssi, 2025-07-15:
Please start with a single BMS. :-) If the ESP is powerful there should be multiple BLE connections possible.