syssi/esphome-seplos-bms #67: Unhandled data received

manalishi70, 2023-09-10:
I have an esp32 connected to the Master BMS of a set 4 batteries with Seplos 10E 200A with BT. The connectivity to the BMS is not very stable and I recieve errors every second in the log. The registers of the 3 slaves are not updated every interval. Sometimes it updates just the registers of one or two packs.

RS485 is powered by 5V. Same behavior with 3.3V. Do youhave any idea what causes this behavior?

`[21:29:38][W][seplos_bms:022]: Unhandled data received (data_len: 0x40): 20.01.46.5A.C0.40.00.01.0C.F1.0C.EF.0B.B9.0B.AD.FA.60.14.B3.58.11.6D.60.03.25.14.B3.00.00.00.00.00.00.03.02.80.41 (38)
[21:29:38][W][seplos_bms:022]: Unhandled data received (data_len: 0x00): 20.02.46.5A.00.00 (6)
[21:29:38][W][seplos_bms:022]: Unhandled data received (data_len: 0x00): 20.02.46.5A.00.00 (6)
[21:29:38][W][seplos_bms:022]: Unhandled data received (data_len: 0x40): 20.02.46.5A.C0.40.00.01.0C.F2.0C.EF.0B.B8.0B.AD.FA.9C.14.B4.59.AE.6D.60.03.33.14.B5.00.00.00.00.00.00.03.02.80.41 (38)
[21:29:38][W][seplos_bms:022]: Unhandled data received (data_len: 0x40): 20.02.46.5A.C0.40.00.01.0C.F2.0C.EF.0B.B8.0B.AD.FA.9C.14.B4.59.AE.6D.60.03.33.14.B5.00.00.00.00.00.00.03.02.80.41 (38)
[21:29:38][W][seplos_bms:022]: Unhandled data received (data_len: 0x00): 20.03.46.5A.00.00 (6)
[21:29:38][W][seplos_bms:022]: Unhandled data received (data_len: 0x00): 20.03.46.5A.00.00 (6)
[21:29:38][W][seplos_bms:022]: Unhandled data received (data_len: 0x40): 20.03.46.5A.C0.40.00.01.0C.F7.0C.E7.0B.AD.0B.A2.FD.5C.14.B4.64.4B.6D.60.03.94.14.B5.00.00.00.00.00.00.03.02.80.41 (38)
[21:29:38][W][seplos_bms:022]: Unhandled data received (data_len: 0x40): 20.03.46.5A.C0.40.00.01.0C.F7.0C.E7.0B.AD.0B.A2.FD.5C.14.B4.64.4B.6D.60.03.94.14.B5.00.00.00.00.00.00.03.02.80.41 (38)`

syssi, 2023-09-10:
Could you provide some more log messages?

syssi, 2023-09-10:
Could you provide your configuration YAML too?

manalishi70, 2023-09-14:
> Could you provide your configuration YAML too?

[seplos_esp32_no_master_yaml.txt](https://github.com/syssi/esphome-seplos-bms/files/12612508/seplos_esp32_no_master_yaml.txt)

manalishi70, 2023-09-14:
> Could you provide some more log messages?

`<html>
<body>

21:21:03 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.03.46.5A.C0.40.00.01.0C.F9.0C.ED.0B.A0.0B.94.FD.EE.14.B8.67.96.6D.60.03.B3.14.BA.00.00.00.00.00.00.03.02.80.41 (38)
21:21:04 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.01.46.5A.00.00 (6)
21:21:04 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.01.46.5A.C0.40.00.01.0C.F4.0C.F2.0B.9F.0B.93.FB.DE.14.B7.62.5E.6D.60.03.83.14.B8.00.00.00.00.00.00.03.02.80.41 (38)
21:21:04 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.02.46.5A.00.00 (6)
21:21:04 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.02.46.5A.C0.40.00.01.0C.F4.0C.F2.0B.9D.0B.91.FC.36.14.B8.62.EB.6D.60.03.88.14.B9.00.00.00.00.00.00.03.02.80.41 (38)
21:21:04 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.03.46.5A.00.00 (6)
21:21:04 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.03.46.5A.C0.40.00.01.0C.F9.0C.ED.0B.A0.0B.94.FD.EC.14.B8.67.95.6D.60.03.B3.14.BA.00.00.00.00.00.00.03.02.80.41 (38)
21:21:05 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.01.46.5A.00.00 (6)
21:21:05 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.01.46.5A.C0.40.00.01.0C.F4.0C.F2.0B.9F.0B.93.FB.DD.14.B7.62.5E.6D.60.03.83.14.B8.00.00.00.00.00.00.03.02.80.41 (38)
21:21:05 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.02.46.5A.00.00 (6)
21:21:05 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.02.46.5A.C0.40.00.01.0C.F4.0C.F2.0B.9D.0B.91.FC.37.14.B8.62.EA.6D.60.03.88.14.B9.00.00.00.00.00.00.03.02.80.41 (38)
21:21:05 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.03.46.5A.00.00 (6)
21:21:05 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.03.46.5A.C0.40.00.01.0C.F9.0C.ED.0B.A0.0B.94.FD.ED.14.B8.67.95.6D.60.03.B3.14.BA.00.00.00.00.00.00.03.02.80.41 (38)
21:21:06 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.01.46.5A.00.00 (6)
21:21:06 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.01.46.5A.C0.40.00.01.0C.F4.0C.F2.0B.9F.0B.93.FB.D3.14.B7.62.5E.6D.60.03.83.14.B8.00.00.00.00.00.00.03.02.80.41 (38)
21:21:06 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.02.46.5A.00.00 (6)
21:21:06 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.02.46.5A.C0.40.00.01.0C.F4.0C.F2.0B.9D.0B.91.FC.2D.14.B8.62.EA.6D.60.03.88.14.B9.00.00.00.00.00.00.03.02.80.41 (38)
21:21:06 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.03.46.5A.00.00 (6)
21:21:06 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.03.46.5A.C0.40.00.01.0C.F9.0C.ED.0B.A0.0B.94.FD.EB.14.B8.67.95.6D.60.03.B3.14.BA.00.00.00.00.00.00.03.02.80.41 (38)
21:21:07 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.01.46.5A.00.00 (6)
21:21:07 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.01.46.5A.C0.40.00.01.0C.F3.0C.F2.0B.9F.0B.93.FB.D0.14.B7.62.5D.6D.60.03.83.14.B8.00.00.00.00.00.00.03.02.80.41 (38)
21:21:07 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.02.46.5A.00.00 (6)
21:21:07 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.02.46.5A.C0.40.00.01.0C.F4.0C.F2.0B.9D.0B.91.FC.2D.14.B8.62.EA.6D.60.03.88.14.B9.00.00.00.00.00.00.03.02.80.41 (38)
21:21:07 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.03.46.5A.00.00 (6)
21:21:07 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.03.46.5A.C0.40.00.01.0C.F9.0C.ED.0B.A0.0B.94.FD.E9.14.B8.67.95.6D.60.03.B3.14.BA.00.00.00.00.00.00.03.02.80.41 (38)
21:21:08 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.01.46.5A.00.00 (6)
21:21:08 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.01.46.5A.C0.40.00.01.0C.F3.0C.F2.0B.9F.0B.93.FB.CD.14.B7.62.5D.6D.60.03.83.14.B8.00.00.00.00.00.00.03.02.80.41 (38)
21:21:08 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.02.46.5A.00.00 (6)
21:21:08 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.02.46.5A.C0.40.00.01.0C.F4.0C.F2.0B.9D.0B.91.FC.29.14.B8.62.EA.6D.60.03.88.14.B9.00.00.00.00.00.00.03.02.80.41 (38)
21:21:08 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.03.46.5A.00.00 (6)
21:21:08 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.03.46.5A.C0.40.00.01.0C.F9.0C.ED.0B.A0.0B.94.FD.EA.14.B8.67.95.6D.60.03.B3.14.BA.00.00.00.00.00.00.03.02.80.41 (38)
21:21:09 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.01.46.5A.00.00 (6)
21:21:09 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.01.46.5A.C0.40.00.01.0C.F3.0C.F2.0B.9F.0B.93.FB.D2.14.B7.62.5D.6D.60.03.83.14.B8.00.00.00.00.00.00.03.02.80.41 (38)
21:21:09 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.02.46.5A.00.00 (6)
21:21:09 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x40): 20.02.46.5A.C0.40.00.01.0C.F4.0C.F2.0B.9D.0B.91.FC.2E.14.B8.62.E9.6D.60.03.88.14.BA.00.00.00.00.00.00.03.02.80.41 (38)
21:21:09 | [W] | [seplos_bms:022] | Unhandled data received (data_len: 0x00): 20.03.46.5A.00.00 (6)
21:21:09 | [D] | [seplos_modbus:175] |  

<br class="Apple-interchange-newline">
</body>
</html>`

manalishi70, 2023-09-14:
> Could you provide some more log messages?

I use this setting:

logger:
  
  level: DEBUG

Can I create a more detailed logfile?

manalishi70, 2023-09-16:
Today I activated "Auto Address Coding" in the BMS Settings. The result gave me a big headache, but during debuging I recognized that "Unhandled data received" messages disappeared in the esp . After deactivation of this feature the messages came back. Maybe this is a hint to the root cause for this messages

Soul-man-dot, 2023-10-01:
any news on this topic- i do have a similar problem.