syssi/esphome-seplos-bms #71: Decoding version 2.5 for bms TDT-6022

wojtbar, 2023-11-02:
Hello and welcome.
I have a BMS as in the topic and it works with seplos ver2.0, but not everything is decoded correctly. Send data:
~25014642E00201FD30
answer:
~25014F4280800001 0E 0DD2 0DD2 0DD3 0DD3 0DCC 0DCC 0DCD 0DCD 0DCE 0DCF 0DCD 0DCE 0DCE 0DCE  06 0BE2 0BB1 0B9A 0B9A 0B9A 0BA2  0000  1354 395A  00 5B68  0000  5B68  133A 09BE 3F  0409  E092
bms is 14s. I have almost all the commands that are sent to the bms. If necessary, I can share the data. I have dump.pcap .
Please help.

syssi, 2023-11-04:
I'm happy about every detail. Could describe what's not working properly in detail?

wojtbar, 2023-11-04:
English is not my native language.
For 14 s it should be Data[77] and for me it is 71.
Therefore port voltage, state of health, charging cycles, state of charge is bad.

wojtbar, 2023-11-04:
Data from Data[0] to Data[61] for 16s match.
Data[63] 0x03 0x13 for seplos it's Data[71] 0x3c
Data[69] 0x 0x changes dynamically depending on the use of CHGon, CHGoff, DSGon,DSGoff.
Data[67] 0x13 0x06 for me port_voltage

wojtbar, 2023-11-05:
it's like this  
//   71     0x3E  0x04     Stage of charge sensor                     15876  * 0.01f            %
  this->publish_state_(this->state_of_charge_sensor_, (float) seplos_get_16bit(offset + 17) * 0.01f);

how to change it would be
//  71    0x3e                     Stage of charge sensor                  62             %

wojtbar, 2023-11-05:
there is none in received frames  seplos-bms state of health
so I dehashed it in seplos_bms.cpp

wojtbar, 2023-11-05:
and it's almost good now

syssi, 2023-11-06:
Could you provide some more examples of the telemetry frame?

syssi, 2023-11-06:
Do you know the position / real value of the charging cycles?

wojtbar, 2023-11-06:
I have all telemetry frames but not all decoded. Which ones should I provide?

wojtbar, 2023-11-06:
CHGoff
>>>  ~0101469AE00201FD22
<<< ~25014F9AE00202FD0B

wojtbar, 2023-11-06:
CHGon
>>>  ~0101469AE00200FD23
>>> ~25014F9AE00200FD04

wojtbar, 2023-11-06:
DSGon
'>>> ~0101469BE00200FD22
'<<< ~25014F9BE00200FD0C

wojtbar, 2023-11-06:
DSGoff
'>>> ~0101469BE00201FD21
'>>> ~25014F9BE00204FD08

syssi, 2023-11-06:
I would be happy about some more status frames (responses to `~25014642E00201FD30`).

wojtbar, 2023-11-06:
Switch off
'>>> ~01014699E00209FD22
'<<< ~25014F990000FD8C

wojtbar, 2023-11-06:
I have a lot of them, I did a wireshark dump of individual commands

wojtbar, 2023-11-06:
~25014F42808000010E0DC40DC60DC60DC60DBF0DBE0DC00DC10DC00DC10DC10DC30DC30DC1060BCE0B9F0B860B860B860B910003134238C7005B6800005B68132A0A4D3E0409E139

wojtbar, 2023-11-06:
~25014F42808000010E0DD10DD20DD30DD30DCD0DCB0DCC0DCC0DCD0DCE0DCD0DCE0DCF0DCE060BDF0BAF0B900B9
A0B9A0BA1000313543957005B6800005B68133D098A3F0409E093
~25014F42808000010E0DD10DD30DD40DD30DCB0DCB0DCC0DCB0DCC0DCE0DCD0DCD0DCE0DCE060BDF0BAF0B900B9A0B9A0BA1000C13543957005B6800005B68133A09C13F0409E08F
~25014F42808000010E0DD10DD30DD40DD40DCB0DCB0DCD0DCC0DCD0DCE0DCD0DCF0DCE0DCD060BDF0BAF0B900B9A0B9A0BA1001013543957005B6800005B68133709923F0409E0AF

wojtbar, 2023-11-06:
at the end there is always 0407, after decoding it is 1033, i.e. the hardware version, I guess
[image]

wojtbar, 2023-11-06:
I modified seplos_bms.cpp a bit and this is the result
[image]

wojtbar, 2023-11-06:
seplos_bms_state_of_change  it's probably 3f before 0407. previously when 60% charged there was 3c, now it's 3e

wojtbar, 2023-11-07:
S/N BarCode
~250146C20000FD99
 ~25014FC260283630303136303136323037323730303031000000F59F

wojtbar, 2023-11-07:
Version 
~250146C10000FD9A
~25014FC16028363032325F3230303134533030305F4E5F323200F549

wojtbar, 2023-11-07:
I already have percentages, but not in a bottle ;-)

syssi, 2023-11-07:
Thanks for your input! Could you provide the position of the `charging cycles`?

wojtbar, 2023-11-07:
I'm very sorry, at the moment the battery is not connected to the inverter and there is no charging and discharging, so there are no cycles.

wojtbar, 2023-11-07:
I think the data is 63 and 64 0x00 0x00

wojtbar, 2023-11-07:
yes, looking at the XDZN software and the file esp32*.yaml address: 0x00 it does not read,
address: 0x01 ok,
  so for another 15 it should work.
I'll know about it soon because I'm making another one just like it and I'll put them together.

wojtbar, 2023-11-07:
My batteries are 14s and the bms is made for 14 ceel  (there is no physical possibility of connecting the 15th and 16th cell)
maybe the BMS firmware is for 14s and for Li-ion
But XDZN software has max 24 cell.

wojtbar, 2023-11-07:
looking at the data that Xdzn sends to bms, it is a sequence of 6 frames repeated from 1 to 6 all the time. reading the bms settings, it sends 16 frames in succession and repeats the sequence from 1 to 6

wojtbar, 2023-11-07:
0xE0, 0X92   ?  CHKSUM

wojtbar, 2023-11-07:
it looks like it was Pace rs232
~25

wojtbar, 2023-11-07:
my interface
https://esp32io.com/tutorials/esp32-rs232

wojtbar, 2023-11-08:
[log-6x.txt](https://github.com/syssi/esphome-seplos-bms/files/13291715/log-6x.txt)

wojtbar, 2023-11-10:
I don't understand what it's like.
When I bought BMS, I received software version 2.4, and now, after conversations, I was sent 3.4. It's just nice:
1. v2.4
>>   ~25014642E00201FD30     (7e3235303134363432453030323031464433300d)
<<  ~25014F42808000010E0DB50DB60DB60DB70DB10DB00DB00DB20DB20DB20DB20DB30DB20DB2060BD00BA00B860B860B860B900001132D2EAC004E2000005B68131409EF3C0409E176

2. v3.4
>> �01014642E00201FD36     (bc3031303134363432453030323031464433360d)
<< ~25014F42808000010E0DB60DB70DB80DB80DB10DB00DB10DB00DB20DB30DB20DB40DB40DB3060BD00BA10B860B860B860B930002132E2EB0004E2000005B681314097F3C0409E187

How to understand ?

wojtbar, 2023-11-10:
problem with temperatures:
temperature 1 is the environment temperature
temperature 2 is the mosfet temperature
temperature 3 is temperature 1
temperature 4 is temperature 2
environment temperature is temperature 3
mosfet temperature is temperature 4

wojtbar, 2023-11-11:
' << ~25014F42808000010E0DB50DB60DB60DB70DB10DB00DB00DB20DB20DB20DB20DB30DB20DB2060BD00BA00B860B860B860B900001132D2EAC004E2000005B681314  09EF  3C 0409 E176      '

-- 09EF   Output Voltage (   external port / MCU )    Sampling Voltage

[image]

wojtbar, 2023-11-11:
[image]

wojtbar, 2023-11-11:
Please check if it is processed correctly
TDT-6022 BMS. I can't do any more.
[TDT-6022.zip](https://github.com/syssi/esphome-seplos-bms/files/13325590/TDT-6022.zip)

wojtbar, 2023-11-11:
interface:

    RS232                                                           UART
┌───┐                                                  ┌───┐                                        ┌───┐
│            │                                                 │             |    <----- RX ----->             │              │
│Seplos |             <---- TX ---->                │RS232  |    <----- TX ----->             │ ESP32/ │
│ BMS   │            <---- RX ---->                 | to TTL │   <----- GND -->            │ ESP8266
│            │            <--- GND -->                 |module|    <----- 3.3V --->            │<-- VCC  |
│            │                                                 │            │                                        │<-- GND|
└───┘                                                  └───┘                                         └───┘

wojtbar, 2024-02-03:
~25014F42808000010E0E610E620E630E640E5B0E5C0E5B0E5C0E5C0E5E0E5E0E5D0E600E5F060BC60B950B7C0B7C0B7C0B85FFFD141E2EBE00521F00015B6813FD0AD4390409E0EF
[tdt_bms.txt](https://github.com/syssi/esphome-seplos-bms/files/14150825/tdt_bms.txt)

wojtbar, 2024-02-03:
You can use this as a new TDT-6022 bms on github.
[tdt-bms.zip](https://github.com/syssi/esphome-seplos-bms/files/14150843/tdt-bms.zip)

syssi, 2024-02-03:
Good job! I will try to compare your code with the one of this project to get an idea how much needs to be changed to support both devices with the same code base.

wojtbar, 2024-02-03:
Question: you are from Poland, and I will translate into English. In simpler Polish, if possible. Dziękuję.

wojtbar, 2024-02-03:
I jeszcze jedno. Interfejs do BMS to esp32 + konwerter rs232 na uart, i do gniazda rj45 ( jest ich 5) do 3 rs232.

syssi, 2024-02-03:
Just my lastname is polish. I cannot speak polish. :-)

wojtbar, 2024-02-03:
Wow.
TDT-bms.yaml
# line 94 address: 0x01
line 94 addres: 0x02

brunopiras, 2024-03-26:
> You can use this as a new TDT-6022 bms on github. [tdt-bms.zip](https://github.com/syssi/esphome-seplos-bms/files/14150843/tdt-bms.zip)

Works fine more than 24h.

Just wait the possibility to add switches entity (CHG and DSG) control.

Thanks u !!

syssi, 2024-05-21:
BLE support isn't implemented yet.

dajpereira, 2024-05-23:
Can you share the conection between the esp32 and the battery? I have an ESP32_C3 that has UART so I just need to connect it to the RS485 port on the BMS?

dajpereira, 2024-05-27:
> You can use this as a new TDT-6022 bms on github. [tdt-bms.zip](https://github.com/syssi/esphome-seplos-bms/files/14150843/tdt-bms.zip)

can you help me with intructions in how to use the zip?

wojtbar, 2024-06-02:
> Czy możesz udostępnić połączenie między esp32 a baterią? Mam ESP32_C3, który ma UART, więc muszę go tylko podłączyć do portu RS485 w BMS?

Port rs232

wojtbar, 2024-06-02:
tdt-6022 for esp-idf and 16s battery.
[tdt-bms.zip](https://github.com/user-attachments/files/15526797/tdt-bms.zip)

wojtbar, 2024-06-02:
[image]
Depending on the type of rs232 to ttl the Vcc voltage can be 5V or 3.3V.

dajpereira, 2024-06-02:
I buy the MAX2323 so it is 3.3V

wojtbar, 2024-06-02:
[image]
 Middle RS232
[image]
RJ11

wojtbar, 2024-06-02:
Communication board  switch DIP  1 on ,  2,3,4 off

brunopiras, 2024-06-03:
> tdt-6022 for esp-idf and 16s battery.
> 
> [tdt-bms.zip](https://github.com/user-attachments/files/15526797/tdt-bms.zip)
> 
> 

New version?

nkinnan, 2024-08-20:
wojtbar: From your provided data frames, it looks like your battery chemistry is not lithium iron, probably why you have a 14s pack rather than 16s, and so your data frame's chemistry is returned as "4F" instead of "46".  

Would you mind sharing the chemistry of your pack?  Or a link to a manufacturer page?  I'm curious to document more about this BMS protocol.

wojtbar, 2024-08-20:
Yes, my BMS is only for 14s. Li-ion lg e78.

[5 more replies]