syssi/esphome-seplos-bms #79: Seplos v3 sensor information

cp6969, 2023-12-16:
I've noticed on v3 some of the sensors have the wrong register type. Some are set as holding - where it should be Read. 
I also noticed the tempreture sensors aren't reading the correct temperatures. They seems to be about 5 degrees higher than the temperatures shown on the actual BMS display?

syssi, 2023-12-17:
Could you enable the `debug` output of the `uart` component and provide some logs? Because of the `debug` mode they will include the RAW traffic and I will able to replay the communication locally.

cp6969, 2023-12-18:
OK great. Attached is the debug file. I think temperature readings are the PCB temps. At the time of this debub the Cell temps were all around 23degrees C 
[seplos-esp_logs.txt](https://github.com/syssi/esphome-seplos-bms/files/13703708/seplos-esp_logs.txt)

cp6969, 2023-12-27:
@syssi I have noticed all the temperature readings are wrong and they only fluctuate within 1°c no matter the actual temps.

cp6969, 2023-12-28:
Ok I have it figured out now. Seplos use a calculation for temps. It's not just the sensor readings. It's the sensor reading minus 2731. See sample sensor feed from the seplos modbus document. So to get correct temps use filters:
  lambda: return x / 10 - 273.15;

syssi, 2023-12-28:
Do you like to provide a pull request to contribute your finding?

syssi, 2023-12-28:
You could apply a negative `offset` filter before the `multiply` filter.

cp6969, 2023-12-28:
I'm very very new to home assistant and yaml. I just asked chat gpt fix my code to deduct 2731 off the sensor readings  and make it work that it has the correct decimal placing🤣 I don't know how pull requests work in GitHub either!! A negative offset filter before the multiply filter should work as well. If you give me the code I'll try it out for you

ferelarg, 2023-12-28:
the cell temperatures are in Kelvin, the register answer in 0.1K.
Kelvin to Celsius: C = K - 273 (C = K - 273.15 if you want to be more precise)
Kelvin to Fahrenheit: F = 9/5(K - 273) + 32 or F = 1.8(K - 273) + 32.

syssi, 2023-12-28:
I will update the configuration example tomorrow. I'm happy about some testing later on!

syssi, 2023-12-29:
> I've noticed on v3 some of the sensors have the wrong register type. Some are set as holding - where it should be Read.

Fixed.

> I also noticed the tempreture sensors aren't reading the correct temperatures. 

Fixed.

Please give it a try using the updated example configuration. Feel free to create more issues if you miss something!

closed 2023-12-29 by syssi