syssi/esphome-seplos-bms #80: Seplos v3 multi battery pack

ferelarg, 2023-12-28:
I have been doing some research on the seplos rs485 bus, and I have programmed a python script to "listen" the data from the complete battery pack (when there is more than one) and publish it to a MQTT server (with auto discovery for home assistant).
When there is only one battery, modbus can be used directly, but when there are several batteries the main one acts as "Master" and there can only be one master, so there is no other option but to listen to what the master asks and decode it.
The code (a draft version) is in: https://github.com/ferelarg/Seplos3MQTT

Privatecoder, 2023-12-28:
you could access the slaves via RS485 directly from the free master's or the last slaves-RS485 port **and** use a second RS485 connection on the CAN-port using a splitter (different BAUD rate of 9600) while also using CAN for Battery<->Inverter communication as CAN and RS485 are using separate pins.

ferelarg, 2023-12-28:
Sorry if I expressed myself wrong, exactly, I connect to any free RS485 port, the one of the master or the one of the last battery.
My first intention was to use CAN (the connection to the inverter) but it does not transmit the cell voltages, which was a data that I was interested in incorporating in Home assistant.

Privatecoder, 2023-12-28:
Maybe you got me wrong too: You can use the CAN-Port using a splitter to connect the CAN-part to the inverter and read the battery-stuff via RS485 simultaneously – including voltages and being able to use the infamous BatteryMonitor supplied by Seplos to connect to it and change its settings etc – **via RS485 through the CAN-port**

ferelarg, 2023-12-28:
I haven't tried that... interesting...
I find the implementation of not being able to modify the parameters of all the batteries centrally very crappy... but well, I understand that it is not common to have to change them on a regular basis. But I have 6 Seplos Mason 280 and it's a pain.
Could I change the configuration of each battery using that other port (the one that shares the can connector)?

ferelarg, 2023-12-28:
This is my home assistant implementation
[image]

ferelarg, 2023-12-29:
I was trying to connect to that port today and it was impossible.
Does it have a different pinout for the RS485?

Privatecoder, 2023-12-29:
I did not check the title – mine is not v3 but the previous version 10E. Have you tried with the baud-rate set to 9600? This works on the 10E BMS.

ferelarg, 2023-12-29:
Yes, I tried, but at least with the pinout I'm using: Can on 4-5 and rs485 on 7-8 it doesn't work.

jamietb, 2024-01-03:
I achieve this using pin 1 = B and pin 2 =A - maybe try that.

Privatecoder, 2024-01-03:
This is from the v3.0 manual for the CAN-Port

[image]

ferelarg, 2024-01-03:
Ok, I will try tomorrow. 
And can you change the ID of each slave on that port? because if so you could set the script as master and control and configure all the batteries centrally..

Privatecoder, 2024-01-03:
No, you can only access the master this way. You can access all slaves via their ID‘s with the second RS485 connection to the _regular_ RS485 Ports on the master or the last slave.

ferelarg, 2024-01-03:
Yes, that's what my script already does, but if you can access the master and can connect in parallel to the others, you should be able to access the others..  In the end it's another RS485 port, you can daisy chain them all.

jolly12f, 2024-01-09:
Hi, I've been trying to read the two batteries via Modbus for months but without getting results, with a single battery these problems didn't exist. If anyone has any news, please let me know!

ferelarg, 2024-01-09:
When you have more than one battery, the main battery is set to "Master" mode and in a RS485 network there can only be one master.
That is why you can read when the battery is alone.

What my script does is to "listen" to what the master asks, and get the data.

jolly12f, 2024-01-09:
With your system were you able to read the values ​​of the two batteries? I already use mqtt with another broker, is it possible to use it for both?

ferelarg, 2024-01-09:
Sure, and you can still use the Seplos app at the same time.

jolly12f, 2024-01-10:
I already use mqtt for another service can I use this together? how can you also use seplos app if port 485 is busy for mqtt?

syssi, 2024-01-10:
@jolly12f A single MQTT broker is required and can be used by tons of different devices simultaneously. Are you asking about the windows application or the Android app?

jolly12f, 2024-01-10:
I use Windows and unfortunately I don't know where to start :(

klatremis, 2024-02-21:
> I have been doing some research on the seplos rs485 bus, and I have programmed a python script to "listen" the data from the complete battery pack (when there is more than one) and publish it to a [...]

Great job! How do we get a ESP32 version of your script for Multi Seplos 3.0? :D

ferelarg, 2024-02-22:
Basically you can use the same methodology: "listen" to the serial port and decode the messages.
I have done a small proof of concept using EspHome and it works fine.
I have not gone any further because in my installation it is more logical to use a rs485 adapter directly.

Privatecoder, 2024-02-22:
@ferelarg an offtopic question: What does the 3.0 BMS send between packs when you just listen (and don't send a telemetry request)? the v16/10e only provides:

- lowest/highest cell
- cells temp 0+1
- dis-/charge current
- total pack-voltage
- residual capacity
- battery capacity
- soc
- port-voltage

and some info I wasn't able to decode yet, probably alarm status..

ferelarg, 2024-02-22:
> EDIT: Nevermind, I found it in your code

No, in my code I only use those registers, but it sends many more... if you look at the protocol manual that is in the "docs" folder you will see that the Master asks for everything (except the serial number and some other minor things).

Privatecoder, 2024-02-22:
I also [wrote a script for the v16/10e](https://github.com/Privatecoder/seplos-mqtt-remote-rs485) which requests telemetry and telesignalization frames and decodes them. they include more data than the frames, requested by the master.. however I am struggling to decode some of the data as there is no documentation about these "intra-pack-communication" frames and the offsets / type of data being used..

jolly12f, 2024-02-22:
> Fondamentalmente puoi usare la stessa metodologia: "ascoltare" la porta seriale e decodificare i messaggi. Ho fatto una piccola prova di concetto utilizzando EspHome e funziona bene. Non sono andato oltre perché nella mia installazione è più logico utilizzare direttamente un adattatore RS485.

 it would be nice to know how to use on esphome too.

clowrey, 2024-02-28:
> Basically you can use the same methodology: "listen" to the serial port and decode the messages. I have done a small proof of concept using EspHome and it works fine. I have not gone any further because in my installation it is more logical to use a rs485 adapter directly.

Could you share your proof of concept ESPhome YAML ? I am hoping to be able to do what you have done but in ESPhome - for easier remote connection and to have less dependence on HA scripts. I also wanted individual cell voltages for all batteries in a multi pack bank.

ferelarg, 2024-02-29:
> Could you share your proof of concept ESPhome YAML ? I am hoping to be able to do what you have done but in ESPhome - for easier remote connection and to have less dependence on HA scripts. I also [...]

For a quick proof of concept in ESPHome I have slightly modified the following script:
https://github.com/htvekov/solivia_esphome
so that the solivia.h looked like this:

```
// *****************************************************************
// *          ESPHome Custom Component Modbus sniffer for             *
// *          Seplosv3 based on Delta Solvia Inverter 3.0 EU G4 TR      *
// *****************************************************************

#include "esphome.h"

class solivia : public PollingComponent, public Sensor, public UARTDevice {
  public:
    solivia(UARTComponent *parent) : PollingComponent(400), UARTDevice(parent) {}
    Sensor *pack_voltage = new Sensor();
    Sensor *current = new Sensor();
    Sensor *remaining_apacity = new Sensor();
    Sensor *total_capacity = new Sensor();
    Sensor *total_discharge_capacity = new Sensor();
    Sensor *soc = new Sensor();
    Sensor *soh = new Sensor();
    Sensor *cycle = new Sensor();
    Sensor *average_cell_voltage = new Sensor();
    Sensor *average_cell_temp = new Sensor();
    Sensor *max_cell_voltage = new Sensor();
    Sensor *min_cell_voltage = new Sensor();
    Sensor *max_cell_temp = new Sensor();
    Sensor *min_cell_temp = new Sensor();
    Sensor *maxdiscurt = new Sensor();
    Sensor *maxchgcurt = new Sensor();
  
  void setup() override {

  }

  std::vector<int> bytes;
  int count = 15;

  //void loop() override {

  void update() {
    while(available() > 0) {
      bytes.push_back(read());      
      //make sure at least 8 header bytes are available for check
      if(bytes.size() < 5)       
      {
        continue;  
      }
      // Check for Delta Solivia Gateway package response.
      if(bytes[0] != 0x01 || bytes[1] != 0x04 || bytes[2] != 0x24) {
        bytes.erase(bytes.begin()); //remove first byte from buffer
        //buffer will never get above 8 until the response is a match
        continue;
      }      
      //ESP_LOGD("custom", "Checking for inverter package: %i", bytes.size());
      
	    if (bytes.size() == 36+5) {
        
        // Some 15 packages are recieved appx. every 10 seconds
        // With this counter we will only update ESPHome sensor appx. every 10 seconds
        // NB. Remove this counter if you're not using a Solivia Gateway. 
    
        TwoByte pack_voltage_data;
        pack_voltage_data.Byte[0] = bytes[1 +3]; // DC voltage lsb
        pack_voltage_data.Byte[1] = bytes[0 +3]; // DC voltage msb

        TwoByte current_data;
[... 224 lines cut ...]

    - name: "max_cell_temp"
      unit_of_measurement: V
      device_class: temperature
      accuracy_decimals: 1

    - name: "min_cell_temp"
      unit_of_measurement: V
      device_class: temperature
      accuracy_decimals: 1

    - name: "maxdiscurt"
      unit_of_measurement: A
      device_class: current
      accuracy_decimals: 2
      filters:

    - name: "maxchgcurt"
      unit_of_measurement: A
      device_class: current
      accuracy_decimals: 2
      filters:
        
        
```

jolly12f, 2024-03-30:
I managed to see something with esphome following your comment, would you just give me some help to see the individual cells even just an example for just one then I'll take care of the rest

DpunktS, 2024-10-02:
@ferelarg könntest du deine YAML mit den Zellspannungen vervollständigen? Aktuell stehe ich nämlich vor dem Problem das ich meine Akkus nicht mehr auslesen kann seitdem ich meinen Speicher erweitert habe.  Deine Lösung mit Seplos3MQTT kommt für mich leider nicht in frage weil ich mir die Werte über ESPHome in den IOBroker hole.

DpunktS, 2024-10-07:
Hier ist eine funktionierende Lösung: https://github.com/syssi/esphome-seplos-bms/issues/112

Sleeper85, 2025-10-29:
> IMO `0x00` is the "master mode". If you want slaves only you should start with `0x01`.

So even with multiple Seplos V3 BMS connected together and numbered starting from 0x01, your YAML for Seplos V3 using Modbus should work?

Why do other comments here say it's not possible to read BMS from Modbus in a multi-BMS configuration ?
Would this only be the case when a BMS has address 0x00 ?

Sleeper85, 2025-11-25:
> I must admit. I don't know. I lost the track here and there are too much wrong informations around.

Other users confirm that with **Multi-BMS SEPLOS V3**, only the version with the [seplos_v3_sniffer](https://github.com/DpunktS/seplos_v3_sniffer/tree/main) seems to work.

See https://github.com/Sleeper85/esphome-yambms/issues/94