syssi/esphome-seplos-bms #86: Invalid Header

sragneau, 2024-02-18:
Hi, 

I have an 10E version of the BMS SEPLOS V2 but i can't receive any data from RS485 bus.
I bought this boad : 
[](https://www.amazon.fr/dp/B0754N794H?psc=1&ref=ppx_yo2ov_dt_b_product_details 
)and 
[](https://www.amazon.fr/dp/B0915RF8S2?psc=1&ref=ppx_yo2ov_dt_b_product_details
)
I checked baudrates and it's 19200 like i need when i connect my laptop to the BMS and no problem when i do that.

I used pin D1 and D2 to solder TX, RX connector.
[image]

I downloaded debug logs, i don't know if there is something i can do now.
[logs_seplos_logs (1).txt](https://github.com/syssi/esphome-seplos-bms/files/14321633/logs_seplos_logs.1.txt)

syssi, 2024-02-18:
How did you connect the ESP to the BMS? Which RS485 module do you use?

sragneau, 2024-02-18:
The first port called canbus is connected to my victron cerbox gx, i connect the esp with the next port called RS485.

syssi, 2024-02-18:
You cannot talk directly to the BMS (AFAIK). You have to translate the UART-TTL signal of the ESP into an RS485 signal using  a converter module which looks like this: https://raw.githubusercontent.com/syssi/esphome-jnge-mppt-controller/main/images/circuit.jpg

sragneau, 2024-02-18:
Ok my links wasn't written into my first post, but i already use this board to convert uart-ttl : 
[image]

to this esp
[image]

syssi, 2024-02-18:
Perfect! If you disconnect the BMS from the RS485 module. Does the TX LED blink periodically while the RX LED stays off?

sragneau, 2024-02-18:
The tx esp ligth blink every 10s. RX led stay off

[image]
[image]

I took pictures of my connection

syssi, 2024-02-18:
Could you provide your YAML configuration too?

sragneau, 2024-02-18:
Yes this my yaml
[seplos.txt](https://github.com/syssi/esphome-seplos-bms/files/14322171/seplos.txt)

sragneau, 2024-02-18:
It may be that communication between my Cerbo GX and the Victron protocol cannot be used simultaneously with RS485 communication. Do you incorporate the Pylontech protocol in your software?