syssi/esphome-seplos-bms #92: about esp32-seplos-v3-example.yaml

cristof48, 2024-04-24:
hello,
I cant find in this code the "cell balance voltage" and the batterie power.
There is something wrong with temperature, please see picture attach

[image]

for SOC I had to add filters / multiply: 0.1

Merci.

syssi, 2024-04-25:
Could you tell me the approximate expected cell temperature? 28°C? Where are you from? ;-)

syssi, 2024-04-25:
Could you increase the log level (see https://github.com/syssi/esphome-seplos-bms?tab=readme-ov-file#debugging) and provide another log in plaintext with some raw traffic? Thanks in advance!

cristof48, 2024-04-25:
I'm from France, I live at an altitude of 1050 meters, and it's cold at the moment.
the temperature given by BMS studio is around 11/13°C (see screenshot) and the esphome logs give around -244°. (see screenshot).
The two screenshots are less than 1 minute apart.
Salutations.
 

Cristof Guerin
La Vaissiere
48220 Le Pont de Montvert
tel :  06 50 44 56 61
      

De : "Sebastian Muszynski"
A : "syssi/esphome-seplos-bms" ,"cristof48" ,"Author"
Envoyé: jeudi 25 Avril 2024 09:39
Objet : Re: [syssi/esphome-seplos-bms] about esp32-seplos-v3-example.yaml (Issue #92)
 

 

Could you tell me the approximate expected cell temperature? 28°C? Where are you from? ;-)

cristof48, 2024-04-25:
‌I'm not very good with esphome, it's the code from the PVBRAIN2 project with Seby in France. I have an esp32 S3 16Mb with a lot of devices in it. Your code is integrated into the PVbrain2 project.
 

Cristof Guerin
La Vaissiere
48220 Le Pont de Montvert
tel :  06 50 44 56 61
      

De : "Sebastian Muszynski"
A : "syssi/esphome-seplos-bms" ,"cristof48" ,"Author"
Envoyé: jeudi 25 Avril 2024 09:47
Objet : Re: [syssi/esphome-seplos-bms] about esp32-seplos-v3-example.yaml (Issue #92)
 

 

Could you increase the log level (see https://github.com/syssi/esphome-seplos-bms?tab=readme-ov-file#debugging) and provide another log in plaintext with some raw traffic? Thanks in advance!

syssi, 2024-04-25:
I've pushed a fix for the temperature sensors. Do you miss something else?

closed 2024-07-05 by syssi