syssi/esphome-seplos-bms #96: unhandled data and bad header

FritsOV, 2024-04-26:
In adition to post: [https://github.com/syssi/esphome-seplos-bms/issues/86](url) and [https://github.com/syssi/esphome-seplos-bms/issues/67](url)
I also use the Seplos BMS 48200-1101-10E-SP118-16S (or 1101-4800-10E-BOM-V16-57FTDJR)   in a Daly Green mason attached to 3 Victron multiplus II. 
I managed to setup a esp8266 lilo with a ttl as described here [https://github.com/syssi/esphome-seplos-bms/discussions/55#discussioncomment-9228944](url)

I am getting some errors like unhandled data, bad header and failed CRC check. 
[logs_esphome-web-761f72_logs.txt](https://github.com/syssi/esphome-seplos-bms/files/15127831/logs_esphome-web-761f72_logs.txt)

how to resolve these errors?

syssi, 2024-04-29:
Some thoughts:

1. The unhandled messages/warnings are fine and can be ignored. These are frames of frame type `0x5A`. I don't know what the frame is used for. This frame type isn't implemented and therefor reported as `unhandled`. Do you know any details? Could you ask your seller/manufacturer about some details of this protocol flavor?
2. I assume the CRC errors can be reduced if a hardware UART is used instead of software serial (GPIO4/GPIO5). If you want to use the hardware UART you have to use GPIO1/GPIO3 and disable the logger (`baud_rate: 0`) here.

In general your setup is working properly and you are missing just a frame every now and then.

FritsOV, 2024-05-06:
I already did and got the offer for an online meeting. I'm not a programmer but if you want to i can set this meeting including you. 
I never could get the code working using the hardware uart. Could give it another try.

closed 2025-07-25 by syssi