syssi/esphome-seplos-bms #99: No response received

otello86, 2024-06-02:
Hello,

i install the program on an ESP32 and  try to use it with 3 batteries in parallel with V3.0 Seplos BMS. I tried with this yaml : https://github.com/syssi/esphome-seplos-bms/blob/main/esp32-seplos-v3-example-multiple-battery-banks.yaml to get only total voltage for starting but i already got this response in mqtt : 
[0;36m[D][modbus_controller:043]: Modbus command to device=1 register=0x1000 countdown=0 no response received - removed from send queue[0m
[0;36m[D][modbus_controller:043]: Modbus command to device=2 register=0x1000 countdown=0 no response received - removed from send queue[0m
for each battery.

I try to change pin 1/2 with 7/8 but this has no effects. I saw the RS485 to TTL flashing led RXD and TXD but nothing on mqtt

Thank you for your response

otello86, 2024-06-02:
i activated log and i got this :

[19:13:21][D][uart_debug:158]: >>> "\x00\x04\x10\x00\x00\x014\xDB"
[19:13:21][D][uart_debug:158]: >>> "\x00\x04\x10\x00\x00\x014\xDB"
[19:13:21][D][modbus_controller:043]: Modbus command to device=0 register=0x1000 countdown=0 no response received - removed from send queue
[19:13:28][D][uart_debug:158]: >>> "\x02\x04\x10\x00\x00\x0159"
[19:13:29][D][uart_debug:158]: >>> "\x02\x04\x10\x00\x00\x0159"
[19:13:29][D][uart_debug:158]: >>> "\x02\x04\x10\x00\x00\x0159"
[19:13:29][D][uart_debug:158]: >>> "\x02\x04\x10\x00\x00\x0159"
[19:13:29][D][uart_debug:158]: >>> "\x02\x04\x10\x00\x00\x0159"
[19:13:29][D][modbus_controller:043]: Modbus command to device=2 register=0x1000 countdown=0 no response received - removed from send queue
[19:13:30][D][uart_debug:158]: >>> "\x01\x04\x10\x00\x00\x015\n"
[19:13:30][D][uart_debug:158]: >>> "\x01\x04\x10\x00\x00\x015\n"
[19:13:30][D][uart_debug:158]: >>> "\x01\x04\x10\x00\x00\x015\n"
[19:13:30][D][uart_debug:158]: >>> "\x00\x04\x10\x00\x00\x014\xDB\x01\x04\x10\x00\x00\x015\n"
[19:13:31][D][uart_debug:158]: >>> "\x00\x04\x10\x00\x00\x014\xDB\x01\x04\x10\x00\x00\x015\n"
[19:13:31][D][modbus_controller:043]: Modbus command to device=1 register=0x1000 countdown=0 no response received - removed from send queue
[19:13:31][D][uart_debug:158]: >>> "\x00\x04\x10\x00\x00\x014\xDB"
[19:13:31][D][uart_debug:158]: >>> "\x00\x04\x10\x00\x00\x014\xDB"
[19:13:31][D][uart_debug:158]: >>> "\x00\x04\x10\x00\x00\x014\xDB"
[19:13:31][D][modbus_controller:043]: Modbus command to device=0 register=0x1000 countdown=0 no response received - removed from send queue

syssi, 2024-07-05:
Sorry for the late response. Do you still have trouble?

Your log indicates no incoming traffic. If you disconnect the BMS which LED is still blinking?