syssi/esphome-seplos-bms #112: Seplos V3.0 BMS Multi Pack

DpunktS, 2024-10-01:
Hallo,

ich habe mein Seplos V3.0 BMS mit einem 2. BMS aufgerüstet (Multi Pack).  Leider bekomme ich sehr viele CRC Fehler für das 2. BMS und somit keine vernünftigen Sensorwerte.

```
[20:52:04][W][modbus:114]: Modbus CRC Check failed! 9E67!=100
[20:52:04][W][modbus:114]: Modbus CRC Check failed! 31C2!=E8
```
Wie kann ich meinen Speicher weiterhin auslesen?

syssi, 2024-10-01:
Wie genau sieht deine YAML-Konfiguration aus? Wie häufig fragst du die Geräte nach neuen Werten?

DpunktS, 2024-10-01:
Im Grunde ist es deine Standard YAML für den ESP32 mit allem Werten und 10s intervall. 

Mir ist auch aufgefallen wenn ich meinen selbst gebauten RS485 USB Adapter verwende erreiche ich mit dem Laptop das erst BMS nicht, das 2. BMS hat angeblich die Adresse 3 und der Wechselrichter hat keine Kommunikation zum BMS mehr.  Laut Seplos Anleitung hat die RJ45 Buchse noch 2 Adressierungspins (4,5). Ich habe mir jetzt nochmal den originalen Adapter bestellt zum testen.  

Einzelne hat mein Setup bis jetzt fehlerfrei funktioniert und auch die Kommunikation zwischen den beiden BMSs funktioniert.

DpunktS, 2024-10-02:
Ich glaube, die Lösung für mein Problem ist dieser Lösungsansatz https://github.com/syssi/esphome-seplos-bms/issues/80

Leider wurde die YAML nicht mit den Zellspannungen vervollständigt.

DpunktS, 2024-10-07:
Ich habe es jetzt geschafft, meine 2 Speicher auszulesen. Es wäre schön, wenn ihr es in euer Projekt aufnehmen könntet, weil ESPHome bald die Unterstützung für "Custom Sensor Component" einstellt.

```
// *****************************************************************
// *          ESPHome Custom Component Modbus sniffer for          *
// *          Seplos v3                                            *
// *****************************************************************
//02 04 10 00 00 12 74 F4 
//02 04 24 15 20 19 EC 36 09 71 48 00 1A 01 DD 03 E8 00 01 0D 34 0B AD 0D 3C 0D 2F 0B B4 0B A9 00 00 00 B4 00 B4 03 E8 88 55 
//02 04 11 00 00 1A 74 CE 
//02 04 34 0D 31 0D 2F 0D 34 0D 35 0D 31 0D 30 0D 39 0D 3C 0D 36 0D 31 0D 30 0D 32 0D 36 0D 3A 0D 32 0D 36 0B AF 0B AA 0B A9 0B B4 0A AB 0A AB 0A AB 0A AB 0B B3 0B AC 1C 77 
//02 01 12 00 00 90 39 2D 
//02 01 12 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 03 00 00 E9 99 
//01 04 24 15 1D 21 39 43 10 71 48 04 99 02 50 03 E3 00 32 0D 32 0B B3 0D 40 0D 2B 0B B6 0B B0 00 00 00 B4 00 B4 03 E8 D8 60 
//01 04 34 0D 2B 0D 31 0D 34 0D 2E 0D 35 0D 31 0D 34 0D 32 0D 40 0D 31 0D 33 0D 30 0D 30 0D 2F 0D 37 0D 2C 0B B0 0B B6 0B B4 0B B5 0A AB 0A AB 0A AB 0A AB 0B B4 0B B2 9F 90 
//01 01 12 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 03 00 00 DA AA 

#include "esphome.h"

class solivia : public PollingComponent, public Sensor, public UARTDevice {
  public:
    solivia(UARTComponent *parent) : PollingComponent(400), UARTDevice(parent) {}
    Sensor *bms1_pack_voltage = new Sensor();
    Sensor *bms1_current = new Sensor();
    Sensor *bms1_remaining_capacity = new Sensor();
    Sensor *bms1_total_capacity = new Sensor();
    Sensor *bms1_total_discharge_capacity = new Sensor();
    Sensor *bms1_soc = new Sensor();
    Sensor *bms1_soh = new Sensor();
    Sensor *bms1_cycle = new Sensor();
    Sensor *bms1_average_cell_voltage = new Sensor();
    Sensor *bms1_average_cell_temp = new Sensor();
    Sensor *bms1_max_cell_voltage = new Sensor();
    Sensor *bms1_min_cell_voltage = new Sensor();
    Sensor *bms1_max_cell_temp = new Sensor();
    Sensor *bms1_min_cell_temp = new Sensor();
    Sensor *bms1_maxdiscurt = new Sensor();
    Sensor *bms1_maxchgcurt = new Sensor();

    Sensor *bms1_cell_1 = new Sensor();
    Sensor *bms1_cell_2 = new Sensor();
    Sensor *bms1_cell_3 = new Sensor();
    Sensor *bms1_cell_4 = new Sensor();
    Sensor *bms1_cell_5 = new Sensor();
    Sensor *bms1_cell_6 = new Sensor();
    Sensor *bms1_cell_7 = new Sensor();
    Sensor *bms1_cell_8 = new Sensor();
    Sensor *bms1_cell_9 = new Sensor();
    Sensor *bms1_cell_10 = new Sensor();
    Sensor *bms1_cell_11 = new Sensor();
    Sensor *bms1_cell_12 = new Sensor();
    Sensor *bms1_cell_13 = new Sensor();
    Sensor *bms1_cell_14 = new Sensor();
    Sensor *bms1_cell_15 = new Sensor();
    Sensor *bms1_cell_16 = new Sensor();
    Sensor *bms1_cell_temp_1 = new Sensor();
    Sensor *bms1_cell_temp_2 = new Sensor();
    Sensor *bms1_cell_temp_3 = new Sensor();
    Sensor *bms1_cell_temp_4 = new Sensor();
    Sensor *bms1_case_temp = new Sensor();
    Sensor *bms1_power_temp = new Sensor();

    Sensor *bms0_pack_voltage = new Sensor();
    Sensor *bms0_current = new Sensor();
    Sensor *bms0_remaining_capacity = new Sensor();
    Sensor *bms0_total_capacity = new Sensor();
    Sensor *bms0_total_discharge_capacity = new Sensor();
    Sensor *bms0_soc = new Sensor();
    Sensor *bms0_soh = new Sensor();
    Sensor *bms0_cycle = new Sensor();
[... 1067 lines cut ...]
      filters:
        - throttle: 1s

    - name: "${bms_name0} cell_temp_4"
      unit_of_measurement: °C
      device_class: temperature
      accuracy_decimals: 1
      filters:
        - throttle: 1s

    - name: "${bms_name0} case_temp"
      unit_of_measurement: °C
      device_class: temperature
      accuracy_decimals: 1
      filters:
        - throttle: 1s

    - name: "${bms_name0} power_temp"
      unit_of_measurement: °C
      device_class: temperature
      accuracy_decimals: 1
      filters:
        - throttle: 1s

```

DpunktS, 2024-10-08:
So sehen die empfangenen Nachrichten aus:

```
2024.10.04 15:28:39:798:Total Len:266
02 04 10 00 00 12 74 F4 02 04 24 15 20 19 D7 36 09 71 48 00 1A 01 DD 03 E8 00 01 0D 33 0B AD 0D 3B 0D 2F 0B B4 0B A9 00 00 00 B4 00 B4 03 E8 EF EB 02 04 11 00 00 1A 74 CE 02 04 34 0D 31 0D 2F 0D 34 0D 35 0D 31 0D 30 0D 39 0D 3B 0D 36 0D 31 0D 30 0D 32 0D 36 0D 39 0D 32 0D 36 0B AF 0B AA 0B A9 0B B4 0A AB 0A AB 0A AB 0A AB 0B B3 0B AC B2 0C 02 01 12 00 00 90 39 2D 02 01 12 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 03 00 00 E9 99 01 04 24 15 1D 21 1F 43 10 71 48 04 99 02 50 03 E3 00 32 0D 31 0B B3 0D 40 0D 2B 0B B6 0B B0 00 00 00 B4 00 B4 03 E8 75 E8 01 04 34 0D 2B 0D 31 0D 33 0D 2E 0D 35 0D 31 0D 34 0D 32 0D 40 0D 31 0D 32 0D 30 0D 30 0D 2F 0D 37 0D 2C 0B B0 0B B6 0B B4 0B B5 0A AB 0A AB 0A AB 0A AB 0B B4 0B B2 08 D5 01 01 12 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 03 00 00 DA AA 

2024.10.04 15:28:40:000:Total Len:266
02 04 10 00 00 12 74 F4 02 04 24 15 20 19 EC 36 09 71 48 00 1A 01 DD 03 E8 00 01 0D 34 0B AD 0D 3C 0D 2F 0B B4 0B A9 00 00 00 B4 00 B4 03 E8 88 55 02 04 11 00 00 1A 74 CE 02 04 34 0D 31 0D 2F 0D 34 0D 35 0D 31 0D 30 0D 39 0D 3C 0D 36 0D 31 0D 30 0D 32 0D 36 0D 3A 0D 32 0D 36 0B AF 0B AA 0B A9 0B B4 0A AB 0A AB 0A AB 0A AB 0B B3 0B AC 1C 77 02 01 12 00 00 90 39 2D 02 01 12 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 03 00 00 E9 99 01 04 24 15 1D 21 39 43 10 71 48 04 99 02 50 03 E3 00 32 0D 32 0B B3 0D 40 0D 2B 0B B6 0B B0 00 00 00 B4 00 B4 03 E8 D8 60 01 04 34 0D 2B 0D 31 0D 34 0D 2E 0D 35 0D 31 0D 34 0D 32 0D 40 0D 31 0D 33 0D 30 0D 30 0D 2F 0D 37 0D 2C 0B B0 0B B6 0B B4 0B B5 0A AB 0A AB 0A AB 0A AB 0B B4 0B B2 9F 90 01 01 12 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 03 00 00 DA AA 

2024.10.04 15:28:40:203:Total Len:266
02 04 10 00 00 12 74 F4 02 04 24 15 20 19 FE 36 09 71 48 00 1A 01 DD 03 E8 00 01 0D 34 0B AD 0D 3C 0D 30 0B B4 0B A9 00 00 00 B4 00 B4 03 E8 BA CC 02 04 11 00 00 1A 74 CE 02 04 34 0D 31 0D 30 0D 34 0D 35 0D 31 0D 30 0D 39 0D 3C 0D 36 0D 31 0D 31 0D 32 0D 36 0D 3A 0D 33 0D 36 0B AF 0B AA 0B A9 0B B4 0A AB 0A AB 0A AB 0A AB 0B B3 0B AC 3F 53 02 01 12 00 00 90 39 2D 02 01 12 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 03 00 00 E9 99 01 04 24 15 1D 21 4E 43 10 71 48 04 99 02 50 03 E3 00 32 0D 32 0B B3 0D 40 0D 2B 0B B6 0B B0 00 00 00 B4 00 B4 03 E8 9C 71 01 04 34 0D 2B 0D 31 0D 34 0D 2E 0D 35 0D 31 0D 34 0D 33 0D 40 0D 31 0D 33 0D 30 0D 30 0D 2F 0D 37 0D 2C 0B B0 0B B6 0B B4 0B B5 0A AB 0A AB 0A AB 0A AB 0B B4 0B B2 83 00 01 01 12 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 03 00 00 DA AA 

```

jolly12f, 2024-10-11:
could you make a guide on the work to be done?

DpunktS, 2024-10-11:
Du musst den ersten Code als seplos_v3_sniffer.h speichern und an den selben Ort der ESPHome YAML schieben.

Der 2. Code ist die Konfiguration für ESPHome die du wie gehabt bearbeitest. 

Ich meinem Beispiel ist keine CRC Prüfung eingebaut weshalb ab und zu mal ein falscher Wert auftaucht.

sunnysingh-84, 2024-10-12:
I attempted this and got the following error when installing:

INFO ESPHome 2024.9.2
INFO Reading configuration /config/esphome/seplos-bms.yaml...
Failed config

esphome: [source /config/esphome/seplos-bms.yaml:6]
  
  'name' is a required option for [esphome].
  includes: 
    - ./seplos_v3_sniffer.h

DpunktS, 2024-10-12:
Der Fehler sagt, dass er die Datei nicht an den angegebenen Speicherort findet.

Alexandros89, 2024-11-10:
> Ho provato a farlo e ho ricevuto il seguente errore durante l'installazione:
> 
> INFO ESPHome 2024.9.2 INFO Lettura configurazione /config/esphome/seplos-bms.yaml... Configurazione non riuscita
> 
> esphome: [sorgente /config/esphome/seplos-bms.yaml:6]
> 
> 'name' è un'opzione obbligatoria per [esphome]. Comprende: - ./seplos_v3_sniffer.h

the point should be the name of the esphome path

nammto, 2024-11-19:
Hi! I'm using the sniffer and it works fantastic for up to two batteries. I have 4 batteries set up and get sometimes weird values for voltages and temperatures in 3rd battery. On 4th battery I get those weird values most of the time.  There I read mostly 4000 degrees battery temperature. If I modify the code to read only 4th battery the issues remain the same. What could be wrong there? Does anyone else have more than 2 batteries in their setup?

DpunktS, 2024-11-19:
Es ist möglich das dein Bus leicht gestört ist und die fehlende CRC-Prüfung es nicht filtert. Oder du hast bei der Erweiterung einen Fehler gemacht.

nammto, 2024-11-20:
Thanks for fast reponse. I suppose if i had a typo in code, there would be false readings all the time. Tried wirh rx buffer 2048 and also with two separate esp32 and each having separate rs485 converter for each pack pair but the result is same. Always 3rd and 4th pack readings fail. Even if i just try to read 4th pack, I get these wierd values.
Below You can see how different pack current reults are received.
[image]
[image]
[image]

What else could I do? Is it possible I miss something in .h file code except battery address?

DpunktS, 2024-11-20:
So wie ich das sehe sind das nur vereinzelte falsche Wert die die CRC-Prüfung filtern würde. Die Akkus senden alles 200ms ihren Status somit fallen ein paar gefilterte Wert nicht auf.

Ich werde es Laie mal versuchen eine CRC-Prüfung zu integrieren :).  Oder vielleicht bekommst du das schneller hin. Es müsste von diesem Projekt https://github.com/ferelarg/Seplos3MQTT auf ESPHome angepasst werden.

nammto, 2024-11-20:
:-) I don't think my knowledge reached that far to do it.

DpunktS, 2024-11-20:
Probiere mal bitte diesen Sniffer mit CRC Prüfung
[seplos_v3_sniffer.zip](https://github.com/user-attachments/files/17827552/seplos_v3_sniffer.zip)

nammto, 2024-11-20:
Got home to test it. I'm now with Your CRC code able to read pack 1&2 data, also pack 3&4 data if I read only 2 packs at a time. If I try to read all 4 packs, I get only data for pack 3 and 4.

nammto, 2024-11-20:
ah I'm dummy, didn't iclude addresses in first If statement

DpunktS, 2024-11-20:
@nammto wie hast du diese Zeile für den 3 . und 4. Akku geändert?
`if (!((bytes[0] == 0x02 && bytes[1] == 0x04 && (bytes[2] == 0x24 || bytes[2] == 0x34)) || (bytes[0] == 0x01 && bytes[1] == 0x04 && (bytes[2] == 0x24 || bytes[2] == 0x34)))) {
 `

nammto, 2024-11-20:
> @nammto wie hast du diese Zeile für den 3 . und 4. Akku geändert? `if (!((bytes[0] == 0x02 && bytes[1] == 0x04 && (bytes[2] == 0x24 || bytes[2] == 0x34)) || (bytes[0] == 0x01 && bytes[1] == 0x04 && (bytes[2] == 0x24 || bytes[2] == 0x34)))) { `

This is for all 4 packs:   if (!((bytes[0] == 0x02 && bytes[1] == 0x04 && (bytes[2] == 0x24 || bytes[2] == 0x34)) || (bytes[0] == 0x01 && bytes[1] == 0x04 && (bytes[2] == 0x24 || bytes[2] == 0x34)) || (bytes[0] == 0x03 && bytes[1] == 0x04 && (bytes[2] == 0x24 || bytes[2] == 0x34)) || (bytes[0] == 0x04 && bytes[1] == 0x04 && (bytes[2] == 0x24 || bytes[2] == 0x34)))) {

DpunktS, 2024-12-18:
@nammto probierst du diesen seplos_v3_sniffer bitte mit deinen 4 Akkus aus. Mit meinen 2 Akkus funktioniert es gut.
https://github.com/DpunktS/seplos_v3_sniffer/tree/main

DpunktS, 2024-12-19:
Ich danke dir für dein Lob. :)
Die Fehler habe ich korrigiert. Dein Vorschlag mit den automatischen anlegen der Sensoren wollte ich auch realisieren leider ist mit kein weg bekannt den ESPHome unterstützt.

tugsi, 2024-12-20:
@DpunktS Kleine Frage am Rande, hast Du zusätzlich den CAN am Wechselrichter angeschlossen? 
Mir ist noch nicht so ganz klar, ob das geht, also CAN am (in meinem Fall) Victron und gleichzeitig mit Deiner Lösung alles per RS485 auslesen. 
Finde SEPLOS irgendwie konfus, habe zwei Packs mit V3 parallel angeschlossen und zack ist einer per Bluetooth nicht mehr erreichbar, der andere ja.
Deswegen würde ich auch gerne mehr Infos auslesen im Betrieb.

DpunktS, 2024-12-20:
Ja meine 2 Seplos V3 sind per CAN am Wechselrichter angeschlossen. Zum Verständnis bei Modbus RS485 kann es nur einen Master geben und das ist bei mehreren Akkus immer bms0 also, der wo CAN angeschlossen ist. Deswegen ist auch nur noch der Master per Bluetooth erreichbar. Der ESP mit den bisherigen Projekten würde auch wieder ein Master sein und deswegen funktioniert die Kommunikation ab 2 Akkus nicht mehr. Meine Lösung liest nur die Kommunikation zwischen Master und Slave's mit und wertet diese aus.

tugsi, 2024-12-20:
Danke für die Rückmeldung. Das Verständnis Modbus habe ich (hab beruflich mit Bus-Systemen zu tun, u.a. auch Modbus). Mir war nur nicht klar, ob gleichzeitig dann noch CAN sauber funktioniert, da in diversen Foren auch vieles widersprüchliche liest. Und dann kommt noch Seplos V3 oder alt etc. 
Dann schau ich mal, wann ich Zeit dazu finde und baue mir das mal nach. Danke Dir.

tugsi, 2024-12-28:
@DpunktS Nur zum Verständnis, hast Du zwischen dem ESP und dem Seplos noch ein RS485-TTL-Wandler dazwischen und wenn ja welchen?

DpunktS, 2024-12-28:
Ich habe diesen verwende. Es kann aber jeder ttl rs485 Konverter verwendet werden.
[image]

tugsi, 2024-12-28:
Danke, ich dachte ich hatte noch einen, aber leider nicht.
Genau den Typ hab ich gerade bei Ali bestellt 10Stck für knapp über 5€, dann hat man Reserve ;-)

syssi, 2024-12-28:
Je nach Konverter ist manuelle Flusskontrolle über einen zusätzlichen GPIO notwendig. Das obige Modul kommt ohne diese Fusskontrolle aus, deshalb nutze ich es lieber.

PostICT, 2025-03-12:
After adding a second battery pack to my setup I had problems to red the BMS data with the ESP32 setup.
I discovered this threat and  configured an ESP32 with this: https://github.com/DpunktS/seplos_v3_sniffer/blob/main/seplos_v3_sniffer.yaml
The data of both bms-en are showing nicely on the webpage of the ESP32, but not under  "Entities" in Homeassistant.
What am I missing?

DpunktS, 2025-03-12:
Du musst für HomeAssistant die Datenpunkte noch vervollständigen damit sie schön angezeigt werden. Zum Beispiel mit:

```
device_class: battery
state_class: measurement
```
Ich selbst verwende iobroker und da ist es nicht nötig. Wenn du eine schön Vorlage hast nehme ich die gerne mit auf.

PostICT, 2025-03-12:
Danke,
habe das volgende versucht:

sensor:
 # bms0
  - platform: seplos_parser
    device_class: voltage
    state_class: measurement
    seplos_parser_id: seplos_v3
    name: bms0 pack_voltage
    unit_of_measurement: "V"
    accuracy_decimals: 2
    filters:
      - throttle: ${time}s
    
Aber ich bin nicht eine pro mit Homeassistant und keine werte "bms0 pack voltage" beim "entities".
Habe vieleicht nog etwas vergessen zu configureren.

sunnysingh-84, 2025-03-12:
Works well for me, maybe try deleting the device from the HA (Settings > Devices > ESPHome > Device;Name) and let it be rediscovered?

PostICT, 2025-03-12:
Thanks sunnysingh.
You pointed me in the right directon as it turnsout that the device is apparantly "disabled by config entry".
Within the ESPhome dashboard on the 3 vertical dots I had to disable the device. There after the "enable" option showed and than I was able to "enable". All battery data than was shown under entities.

broekieman, 2025-04-01:
> Thanks sunnysingh. You pointed me in the right directon as it turnsout that the device is apparantly "disabled by config entry". Within the ESPhome dashboard on the 3 vertical dots I had to disable [...]

Would you share your yaml with me?

broekieman, 2025-04-01:
Could somebody please share his yaml with me?
I have the following and no error but also no data:

esphome:
  name: mi_seplos_multibms

esp8266:
  board: d1_mini

# Disable logging over UART (required)
logger:
  baud_rate: 0

# Enable Home Assistant API
api: 

# Enable Home Assistant API

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.178.85
    gateway: 192.168.178.1
    subnet: 255.255.255.0

ota:
  platform: esphome

external_components:
  - source:
      type: git
      url: https://github.com/DpunktS/seplos_v3_sniffer

uart:
- id: seplos
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 19200
  rx_buffer_size: 2048

substitutions:
  time: "5" # update interval for all data points
  
seplos_parser:
  id: seplos_v3
  bms_count: 2  # Anzahl der Akku's (max. 16)
  update_interval: ${time}
  uart_id: seplos  # SeplosParser verwendet diese UART ID

sensor:
 # bms0
  - platform: seplos_parser
    seplos_parser_id: seplos_v3
    name: bms0 pack_voltage
    unit_of_measurement: "V"
    accuracy_decimals: 2
    filters:
      - throttle: ${time}s
  - platform: seplos_parser
    seplos_parser_id: seplos_v3
    name: bms0 current
    unit_of_measurement: "A"
    accuracy_decimals: 2
    filters:
      - throttle: ${time}s
  - platform: seplos_parser
    seplos_parser_id: seplos_v3
    name: bms0 remaining_capacity
[... 551 lines cut ...]
  - platform: seplos_parser
    seplos_parser_id: seplos_v3
    name: bms0 active_protections
 # bms1
  - platform: seplos_parser
    seplos_parser_id: seplos_v3
    name: bms1 system_status
  - platform: seplos_parser
    seplos_parser_id: seplos_v3
    name: bms1 active_balancing_cells
  - platform: seplos_parser
    seplos_parser_id: seplos_v3
    name: bms1 cell_temperature_alarms
  - platform: seplos_parser
    seplos_parser_id: seplos_v3
    name: bms1 cell_voltage_alarms
  - platform: seplos_parser
    seplos_parser_id: seplos_v3
    name: bms1 FET_status
  - platform: seplos_parser
    seplos_parser_id: seplos_v3
    name: bms1 active_alarms
  - platform: seplos_parser
    seplos_parser_id: seplos_v3
    name: bms1 active_protections

DpunktS, 2025-06-06:
Hier ist alles beschrieben.

https://github.com/DpunktS/seplos_v3_sniffer

jjtoard, 2025-06-06:
Hi,

I have 3 rs485 ports.
How should I connect them ?
Nothing is described in the manual.

[image]

jjtoard, 2025-06-06:
Its brand new.  Just put the battery together and turned it on.
Didn't configure it yet with the software.  Will do that tomorrow.  Was just hoping to get the esp32 read it out already but as always what is expecte to take just 10 min is taking a lot more time.

galaxinetwisp-droid, 2025-09-04:
Hi, I'm still having trouble viewing the data from the two batteries with the sniffer. The data arrives at the RS485 TTL converter (the LEDs flash), but when I connect the TTL module's Rx to the ESP32's Tx, it stops flashing and there is no communication.