syssi/esphome-seplos-bms #116: Seplov V16-10E Multi Pack

Wiki591, 2024-12-08:
I am reading pack 1 (adress 0x01) and pack 2 (adress 0x02) from three packs, Seplos BMS V16-10E. The master attached per CAN bus to the inverter is not accessible as known. But anyway, at least I get an idea of the drift of the SOC of the two batteries. Detailed information, i.e. voltages of cells, I am reading with BLE from three attached Neey balancer.

Having a look at the debug protocol I can see, that on the RS485 line a device with adress 0x03 ist sending some unrecognized data. I suppose, this is the master which sends the cummulated/summarized information of the whole packs which are sent to the inverter, too.

The entries look like this: 

```
[10:48:14][W][seplos_modbus:133]: Got SeplosModbus frame from unknown address 0x03! 
[10:48:15][W][seplos_bms:022]: Unhandled data received (data_len: 0x00): 20.01.46.5A.00.00 (6)
[10:48:15][W][seplos_bms:022]: Unhandled data received (data_len: 0x40): 20.01.46.5A.C0.40.00.01.0D.2D.0D.2A.0B.6A.0B.63.02.41.15.12.6B.68.6D.60.03.D6.15.16.00.00.00.00.00.00.03.02.80.82 (38)
```
Is there any way to translate these informations, too? Would be a nice-to-have for i.e. estimating the SOC of the master.

syssi, 2024-12-08:
To which and how many RS485 ports did you attach the ESP?

Wiki591, 2024-12-08:
The BMS are cascaded: master pack RS485->pack1->pack2->ESP32 TTL RS485 adaptor

So there are four devices on the RS485, three Seplos and the ESP (connected to pack 2).

Dip switches according to Seplos documentation: Master: Dip 6 on (master for three packs)
Pack1; Dip 1 on (slave #1, answering on adress 0x01)
Pack2; Dip 2 on (slave #2, answering on adress 0x02)
All other Dips off

protocol version 0x20, 19200 baud
Works so far for pack 1 and pack 2

syssi, 2024-12-08:
Could you share your configuration YAML? In worst-case you have two Modbus masters on the same bus at the moment.

Wiki591, 2024-12-08:
Sure, here we go:

```
substitutions:
  name: seplos-bms
  battery_bank1: "Seplos Pack1"
  battery_bank2: "Seplos Pack2"
  device_description: "Monitoring Seplos BMS via RS485"
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin: GPIO4
  rx_pin: GPIO5

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.seplos-bms"
    version: 0.0.2

esp32:
  board: wemos_d1_mini32

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: DEBUG

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

uart:
  id: uart_0
  baud_rate: 19200
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  rx_buffer_size: 384

seplos_modbus:
  id: modbus0
  uart_id: uart_0

seplos_bms:
  - id: battery_bank1
    address: 0x01
    protocol_version: 0x20
    seplos_modbus_id: modbus0
    update_interval: 10s
  - id: battery_bank2
    address: 0x02
    protocol_version: 0x20
    seplos_modbus_id: modbus0
    update_interval: 10s

sensor:
  - platform: seplos_bms
    seplos_bms_id: battery_bank1
    min_cell_voltage:
[... 139 lines cut ...]
      name: "${battery_bank2} Gesamtspannung"
    current:
      name: "${battery_bank2} Strom"
    power:
      name: "${battery_bank2} Leistung"
    charging_power:
      name: "${battery_bank2} Ladeleistung"
    discharging_power:
      name: "${battery_bank2} Entladeleistung"
    residual_capacity:
      name: "${battery_bank2} verbleibende Kapazität"
    battery_capacity:
      name: "${battery_bank2} Batterie Kapazität"
    rated_capacity:
      name: "${battery_bank2} Nominalkapazität"
    state_of_charge:
      name: "${battery_bank2} SOC"
    charging_cycles:
      name: "${battery_bank2} Zyklen"
    state_of_health:
      name: "${battery_bank2} SOH"
    port_voltage:
      name: "${battery_bank2} Anschlussspannung"

```

Wiki591, 2024-12-08:
I forgot:

Thank you for your fast reaction!

syssi, 2024-12-08:
Let's talk german to speed things up. :-) 

Aktuell hast du zwei Geräte auf dem Bus, welche die Master-Rolle einnehmen. Ich versuche kurz zu beschrieben, wie es normalerweise aussieht und an welcher Stelle dein Setup von diesem Konzept abweicht:

Normalerweise hat man einen aktiven Master auf dem Bus. Dieser pollt nacheinander alle Slaves und bittet einen nach dem anderen um eine Meldung. Die Slaves nehmen also eine passive Rolle ein und dürfen nur sprechen, wenn sie kurz vorher aufgefordert wurden. Wenn sich alle Partner an diese Regeln halten sollte es keine Kollisionen auf dem Bus geben. Gesprochen wird immer nur, wenn man gefragt wurde.

Stellt man ein Seplos BMS auf die Adresse `0x00`, dann nimmt dieses die Master-Rolle ein und stellt ab diesem Moment periodisch allen anderen Geräten (`0x01 ... 0xff`) auf dem Bus fragen, aggregiert/summiert die Informationen aus den Antworten und stellt sie ueber CAN-Bus als Aggregat (= z.B. SOC des gesamten Setups) zur Verfügung.

Du hast nun einen zweiten Master an den Bus gehangen. Dieser quatscht nun periodisch dazwischen und fordert ebenfalls einzelne Packs auf etwas über sich zu verraten. Kollisionen sind deshalb vorprogrammiert und im schlimmsten Fall verschluckt sich sogar die Seplos Firmware, welche bereits die Master-Rolle einnimmt (ist nur eine These!), weil sie auf eine Frage eine völlig falsche Antwort erhält.

Es gibt jetzt zwei Möglichkeiten:

1. Dieses Projekt geht davon aus, dass kein Seplos BMS die Master-Rolle einnimmt und stattdessen der ESP alleiniger Master ist. Du stellt die einzelnen Packs also auf Adresse `0x01`, `0x02` und `0x03` und genießt den Vorteil, dass du Auskunft über alle Packs erhälst ohne eine Ausnahme. In dieser Variante kann kein Inverter per CAN-Bus mit Infos versorgt werden.

2. Du haelst an deinem aktuellen Setup fest aber du knippst du TX-Leitung zwischen ESP und RS485 Converter durch oder setzt das `update_interval` der `seplos_bms`-Komponente auf `never`. Das Ziel ist, dass der ESP sich auf dem Bus ruhig verhält und zu einem passiven Zuhörer wird. Die Informationen die Fließen, sollten weiterhin decodiert werden, sofern die Implementierung sie versteht/kennt. Du erhaelst also weiterhin Informationen über Slave `0x01` und `0x02`, jedoch quasselt dein ESP nicht mehr dazwischen.

Wiki591, 2024-12-08:
OK, verstehe ich und werde die Variante "never" nutzen und eben halt nur lauschen, denn der WR braucht die aggregierten Infos aus dem Gesamtpaket bzw. soll sie bekommen. Mir war nicht klar, daß der ESP sich zum Master aufschwingt und bedanke mich für den Hinweis. Hab ich mir ganz ehrlich auch keine Gedanken drum gemacht, läuft halt.

Was mir nur aufgefallen ist: Laut Warnings sendet ein Geräte mit Adresse 0x03 iwas an Daten. Könnte das der Master sein, der ja bei Seplos unter 0x00 nicht erreichbar ist (selbst von deren PC-Software aus nicht) und wären die iwie nutzbar?

Das, was mich interessieren würde, wäre der SOC des Master, da die drei Dinger mit ihrem SOC sehr weit driften, die Shunts scheinen nicht besonders akkurat zu sein (oder die AD-Wandler). Volladungen sind mit meiner kleine Anlage nicht sehr häufig, so daß deren "Selbstkalibrierungen" zum Ladeende manchmal zeitlich weit auseinander liegen. Ich greife sporadisch manuell ein um die wieder auf einen realen Wert zu setzen. Nur ist die Menschheit von Natur aus faul, so auch ich, und würde mir den Weg in den Keller für den Blick auf die Anzeige der BMS sparen wollen.

Aber ist ein ganz klares nice-to-have!

syssi, 2024-12-08:
Lass uns gerne den Verkehr auf dem Bus genauer unter die Lupe nehmen. Dein kurzer Logauszug oben enthielt nur zwei Frames von Slave `0x01` an der Zahl hinter der `20` erkennbar. Der Verkehr von `0x03` wurde verworfen. Er sollte sichtbar werden, wenn du den Debug-Modus der `uart` -Komponente aktivierst:

https://github.com/syssi/esphome-seplos-bms?tab=readme-ov-file#debugging

Kannst du einen weiteren Logauszug zur Verfügung stellen, welcher die Warnung als auch die Debug-Ausgaben enthält?

Wiki591, 2024-12-08:
Danke für den Hinweis, mit tieferem Hinsehen hätte ich schon im Vorwege sehen können, daß der angemeckerte Datensatz tatsächlich vom Slave 1 kommt.

Ein Debug-Log kann ich gerne machen. Nur ist die Frage, macht das Sinn? Der von mir aufgefangene Snippet ist wohl eine Antwort auf eine Anfrage des Masters, Function Code 5A (Byte 3) kann ich in der Doku des Protokolles V 2.0 nicht finden.

Hab jetzt grad mal probeweise nur gelauscht, keinerlei verständlicher Verkehr mehr.

syssi, 2024-12-08:
Die Kommunikation gibt es in vielen Ausprägungen, deshalb wäre ein Mitschnitt spannend als Basis für eine Diskussion. :-) 

Die paar nicht dekodierbaren Frames von Slave `0x01` sind vermutlich wirklich wertlos insbesondere, weil das kurze Frames `0 Bytes Daten` in sich traegt. 

Trotzdem finde ich spannend, was das Gerät `0x03` grundsätzlich zu melden hat. Möglicherweise sonst es wirklich nur die Requests von Master ohne jede Antwort. Aber das würde erst die Mitschnitt zeigen.

Wiki591, 2024-12-08:
Soderle,

aus akademischem Interesse den ESP auf Lauschangriff gestellt, es werden im Sekundentakt vom Slave 1 und Slave 2 wohl cummulierte Daten gesendet, ein Gerät mit Adresse 0x03 taucht dabei nicht auf:
```
[18:46:07][D][uart_debug:158]: <<< "~2001465A0000FD9D\r"
[18:46:07][W][seplos_bms:022]: Unhandled data received (data_len: 0x00): 20.01.46.5A.00.00 (6)
[18:46:07][D][uart_debug:158]: <<< "~2001465AC04000010CE10CDE0B5C0B53FFFE14983FC26D600246149A00000000000003028010EFE8\r"
[18:46:07][W][seplos_bms:022]: Unhandled data received (data_len: 0x40): 20.01.46.5A.C0.40.00.01.0C.E1.0C.DE.0B.5C.0B.53.FF.FE.14.98.3F.C2.6D.60.02.46.14.9A.00.00.00.00.00.00.03.02.80.10 (38)
[18:46:07][D][uart_debug:158]: <<< "~2002465A0000FD9C\r"
[18:46:07][W][seplos_bms:022]: Unhandled data received (data_len: 0x00): 20.02.46.5A.00.00 (6)
[18:46:07][D][uart_debug:158]: <<< "~2002465AC04000010CE00CDD0B620B5A0001149741236D600253149700000000000003028010F072\r"
[18:46:07][W][seplos_bms:022]: Unhandled data received (data_len: 0x40): 20.02.46.5A.C0.40.00.01.0C.E0.0C.DD.0B.62.0B.5A.00.01.14.97.41.23.6D.60.02.53.14.97.00.00.00.00.00.00.03.02.80.10 (38)

```
Jetzt stelle ich den ESP mal auf sabbeln und logge das dann wieder mit.

...to be continued...

Wiki591, 2024-12-08:
Soderle, mit der Doku des Protokolls 2.0 in der Hand, ESP auf sabbeln gesstellt, schön an diesem Snippet zu sehen, der Request an Slave 2 zum Senden der Telemetriedaten (Function Code 42) durch den ESP wird sauber vom Slave 2 beantwortet mit einem Datensatz in der zu erwartenden Länge von 81 Bytes, Function Code 00 für alles iO:
```

[19:00:37][D][uart_debug:158]: >>> "~20024642E00202FD33\r"
[19:00:37][D][uart_debug:158]: <<< "~2002460010960002100CDD0CDD0CDE0CDE0CDE0CDE0CDE0CDE0CDF0CDF0CDE0CDE0CDE0CDF0CDF0CDE060B5B0B620B600B620B890B650000149741220A6D6002536D60003903E81497FFF"
[19:00:37][D][uart_debug:158]: <<< "B000D1C801B24DBAB\r"
[19:00:37][I][seplos_bms:030]: Telemetry frame (81 bytes) received
``` 

Der Datensatz mit Null Daten scheint mit Function Code 5A die Aufforderung des Master an die Slaves zu sein, cummulierte Daten zu senden, die auch prompt beantwortet wird, ebenfalls mit 5A an entsprechender Stelle. Das bleibt auch so, wenn der ESP dazwischensabbelt.

Imho ist damit eigentlich keine Kollision zweier Master zu befürchten, da über die Funktion eindeutig gesagt wird, wer was haben will und es auch mundgerecht geliefert bekommt. Oder?

Nur schade, daß vom Master tatsächlich nix aufzutauchen scheintaußer seinen Requests. Die Meldung "0x03" buche ich mal unter geshreddertem Datensatz ab.

Wiki591, 2024-12-08:
...sollte der Wunsch nach nem längeren Logging bestehen - interessehalber - würde ich den ESP auf einen anderen MQTT Broker hetzen zum Loggen, da ich bei der Menge an Meldungen Störung in meinem Smarthome befürchte. Die Hardware, auf dem das Ding läuft,ist nicht besonders leistungsfähig...

syssi, 2024-12-08:
> Imho ist damit eigentlich keine Kollision zweier Master zu befürchten, da über die Funktion eindeutig gesagt wird, wer was haben will und es auch mundgerecht geliefert bekommt. Oder?

Mit Kollisionen meine ich "gleichzeitiges Senden". Dieses Risiko besteht unmittelbar, sobald mehr als ein Master auf dem Bus unterwegs ist. Du kannst dir die Kommunikation wie Zeitslots vorstellen. Im ersten Zeitfenster (z.B. 250ms) sendet der Master einen Request. Im naechsten Fenster (250ms) antwortet der Slave. Hoert der Master 250ms lang keine Antwort, dann ist die Frage entweder verloren gegangen oder der adressierte Slave existiert nicht auf dem Bus.

In einer anderen Disziplin ist deine Annahme aber richtig: Die Wahrscheinlichkeit, dass der Master eine `function code 42`-Response konsumiert an der Stelle, wo er eigentlich eine `0x5A`-Antwort erwartet ist hoffentlich gering. Es sei denn die Seplos-Firmware ist sehr schlecht geschrieben.

syssi, 2024-12-08:
Was diesem Projekt fehlt ist ein Decoder für die `0x5A`-Response. Mal schauen, was wir in dem Frame finden können:

```
      0  0x20
     1  0x01
     2  0x46
     3  0x5A
     4  0xC0
     5  0x40   data_len=64/2=32 bytes

     6  0x00
     7  0x01
     8  0x0C 0xE1    3297  3.297V?
    10  0x0C 0xDE    3294  3.294V?
    12  0x0B 0x5C    2908
    14  0x0B 0x53    2131
    16  0xFF 0xFE
    18  0x14 0x98    5272  52.72V?
    20  0x3F 0xC2    16322
    22  0x6D 0x60    28000  280.00Ah?
    24  0x02 0x46    582
    26  0x14 0x9A    5274
    28  0x00 0x00
    30  0x00 0x00
    32  0x00 0x00
    34  0x03 0x02    770
    36  0x80 0x10    32784
```

Sehen irgendwelche dieser Werte für die plausibel aus?

Privatecoder, 2024-12-08:
https://github.com/Privatecoder/seplos-mqtt-remote-rs485/blob/eed8976cca3f3fab946383f7d1ebefb54600289e/src/fetch_bms_data.py#L424-L455

Privatecoder, 2024-12-08:
das ist allerdings nicht vollständig.. ich weiß, dass ein Teil der Daten auch _alarm_-flags enthält

syssi, 2024-12-08:
Mir ist keine offizielle Beschreibung der Frames bekannt. Dein erster Aufschlag sieht aber bereits vielversprechend aus.

Privatecoder, 2024-12-08:
ich hatte schonmal bei Seplos angefragt, aber die haben mir dann ne Pylontech-Protokoll-Doku geschickt, die ich mit ehrlicherweise nicht angesehen habe. Da sie sicher viel davon kopiert haben, findet man dort ggf. etwas. Ich schaue die Tage mal.

In jedem Fall enthalten die _Intra-Pack-Daten_ aber weniger Informationen.

Zurück zum Thema:

Zwei Master auf dem Bus bereiten bei mir auch auf lange Sicht mit 12 Packs bisher keine Probleme. Den Master frage ich zusätzlich mit 9600er Baud über einen Splitter am CAN ab.

Wiki591, 2024-12-08:
> Zwei Master auf dem Bus bereiten bei mir auch auf lange Sicht mit 12 Packs bisher keine Probleme. Den Master frage ich zusätzlich mit 9600er Baud über einen Splitter am CAN ab.

Ersteinmal bin ich für das Teilen Deiner Erfahrung recht dankbar. Und das Abgreifen des Masters über die 9600 am CAN wäre eigentlich das Datenschnipsel, das mir fehlen würde. Mit Splitter meinst Du sicherlich nen Y-Adapter - laienhaft gefragt?

Privatecoder, 2024-12-08:
> Was in übersetzten Daten identifizieren kann:
> 
> ```
>      8  0x0C 0xE1    3297  3.297V? Spannung Max Zelle
>     10  0x0C 0xDE    3294  3.294V? Spannung Min Zelle
>     12  0x0B 0x5C    2908-2731=17.7°C Könnte Max Temp Sensor sein
>     14  0x0B 0x53    2899-2731=16.8°C Könnte Min Temp Sensor sein
>     16  0xFF 0xFE
>     18  0x14 0x98    5272  52.72V? Gesamtspannung
>     20  0x3F 0xC2    16322 Derzeitige Ladekapazität
>     22  0x6D 0x60    28000  280.00Ah? Nominalkapazität
>     24  0x02 0x46    582 Ladung in Prozent 58,2%
>     26  0x14 0x9A    5274 Anschlussspannung 52,74V
>     28  0x00 0x00
>     30  0x00 0x00
>     32  0x00 0x00
>     34  0x03 0x02    770 k.A.
>     36  0x80 0x10    32784 k.A.
> ```

siehe mein Link, da hab ich schon recht viel dekodiert, aber eben nicht alles und insb. die Alarm-Daten nicht.

Privatecoder, 2024-12-08:
> Ersteinmal bin ich für das Teilen Deiner Erfahrung recht dankbar. Und das Abgreifen des Masters über die 9600 am CAN wäre eigentlich das Datenschnipsel, das mir fehlen würde. Mit Splitter meinst Du [...]

Korrekt, ein Y-Adapter wie [diesen hier](https://www.amazon.de/gp/product/B00D3KIQXC)

Wiki591, 2024-12-08:
> siehe mein Link, da hab ich schon recht viel dekodiert, aber eben nicht alles und insb. die Alarm-Daten nicht.

..bin grad schon imPython versunken...

Privatecoder, 2024-12-08:
es geht auch die Variante, den Master via Bluetooth-Display zu pollen..

Wiki591, 2024-12-08:
Ich geh mal die Variante mit 9600 Baud auf dem CAN Anschluss, die Hardware hab ich hier und werde beizeiten berichten....

syssi, 2024-12-08:
Ich habe einen kleinen Decoder implementiert, der das dekodierte Frame erstmal ins Log schreibt:

```
[20:42:00][I][seplos_bms:165]: Intra pack communication frame (38 bytes) received
[20:42:00][D][seplos_bms:186]: Maximum Cell Voltage: 3.296 V
[20:42:00][D][seplos_bms:189]: Minimum Cell Voltage: 3.293 V
[20:42:00][D][seplos_bms:192]: Temperature 1: 18.3 °C
[20:42:00][D][seplos_bms:195]: Temperature 2: 17.5 °C
[20:42:00][D][seplos_bms:199]: Current: 0.01 A
[20:42:00][D][seplos_bms:203]: Total voltage: 52.71 V
[20:42:00][D][seplos_bms:206]: Residual capacity: 166.75 Ah
[20:42:00][D][seplos_bms:209]: Battery capacity: 280.00 Ah
[20:42:00][D][seplos_bms:212]: State of charge: 59.5 %
[20:42:00][D][seplos_bms:215]: Port voltage: 52.71 V
[20:42:00][D][seplos_bms:218]: Unknown28: 0x00 0x00 (0.000000)
[20:42:00][D][seplos_bms:221]: Unknown28: 0x00 0x00 (0.000000)
[20:42:00][D][seplos_bms:224]: Unknown28: 0x00 0x00 (0.000000)
[20:42:00][D][seplos_bms:227]: Unknown28: 0x03 0x02 (770.000000)
[20:42:00][D][seplos_bms:230]: Unknown28: 0x80 0x10 (32784.000000)
```

Wenn du den `external_components_source` auf den Feature-Branch aenderst kannst du den neuen Stand ausprobieren:

```
substitutions:
  external_components_source: github://syssi/esphome-seplos-bms@decode-intra-pack-communicatio
```

syssi, 2024-12-08:
Über den Mehrwert muss ich aber noch einmal eine Nacht schlafen. ;-)

Wiki591, 2024-12-08:
Tolle Arbeit, vor Allem schnell.

Für den Mehrwert: wohl eher nicht, weil immernoch der Master fehlt und die schmale Datentiefe wohl nur für die Aufsummierung für den WR Sinn macht.

Am CAN Port des Masters kann mit 9600 Baud der Master wohl tatsächlich mit dem Protokoll 2.0 ausgelesen werden, hab das mal q&d ausprobiert. Der ESP32 sollte in der Lage sein, die Daten aller drei ruckelfrei zu verarbeiten, werde wohl morgen oder so das Projektchen mit nem Splitter und einer zweiten RS485 aufpimpen und dann berichten. Ein Dauerlauf muß dann die Stabilität des CAN Bus beweisen, weil mein Solis WR bei Schluckauf auf dem CAN die am Notstrom angeschlossenen Geräte abwirft - was blöde ist.

Danke auf jeden Fall für den umfänglichen und schnellen Support, auch an @Privatecoder für den Tipp per Splitter und 9600er Datenrate an den CAN Bus zu gehen.

Wiki591, 2024-12-09:
Zwischenbericht:
Auslesen mit ESP32 mini D1, zwei TTL->RS485, einer mit 9600 Baud am CAN Bus für den Master, einer mit 19200 originär an RS485 funktioniert für alle drei Packs. Zusammenarbeit zum WR kann ich im Moment nicht beurteilen, mein eigentlich vorhandener RJ45-Splitter ist irgendwo unter die Räder gekommen seit ISDN-Feierabend.

Wiki591, 2024-12-09:
...und Ernüchterung: CAN und RS485 zusammen über Splitter kommt nix vom Seplos zurück. CAN zum WR funktioniert.
Ich vermute mal, die Endwiderstände des CAN machen den Pegel vom max485 (betrieben mit 3,3V) kaputt.
Anyone any idea?

syssi, 2024-12-09:
Ich hatte es bisher so verstanden, dass auf der gleichen Buchse auf unterschiedlichen Pinnen unterschiedliche Signale anliegen. Auf welchen Pinnen liegt das CAN- und auf welchen Pinnen das RS485-Signal? Ich wuerde die zwei relevanten Leitungen zum ESP/RS485-Converter führen und die CAN-Leitungen zum Inverter. Verstehe ich etwas falsch?

Wiki591, 2024-12-09:
Nö, verstehst wohl nix falsch, hast Recht, wie ein tiefer Blick in die Unterlagen zeigt. CAN liegt auf 4+5, RS485 auf 1/8+2/7. Aber gemeinsam GND auf 3/6. Meine hemdsärmelige Theorie also für die Tonne.
In dem Moment, wenn ich vom Splitter den CAN vom WR abziehe ist alles wieder tutti. Ich werde es mal mit separierten Kabeln versuchen, wie Du es vorschlägst bzw. die Pins 1,2,7 und 8 des WR abklemmen.

Wiki591, 2024-12-09:
...nicht zu fassen, kaum macht man was richtig und arbeitet den Dokumentationen der Gerätehersteller entsprechend und sauber und ordentlich, funktioniert es auch...Peinlich...
[image]

Wiki591, 2024-12-09:
Jepp. Wünschte mir nur, mit yaml soweit firm zu sein, daß der u.s. Spaghetticode etwas kompakter daherkommen würde.

Falls jemand das nachbauen möchte (könnte auch gerne in die Beispiele einfliessen):

```##############################################################
# Battery Pack 1xMaster, 2xSlaves 
# Seplos 1101-10E
# RS485 connection master pack using splitter on CAN Bus port
# in parallel with CAN Bus connection to inverter
# RS485 connection slave packs using original RS485 port
##############################################################
substitutions:
  name: seplos-bms
  battery_bank0: "Seplos Pack0"
  battery_bank1: "Seplos Pack1"
  battery_bank2: "Seplos Pack2"
  device_description: "Monitoring 3 stacked Seplos BMS, 1 Master, 2 Slaves via RS485"
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin0: GPIO16
  rx_pin0: GPIO17
  tx_pin1: GPIO4
  rx_pin1: GPIO5

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "Wiki.Seplos BMS 1+2 basierend auf syssi/esphome-seplos-bms"
    version: 0.1.0

# esp8266:
#  board: d1_mini

esp32:
  board: wemos_d1_mini32

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: NONE

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
mqtt:
  broker: !secret mqtt_host
#  broker: !secret mqtt_host_elvira
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client
  # Workaround to prevent Homeassistant's message 'unavailable'
  # on MQTT disconnect using different topics for
  # LWT and Birth - kinda crowbar but works
  birth_message:
    topic: imhere/topic
    payload: online
  will_message:
    topic: imaway/topic
    payload: offline

# api:
[... 517 lines cut ...]
      name: "${battery_bank2} Batterie Kapazität"
      filters:
      - delta: 0.001
    rated_capacity:
      name: "${battery_bank2} Nominalkapazität"
      filters:
      - delta: 0.001
    state_of_charge:
      name: "${battery_bank2} SOC"
      filters:
      - delta: 0.001
    charging_cycles:
      name: "${battery_bank2} Zyklen"
      filters:
      - delta: 0.001
    state_of_health:
      name: "${battery_bank2} SOH"
      filters:
      - delta: 0.001
    port_voltage:
      name: "${battery_bank2} Anschlussspannung"
      filters:
      - delta: 0.001

```

Wiki591, 2024-12-09:
Für mich ist das jetzt ersteinmal fertig.

Wiki591, 2024-12-09:
Weil es doch die eine oder andere Kollision auf dem originären RS485 Port gibt (Master fordert sekündlich die Infos von den Slaves) und die RS485 Schnittstelle von allen dreien komplette Ruhe zeigt bar jeglicher Kommunikation werde ich morgen - oder so - mal ein Kabel basteln, um die drei RS485 Ports, die auf 9600 Baud laufen und parallel zum CAN-Anschluß vorhanden sind, miteinander zum Bus zu kombinieren und entsprechend berichten.

Wiki591, 2024-12-09:
Hat mir nu ja keine Ruhe gelassen. Der von Dir angestoßene tiefe Blick in die Dokumentation des Boards sowie eine kurze Kontrolle am Board selber hat ergeben, daß der RS485-Teil des CAN Ports auf dem Board schon fertig durchgeschleift ist mit
1->8 (B-) 
2->7 (A+) 
3->6 (Gnd). 
Damit war es völlig easy, ein Kabel mit 3xRJ45 zu krimpen, das mir jetzt einen sauberen, dem ESP gehörigen RS485 Bus zur Verfügung stellt, Splitter(am Master)->Pack 1->Pack 2->ESP. Zwar "nur" mit 9600 Baud, aber was solls? Funktioniert, alle Beteiligten sind glücklich und zufrieden, neben klein Wiki auch die insgesamt fünf Geräte (WR, 3xSeplos, ESP).
Ganz wichtig war, daß der WR tatsächlich nur mit den beiden Kabeln CAN H und CAN L angeschlossen wird.
Deine diversen Hinweise haben mir ermöglicht: Habe fertig.
Nochmal, falls jemand nachbauen will oder auch zur Aufnahme in Deine Beispielpakete mit meiner jetzigen Konfiguration:

```
##############################################################
# Battery Pack 1xMaster, 2xSlaves 
# Seplos 1101-10E
# RS485 connection master pack using splitter on CAN Bus port
# in parallel with CAN Bus connection to inverter
# RS485 connection slave packs using original RS485 port
##############################################################
substitutions:
  name: seplos-bms
  battery_bank0: "Seplos Pack0"
  battery_bank1: "Seplos Pack1"
  battery_bank2: "Seplos Pack2"
  device_description: "Monitoring 3 stacked Seplos BMS, 1 Master, 2 Slaves via RS485"
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin0: GPIO16
  rx_pin0: GPIO17

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "Wiki.Seplos BMS 1+2 basierend auf syssi/esphome-seplos-bms"
    version: 0.2.0

# esp8266:
#  board: d1_mini

esp32:
  board: wemos_d1_mini32

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: NONE

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
mqtt:
  broker: !secret mqtt_host
#  broker: !secret mqtt_host_elvira
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client
  # Workaround to prevent Homeassistant's message 'unavailable'
  # on MQTT disconnect using different topics for
  # LWT and Birth - kinda crowbar but works
  birth_message:
    topic: imhere/topic
    payload: online
  will_message:
[... 512 lines cut ...]
    battery_capacity:
      name: "${battery_bank2} Batterie Kapazität"
      filters:
      - delta: 0.001
    rated_capacity:
      name: "${battery_bank2} Nominalkapazität"
      filters:
      - delta: 0.001
    state_of_charge:
      name: "${battery_bank2} SOC"
      filters:
      - delta: 0.001
    charging_cycles:
      name: "${battery_bank2} Zyklen"
      filters:
      - delta: 0.001
    state_of_health:
      name: "${battery_bank2} SOH"
      filters:
      - delta: 0.001
    port_voltage:
      name: "${battery_bank2} Anschlussspannung"
      filters:
      - delta: 0.001
```

closed 2024-12-09 by syssi

Privatecoder, 2024-12-23:
@Wiki591 @syssi ich konnte es noch nicht testen, aber es gibt für die 10E / v2 eine [neue Firmware](https://diysolarforum.com/resources/seplos-v2-10e-latest-firmware.464/)

Bei Akku-Doktor las ich zum Vorgägner `16.06.10`

`16.06.10 fixes the problem that 485 communication cannot understand the Bluetooth APP. And 485 and CAN can now be used in the same program. The subsequent volitronic inverter can communicate with it as long as the pin is changed, without updating the firmware. And you can view its information in the Bluetooth APP. The previous version fixed the functions of communication anti-interference and parallel 1CC charging. There are also some customized programs that I am not convenient to talk about.`

Vielleicht kann man damit nun auch am Master ohne Splitter über einen der freien RS485-Ports alle Packs auslesen

Wiki591, 2024-12-23:
Danke für die Info, nur bei FW Updates folge ich gerne der "Never touch a running system"-Philosophie. 

Wie oben beschrieben, hab ich mir ein Kabel gecrimpt, mit dem ich den auf den Boards parallel zum CAN durchgeschleiften RS485 Port aller Packs zu einem Bus verbunden habe, zusätzlich am RJ45 ziehe ich die beiden Pins vom CAN aus dem Master raus zum WR - selbstgecrimpter Splitter halt. Funktioniert bis jetzt völlig problemlos und bis jetzt ohne Schluckäufe.

Da ich auf dem originären RS485 Port bei meinen ersten Versuchen immer wieder Kollisionen mit der sekündlichen Abfrage des Masters an seine Slaves beobachtet habe, lasse ich den jetzt in Ruhe, da die Kommunikation ja auch zukünftig erforderlich ist. 

Benötigt wird der separate RS485 (dann mit 19200bd) in meinem Setup nur dann, wenn ich mit dem BMS Monitor von Seplos die Einstellungen runterladen und ggf. ändern möchte - was bei einem Produktiv-System, das seit 1 1/2 Jahren läuft sehr selten ist. Der Batteriemonitor von Seplos gibt mir über den o.g. RS485-Bus (nochmals drei Pins rausgezogen) jetzt auch die Meßwerte aller drei Packs, nur halt das Setup tuts nicht.

Privatecoder, 2025-02-18:
@Wiki591 
> Hat mir nu ja keine Ruhe gelassen. Der von Dir angestoßene tiefe Blick in die Dokumentation des Boards sowie eine kurze Kontrolle am Board selber hat ergeben, daß der RS485-Teil des CAN Ports auf [...]

Kannst du nochmal erklären, wie du das jetzt genau verkabelt hast (vielleicht mit Foto)?

> Splitter(am Master)->Pack 1->Pack 2->ESP

Hieraus entnehmen ich: Master CAN-Port mit Y-Kabel (zwei Adern (CAN) an WR und 3x RS485 an ESP). Du hast die Packs unterinenader aber über die RS485 ganz normal verbunden?

Danke schon mal

Wiki591, 2025-02-18:
Am Master hab ich einen dreifachen Splitter gecrimpt, Pin 4&5 -> CAN, Pin 7&8 -> CAN Port Slave 1 Pin 1&2, CAN Port Slave1 Pin 7&8 -> CAN Port Slave 2 Pin 1&2, CAN Port Slave 2 Pin 7&8 -> ESP mit 1xRS485 TTL. Alles läuft an diesem Bus mit 9,6kBd.
Am Master hab ich Pin 1&2 auf einen RJ45-Stecker Pin 1&2 gelegt, mittels RJ45 Kupplung kann ich mit dem BMS-Manager von Seplos auf dem PC ber nen RS485-USB TTL alle drei Packs sehen (mit 9,6kBd, klar), aber nicht konfigurieren. Das ginge nur über den originären RS485 Port mit 19,2kBd. Dazu muss dann am Master wieder der Dip 1 auf 0 gesetzt werden.
Den originären RS485 Port aller drei Packs lasse ich für deren interne Kommunikation, normal gestackt Master->Slave 1->Slave 2

[image]

Privatecoder, 2025-02-18:
ich danke!

PanzerLemmy, 2025-04-26:
Hallo, kurze Frage, das geht nur mit der V3.0 Version, oder ist das auch mit dem Seplos BMS V2.0 möglich?
Danke.

Wiki591, 2025-04-27:
Öhm, eine Version V2.0 kenne ich nicht. Das V16-10E (siehe Titel) ist meines Wissens nach der direkte Vorgänger vom 3.0 (s.a. https://www.akkudoktor.net/t/seplos-bms-faq-sammlung/8843) .
Da ich ausgenutzt habe, daß der RS485 Port am CAN auf dem Board hardwareseitig durchgeschleift ist, hängt das Ergebnis auch nicht aneinem besonderen Firmware-Stand.

Privatecoder, 2025-04-27:
Also wenn man es ganz genau nimmt, kommt das BMS von SH energy und dort beginnen die Modelle alle mit _EHV_ oder _EMU_. Ein _V2.0 / 10E / V16_ ist also tatsächlich ein **EMU-1101** – wird aber in den Weiten des Internets mithin eben _V2.0 / 10E / V16_ genannt, weil Seplos vorher das 10C anbot (was dann die _V1.0_ wäre)

syssi, 2025-04-28:
Mach mal bitte ein Foto deines RS485-Converters. Musst du dich möglicherweise um die Flusskontrolle kümmern?

PanzerLemmy, 2025-04-28:
[image]

syssi, 2025-04-28:
Dieses Modul kümmert sich automatisch um die Flusskontrolle. Daran liegt es also nicht. Im Battery Monitor müsstest du den rohen Datenverkehr in einer Exceldatei exportieren können. Kannst du diesen Datei zur Verfügung stellen? Ich würde gerne den Verkehr dem implementierten Protokoll hier im Projekt vergleichen.

PanzerLemmy, 2025-04-28:
Ich mag es kaum sagen, aber es geht jetzt.
Wie gesagt, ich habe die gleiche Hardware an meinem Wechselrichter und habe dort aus diesem Video
www.youtube.com/watch?v=PGaL0th2dh4 die Schaltung nachgebaut.
Ich habe da jetzt einfach mal die RXD / TXD GPIOs auf 21/22 wie in dem Video gelegt und die YAML angepasst und jetzt geht es wie es soll. Verstehe ich zwar nicht, da ich erst den gleichen ESP32 hatte, den man hier auf Bildern gesehen hat. Jetzt ist dieser im Einsatz(siehe Bild). ich würde gerne wissen, warum es so klappt bzw. was ich da wohl falsch gemacht habe...Ich sags ja, bin da kein Profi.....;-)

[image]

syssi, 2025-04-28:
Ohne eine Beschreibung vom vorher & nachher zu kennen, lässt sich leider nur raten.

PanzerLemmy, 2025-04-28:
Ursprungscode :esp32-example.yaml

substitutions:
  name: seplos-bms
  device_description: "Monitor a Seplos BMS via RS485"
  external_components_source: github://syssi/esphome-seplos-bms@main
 # tx_pin: GPIO16
 # rx_pin: GPIO17

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-seplos-bms"
    version: 1.2.0

esp32:
  board: esp32dev
  framework:
    type: arduino
    
    
external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
#mqtt:
#  broker: !secret mqtt_host
#  username: !secret mqtt_username
#  password: !secret mqtt_password
#  id: mqtt_client

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

uart:
  id: uart_0
  # Please set the default baudrate of your Seplos BMS model here. It's sometimes 19200 baud instead of 9600.
  baud_rate: 19200
  tx_pin: GPIO21
  rx_pin: GPIO22
  # The increased RX buffer size is important because
  # the full BMS response must fit into the buffer
  rx_buffer_size: 384

seplos_modbus:
  id: modbus0
  uart_id: uart_0
  rx_timeout: 150ms

seplos_bms:
  id: bms0
  # Dip switch configuration of a single pack setup / address 0x0
  #  8    7    6    5    4    3    2    1
  # off, off, off, off, off, off, off, off
  address: 0x00
  # Known protocol versions: 0x20 (Seplos), 0x26 (Boqiang)
  protocol_version: 0x20
  seplos_modbus_id: modbus0
  update_interval: 10s

sensor:
  - platform: seplos_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    temperature_1:
      name: "${name} temperature 1"
    temperature_2:
      name: "${name} temperature 2"
    temperature_3:
      name: "${name} temperature 3"
    temperature_4:
      name: "${name} temperature 4"
    temperature_5:
      name: "${name} environment temperature"
    temperature_6:
      name: "${name} mosfet temperature"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    residual_capacity:
      name: "${name} residual capacity"
    battery_capacity:
      name: "${name} battery capacity"
    rated_capacity:
      name: "${name} rated capacity"
    state_of_charge:
      name: "${name} state of charge"
    charging_cycles:
      name: "${name} charging cycles"
    state_of_health:
      name: "${name} state of health"
    port_voltage:
      name: "${name} port voltage"

Ich habe da das Board, API und die RXD TXD angepasst, MQTT inaktiv gesetzt.Die Baud Rate hatte ich als erstes auf 19200 gesetzt.

Hier sind noch die Logdateien, falls die Aufzeichnung länger sein soll, kein Problem.

[DataRecord_2025.04.28_072431.xls](https://github.com/user-attachments/files/19945428/DataRecord_2025.04.28_072431.xls)
[DataRecord_2025.04.28_072522.xls](https://github.com/user-attachments/files/19945429/DataRecord_2025.04.28_072522.xls)

privatecoder, 2026-10-09:
@syssi Ich habe mir mal die umfangreiche Mühe gemacht, den ganzen verfügbaren Kram aus den Weiten des Internets zusammenzutragen und dann endlose Sessions mit Claude zu führen um herauszufinden, was / wie / wo zu diesem BMS zu wissen ist: https://github.com/privatecoder/seplos-emu1101-docs

Mein Addon und Script habe ich anpassen und prüfen lassen. Intra-Pack-Kommunikation ist auch vollständig dekodiert.

Für dieses Projekt **sicher auch interessant**:

Commit: 1218206 „v2.0.0-beta1: firmware-based decoding fixes, extended status, system device" vom 08.10.2026, Release v2.0.0-beta1. Das System-Gerät kam dort zusammen mit weiteren Änderungen hinzu. Später kamen zwei Ergänzungen dazu: beta3 (f922d32) legt die Entity-IDs fest, beta5 (2ef833e) zeigt die echte Firmware-Version in den Geräte-Informationen.

  **Was das Script zusätzlich ausliest:**
  - **Woher: Es fragt den Master per Modbus RTU ab.** Gelesen werden Funktion 03, Slave-ID 1, die 21 Register ab 0x1000.Die Anfrage lautet 01 03 10 00 00 15 80 C5.
  - Über welchen Bus: über denselben RS485-Bus in der CAN-Buchse (9600 Baud), über den das Add-on die Packs abfragt. Dort beantwortet nur der Master (DIP 0) Modbus. Seplos dokumentiert
    diesen Registerblock nicht; er steht seit Firmware 16.06.04 in der Firmware.
  - **Was: Das sind die Systemwerte, die der Master über alle Packs zusammenrechnet und so auch an den Wechselrichter meldet:**
    - **Systemzustand**
    - **Laden und Entladen erlaubt**
    - **Systemspannung, Systemstrom, System-SOC, Gesamtkapazität**
    - **höchste und niedrigste Zellspannung bzw. -temperatur, jeweils mit Pack-Nummer**
    - **Lade- und Entladegrenzen (Spannung und Strom)**
    - **zwei Alarmwörter und den Pack-Kommunikationsfehler**
  - Wann: nur im Mehr-Pack-Betrieb am Bus der CAN-Buchse. Im Code ist das die Klasse SystemData, veröffentlicht unter dem MQTT-Topic seplos/system/sensors. In Home Assistant erscheint
    daraus das Gerät „Seplos BMS System", angelegt von create_system_sensors() in ha_auto_discovery.py.

Vorhanden ab Firmware 16.06.04. Laut Firmware-Vergleich ist der Modbus-Block 0x1000 × 21 dort neu hinzugekommen und in allen späteren Versionen unverändert vorhanden: Bei 16.04.06 und 16.06.03 meldet der Master auf diese Anfrage nur einen Fehler (unbekannte Startadresse). Das System-Gerät bekommt dann keine Werte.

Sieht dann so aus:
[image]