syssi/esphome-seplos-bms #122: 3 Seplos V2 on one ESP32

rira2005, 2025-02-11:
Hi,

I currently have three Seplos 2.0 BMS, each connected to its own ESP32 (so three ESP32s in total).
This feels like an overhead to me. My idea is to use a single ESP32 for all three UART connections.

But how should the ESPHome configuration look like? Does anyone have an idea?
Thanks!

substitutions:
  name: seplos-bms1-2-3
  friendly_name: Seplos BMS1-2-3
  device_description: "Monitor a Seplos BMS via RS485"
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin_bms1: GPIO4
  rx_pin_bms1: GPIO5
  tx_pin_bms2: GPIO16
  rx_pin_bms2: GPIO17
  tx_pin_bms3: GPIO22
  rx_pin_bms3: GPIO23
  id1: bms1

esphome:
  name: "seplos-bms1-2-3"
  friendly_name: Seplos BMS1-2-3

external_components:
  - source: ${external_components_source}
    refresh: 0s

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:

ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: Seplos_BMS1_2_3

captive_portal:

uart:
  id: uart_0
  # Please set the default baudrate of your Seplos BMS model here. It's sometimes 19200 baud instead of 9600.
  baud_rate: 19200
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  # The increased RX buffer size is important because
  # the full BMS response must fit into the buffer
  rx_buffer_size: 384
  debug:
    dummy_receiver: false
    direction: BOTH
    # after:
    #   delimiter: "\r"
    # sequence:
    #   - lambda: UARTDebug::log_string(direction, bytes);

seplos_modbus:
  id: modbus0
  uart_id: uart_0
  rx_timeout: 150ms

seplos_bms:
  id: bms1
  # Dip switch configuration of a single pack setup / address 0x0
  #  8    7    6    5    4    3    2    1
  # off, off, off, off, off, off, off, off
  address: 0x01
  # Known protocol versions: 0x20 (Seplos), 0x26 (Boqiang)
  protocol_version: 0x20
  seplos_modbus_id: modbus0
  update_interval: 60s

sensor:
  - platform: seplos_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    temperature_1:
      name: "${name} temperature 1"
    temperature_2:
      name: "${name} temperature 2"
    temperature_3:
      name: "${name} temperature 3"
    temperature_4:
      name: "${name} temperature 4"
    temperature_5:
      name: "${name} environment temperature"
    temperature_6:
      name: "${name} mosfet temperature"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    residual_capacity:
      name: "${name} residual capacity"
    battery_capacity:
      name: "${name} battery capacity"
    rated_capacity:
      name: "${name} rated capacity"
    state_of_charge:
      name: "${name} state of charge"
    charging_cycles:
      name: "${name} charging cycles"
    state_of_health:
      name: "${name} state of health"
    port_voltage:
      name: "${name} port voltage"
  - platform: uptime
    type: seconds
    name: Uptime Sensor       

switch:
  - platform: restart
    name: "Restart ESP"

  - platform: safe_mode
    name: "Restart ESP (Safe Mode)"

syssi, 2025-02-11:
You configuration must look like this:

```
substitutions:
  name: seplos-bms1-2-3
  friendly_name: Seplos BMS1-2-3
  device_description: "Monitor a Seplos BMS via RS485"
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin_bms1: GPIO4
  rx_pin_bms1: GPIO5
  tx_pin_bms2: GPIO16
  rx_pin_bms2: GPIO17
  tx_pin_bms3: GPIO22
  rx_pin_bms3: GPIO23

esphome:
  name: "seplos-bms1-2-3"
  friendly_name: Seplos BMS1-2-3

external_components:
  - source: ${external_components_source}
    refresh: 0s

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

api:
  encryption:

ota:
  platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: Seplos_BMS1_2_3

captive_portal:

uart:
  - id: uart_bms0
    baud_rate: 19200
    tx_pin: ${tx_pin_bms0}
    rx_pin: ${rx_pin_bms0}
    rx_buffer_size: 384
  - id: uart_bms1
    baud_rate: 19200
    tx_pin: ${tx_pin_bms1}
    rx_pin: ${rx_pin_bms1}
    rx_buffer_size: 384
  - id: uart_bms2
    baud_rate: 19200
    tx_pin: ${tx_pin_bms2}
    rx_pin: ${rx_pin_bms2}
    rx_buffer_size: 384
    
seplos_modbus:
  - id: modbus_bms0
    uart_id: uart_bms0
    rx_timeout: 150ms
  - id: modbus_bms1
    uart_id: uart_bms1
    rx_timeout: 150ms
  - id: modbus_bms2
    uart_id: uart_bms2
    rx_timeout: 150ms

seplos_bms:
  - id: bms0
    seplos_modbus_id: modbus_bms0
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s
  - id: bms1
    seplos_modbus_id: uart_bms1
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s
  - id: bms2
    seplos_modbus_id: uart_bms2
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s

sensor:
  - platform: seplos_bms
    seplos_bms_id: bms0
    total_voltage:
      name: "bsm0 total voltage"
  - platform: seplos_bms
    seplos_bms_id: bms1
    total_voltage:
      name: "bms1 total voltage"
  - platform: seplos_bms
    seplos_bms_id: bms1
    total_voltage:
      name: "bms1 total voltage"
# ...
```

rira2005, 2025-02-11:
Jep i tried this allready but get an error back...

INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/seplos-bms1-2-3.yaml...
INFO Updating https://github.com/syssi/esphome-seplos-bms.git@main
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

seplos_bms: [source /config/esphome/seplos-bms1-2-3.yaml:78]
  - id: bms0
    seplos_modbus_id: modbus_bms0
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s
    override_cell_count: 0
  - id: bms1
    
    ID 'uart_bms1' of type uart::ESP32ArduinoUARTComponent doesn't inherit from seplos_modbus::SeplosModbus. Please double check your ID is pointing to the correct value.
    seplos_modbus_id: uart_bms1
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s
    override_cell_count: 0
  - id: bms2
    
    ID 'uart_bms2' of type uart::ESP32ArduinoUARTComponent doesn't inherit from seplos_modbus::SeplosModbus. Please double check your ID is pointing to the correct value.
    seplos_modbus_id: uart_bms2
    address: 0x01
    protocol_version: 0x20

substitutions:
  name: seplos-bms1-2-3
  friendly_name: Seplos BMS1-2-3
  device_description: "Monitor a Seplos BMS via RS485"
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin_bms0: GPIO4
  rx_pin_bms0: GPIO5
  tx_pin_bms1: GPIO16
  rx_pin_bms1: GPIO17
  tx_pin_bms2: GPIO22
  rx_pin_bms2: GPIO23

esphome:
  name: "seplos-bms1-2-3"
  friendly_name: Seplos BMS1-2-3

external_components:
  - source: ${external_components_source}
    refresh: 0s

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

# Enable Home Assistant API
api:
  encryption:

ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: Seplos_BMS1_2_3

uart:
  - id: uart_bms0
    baud_rate: 19200
    tx_pin: ${tx_pin_bms0}
    rx_pin: ${rx_pin_bms0}
    rx_buffer_size: 384
  - id: uart_bms1
    baud_rate: 19200
    tx_pin: ${tx_pin_bms1}
    rx_pin: ${rx_pin_bms1}
    rx_buffer_size: 384
  - id: uart_bms2
    baud_rate: 19200
    tx_pin: ${tx_pin_bms2}
    rx_pin: ${rx_pin_bms2}
    rx_buffer_size: 384
    
seplos_modbus:
  - id: modbus_bms0
    uart_id: uart_bms0
    rx_timeout: 150ms
  - id: modbus_bms1
    uart_id: uart_bms1
    rx_timeout: 150ms
  - id: modbus_bms2
    uart_id: uart_bms2
    rx_timeout: 150ms

seplos_bms:
  - id: bms0
    seplos_modbus_id: modbus_bms0
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s
  - id: bms1
    seplos_modbus_id: uart_bms1
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s
  - id: bms2
    seplos_modbus_id: uart_bms2
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s

sensor:
  - platform: seplos_bms
    seplos_bms_id: bms0
    total_voltage:
      name: "bsm0 total voltage"
  - platform: seplos_bms
    seplos_bms_id: bms1
    total_voltage:
      name: "bms1 total voltage"
  - platform: seplos_bms
    seplos_bms_id: bms2
    total_voltage:
      name: "bms1 total voltage"

  - platform: uptime
    type: seconds
    name: Uptime Sensor       

switch:
  - platform: restart
    name: "Restart ESP"

  - platform: safe_mode
    name: "Restart ESP (Safe Mode)"

syssi, 2025-02-11:
My fault. This is the fixed / a valid version:

```
substitutions:
  name: seplos-bms1-2-3
  friendly_name: Seplos BMS1-2-3
  device_description: "Monitor a Seplos BMS via RS485"
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin_bms0: GPIO4
  rx_pin_bms0: GPIO5
  tx_pin_bms1: GPIO16
  rx_pin_bms1: GPIO17
  tx_pin_bms2: GPIO22
  rx_pin_bms2: GPIO23

esphome:
  name: "seplos-bms1-2-3"
  friendly_name: Seplos BMS1-2-3

external_components:
  - source: ${external_components_source}
    refresh: 0s

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

api:

ota:
  platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: Seplos_BMS1_2_3

captive_portal:

uart:
  - id: uart_bms0
    baud_rate: 19200
    tx_pin: ${tx_pin_bms0}
    rx_pin: ${rx_pin_bms0}
    rx_buffer_size: 384
  - id: uart_bms1
    baud_rate: 19200
    tx_pin: ${tx_pin_bms1}
    rx_pin: ${rx_pin_bms1}
    rx_buffer_size: 384
  - id: uart_bms2
    baud_rate: 19200
    tx_pin: ${tx_pin_bms2}
    rx_pin: ${rx_pin_bms2}
    rx_buffer_size: 384

seplos_modbus:
  - id: modbus_bms0
    uart_id: uart_bms0
    rx_timeout: 150ms
  - id: modbus_bms1
    uart_id: uart_bms1
    rx_timeout: 150ms
  - id: modbus_bms2
    uart_id: uart_bms2
    rx_timeout: 150ms

seplos_bms:
  - id: bms0
    seplos_modbus_id: modbus_bms0
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s
  - id: bms1
    seplos_modbus_id: modbus_bms1
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s
  - id: bms2
    seplos_modbus_id: modbus_bms2
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s

sensor:
  - platform: seplos_bms
    seplos_bms_id: bms0
    total_voltage:
      name: "bsm0 total voltage"
  - platform: seplos_bms
    seplos_bms_id: bms1
    total_voltage:
      name: "bms1 total voltage"
  - platform: seplos_bms
    seplos_bms_id: bms1
    total_voltage:
      name: "bms1 total voltage"
```

rira2005, 2025-02-11:
Okay compile works without error... but after this the esp32 runs into a bootloop, ab the to safe mode!?
Any Sugestions?

[10:58:24]ets Jun  8 2016 00:22:57
[10:58:24]
[10:58:24]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[10:58:24]configsip: 0, SPIWP:0xee
[10:58:24]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[10:58:24]mode:DIO, clock div:2
[10:58:24]load:0x3fff0030,len:1184
[10:58:24]load:0x40078000,len:13132
[10:58:24]load:0x40080400,len:3036
[10:58:24]entry 0x400805e4
[10:58:25][I][logger:171]: Log initialized
[10:58:25][C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[10:58:25][D][esp32.preferences:114]: Saving 1 preferences to flash...
[10:58:25][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[10:58:25][I][app:029]: Running through setup()...
[10:58:25][C][uart.arduino_esp32:077]: Setting up UART...
[10:58:25][C][uart.arduino_esp32:077]: Setting up UART...
[10:58:25][C][uart.arduino_esp32:077]: Setting up UART...
[10:58:25][W][uart.arduino_esp32:122]: Maximum number of UART components created already.
[10:58:25][E][component:119]: Component uart was marked as failed.
[10:58:25][E][component:164]: Component uart set Error flag: unspecified
[10:58:25][C][wifi:048]: Setting up WiFi...
[10:58:25][C][wifi:061]: Starting WiFi...
[10:58:25][C][wifi:062]:   Local MAC: 3C:71:BF:D5:38:E8
[10:58:25][D][wifi:482]: Starting scan...
[10:58:25]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[10:58:25]
[10:58:25]Core  1 register dump:
[10:58:25]PC      : 0x4018baa3  PS      : 0x00060730  A0      : 0x800e2704  A1      : 0x3ffb2570  
[10:58:25]A2      : 0x3ffb320c  A3      : 0x3ffd0704  A4      : 0x00000014  A5      : 0x0000ff00  
[10:58:25]A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0xa5064800  A9      : 0x3ffb2500  
[10:58:25]A10     : 0x00000000  A11     : 0x3f40311e  A12     : 0x00000014  A13     : 0x3f4031fe  
[10:58:25]A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
[10:58:25]EXCVADDR: 0x00000000  LBEG    : 0x400899d5  LEND    : 0x400899e5  LCOUNT  : 0xfffffffa  
[10:58:25]
[10:58:25]
[10:58:25]Backtrace:0x4018baa0:0x3ffb25700x400e2701:0x3ffb2590 0x400e1f86:0x3ffb2610 0x4018bda5:0x3ffb2630 0x400e862b:0x3ffb2650 0x400e6ea2:0x3ffb26a0 0x400e9846:0x3ffb26d0 0x400f15ce:0x3ffb2820 
[10:58:25]
[10:58:25]
[10:58:25]
[10:58:25]
[10:58:25]ELF file SHA256: 0000000000000000
[10:58:25]
[10:58:25]Rebooting...
[10:58:25]ets Jun  8 2016 00:22:57
[10:58:25]
[10:58:25]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[10:58:25]configsip: 0, SPIWP:0xee
[10:58:25]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[10:58:25]mode:DIO, clock div:2
[10:58:25]load:0x3fff0030,len:1184
[10:58:25]load:0x40078000,len:13132
[10:58:25]load:0x40080400,len:3036
[10:58:25]entry 0x400805e4
[10:58:25][I][logger:171]: Log initialized
[10:58:25][C][safe_mode:079]: There have been 1 suspected unsuccessful boot attempts
[10:58:25][D][esp32.preferences:114]: Saving 1 preferences to flash...
[10:58:25][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[10:58:25][I][app:029]: Running through setup()...
[10:58:25][C][uart.arduino_esp32:077]: Setting up UART...
[10:58:25][C][uart.arduino_esp32:077]: Setting up UART...
[10:58:25][C][uart.arduino_esp32:077]: Setting up UART...
[10:58:26][W][uart.arduino_esp32:122]: Maximum number of UART components created already.
[10:58:26][E][component:119]: Component uart was marked as failed.
[10:58:26][E][component:164]: Component uart set Error flag: unspecified
[10:58:26][C][wifi:048]: Setting up WiFi...
[10:58:26][C][wifi:061]: Starting WiFi...
[... 435 lines cut ...]
[10:58:40][C][wifi:436]:   IP Address: 192.168.253.62
[10:58:40][C][wifi:440]:   BSSID: [redacted]
[10:58:40][C][wifi:441]:   Hostname: 'seplos-bms1-2-3'
[10:58:40][C][wifi:443]:   Signal strength: -44 dB ▂▄▆█
[10:58:40][C][wifi:447]:   Channel: 11
[10:58:40][C][wifi:448]:   Subnet: 255.255.255.0
[10:58:40][C][wifi:449]:   Gateway: 192.168.253.254
[10:58:40][C][wifi:450]:   DNS1: 192.168.253.10
[10:58:40][C][wifi:451]:   DNS2: 192.168.253.11
[10:58:40][C][logger:185]: Logger:
[10:58:40][C][logger:186]:   Level: DEBUG
[10:58:40][C][logger:188]:   Log Baud Rate: 115200
[10:58:40][C][logger:189]:   Hardware UART: UART0
[10:58:40][C][captive_portal:089]: Captive Portal:
[10:58:40][C][mdns:116]: mDNS:
[10:58:40][C][mdns:117]:   Hostname: seplos-bms1-2-3
[10:58:40][C][esphome.ota:073]: Over-The-Air updates:
[10:58:40][C][esphome.ota:074]:   Address: seplos-bms1-2-3.local:3232
[10:58:40][C][esphome.ota:075]:   Version: 2
[10:58:40][C][esphome.ota:078]:   Password configured
[10:58:40][C][safe_mode:018]: Safe Mode:
[10:58:40][C][safe_mode:020]:   Boot considered successful after 60 seconds
[10:58:40][C][safe_mode:021]:   Invoke after 10 boot attempts
[10:58:40][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[10:58:40][W][safe_mode:031]: SAFE MODE IS ACTIVE

rira2005, 2025-02-11:
**1.try:**
uart:
  - id: uart_bms0
    baud_rate: 19200
    tx_pin: ${tx_pin_bms0}
    rx_pin: ${rx_pin_bms0}
    rx_buffer_size: 384

seplos_modbus:
  - id: modbus_bms0
    uart_id: uart_bms0
    rx_timeout: 150ms

seplos_bms:
  - id: bms0
    seplos_modbus_id: modbus_bms0
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s

sensor:
  - platform: seplos_bms
    seplos_bms_id: bms0
    total_voltage:
      name: "bms0 total voltage"

**works!**

2.)
uart:
  - id: uart_bms0
    baud_rate: 19200
    tx_pin: ${tx_pin_bms0}
    rx_pin: ${rx_pin_bms0}
    rx_buffer_size: 384
  - id: uart_bms1
    baud_rate: 19200
    tx_pin: ${tx_pin_bms1}
    rx_pin: ${rx_pin_bms1}
    rx_buffer_size: 384

seplos_modbus:
  - id: modbus_bms0
    uart_id: uart_bms0
    rx_timeout: 150ms
  - id: modbus_bms1
    uart_id: uart_bms1
    rx_timeout: 150ms

seplos_bms:
  - id: bms0
    seplos_modbus_id: modbus_bms0
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s
  - id: bms1
    seplos_modbus_id: modbus_bms1
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s

sensor:
  - platform: seplos_bms
    seplos_bms_id: bms0
    total_voltage:
      name: "bms0 total voltage"
  - platform: seplos_bms
    seplos_bms_id: bms1
    total_voltage:
      name: "bms1 total voltage"

**works!**

3.)
uart:
  - id: uart_bms0
    baud_rate: 19200
    tx_pin: ${tx_pin_bms0}
    rx_pin: ${rx_pin_bms0}
    rx_buffer_size: 384
  - id: uart_bms1
    baud_rate: 19200
    tx_pin: ${tx_pin_bms1}
    rx_pin: ${rx_pin_bms1}
    rx_buffer_size: 384
  - id: uart_bms2
    baud_rate: 19200
    tx_pin: ${tx_pin_bms2}
    rx_pin: ${rx_pin_bms2}
    rx_buffer_size: 384

seplos_modbus:
  - id: modbus_bms0
    uart_id: uart_bms0
    rx_timeout: 150ms
  - id: modbus_bms1
    uart_id: uart_bms1
    rx_timeout: 150ms
  - id: modbus_bms2
    uart_id: uart_bms2
    rx_timeout: 150ms

seplos_bms:
  - id: bms0
    seplos_modbus_id: modbus_bms0
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s
  - id: bms1
    seplos_modbus_id: modbus_bms1
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s
  - id: bms2
    seplos_modbus_id: modbus_bms2
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s

sensor:
  - platform: seplos_bms
    seplos_bms_id: bms0
    total_voltage:
      name: "bms0 total voltage"
  - platform: seplos_bms
    seplos_bms_id: bms1
    total_voltage:
      name: "bms1 total voltage"
  - platform: seplos_bms
    seplos_bms_id: bms2
    total_voltage:
      name: "bms2 total voltage"

produces the bootloop and safemode...
my i can try to change die address: 0x01 for the second/third bms to 0x02/0x03?, what do you think?

tia
raphael

rira2005, 2025-02-11:
this is the last working config:
uart:
  - id: uart_bms0
    baud_rate: 19200
    tx_pin: ${tx_pin_bms0}
    rx_pin: ${rx_pin_bms0}
    rx_buffer_size: 384
  - id: uart_bms1
    baud_rate: 19200
    tx_pin: ${tx_pin_bms1}
    rx_pin: ${rx_pin_bms1}
    rx_buffer_size: 384
  - id: uart_bms2
    baud_rate: 19200
    tx_pin: ${tx_pin_bms2}
    rx_pin: ${rx_pin_bms2}
    rx_buffer_size: 384

seplos_modbus:
  - id: modbus_bms0
    uart_id: uart_bms0
    rx_timeout: 150ms
  - id: modbus_bms1
    uart_id: uart_bms1
    rx_timeout: 150ms

seplos_bms:
  - id: bms0
    seplos_modbus_id: modbus_bms0
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s
  - id: bms1
    seplos_modbus_id: modbus_bms1
    address: 0x01
    protocol_version: 0x20
    update_interval: 60s
 
sensor:
  - platform: seplos_bms
    seplos_bms_id: bms0
    total_voltage:
      name: "bms0 total voltage"
  - platform: seplos_bms
    seplos_bms_id: bms1
    total_voltage:
      name: "bms1 total voltage"

if i add 
  - id: modbus_bms2
    uart_id: uart_bms2
    rx_timeout: 150ms

the device is rebooting(loop) and going to safemode...

maybe a bug?
tia
Raphael

rira2005, 2025-02-11:
HI Sebastian!
^^^^^^^^^^^^^^^^^^^^^^
**Many thanks, that did the Job**! 
Good to know the esp32 only supports 3 uart register, because i get a 4th batterypack, so i will use 2 esp32 with 2uart for the future, Pls, Close the issue!

Greetings, Raphael

syssi, 2025-02-11:
Another option is to use an Expander like this one: https://esphome.io/components/weikai.html