syssi/esphome-seplos-bms #123: Decoding new version bms tdt-6032

wojtbar, 2025-02-15:
Hello.
I have decoded data from bms tdt-6032. Please help me finish the software modification.

syssi, 2025-02-15:
What do you need? Do you like to share your work?

wojtbar, 2025-02-15:
The start of frame in tdt-6022 is 7E ( ~ ) : it's ok

wojtbar, 2025-02-15:
Start of frame in tdt-6032 is BC (   ?  ), what character to insert.

wojtbar, 2025-02-15:
" >>  bc3031303134363432453030323031464433360d "  ( �01014642E00201FD36  )  "

wojtbar, 2025-02-15:
" << ~01014F4230940001100CD40CD40CD50CD60CD60CD60CD60CD60CD60CD70CD70CD70CD70CD80CD80CD5060B4E0B370B2C0B2C0B2C0B3100191489167D0049C6003180E8148F00001E0380000000000002DD4B "

wojtbar, 2025-02-15:
The response from tdt-6032 starts with 7E,
version 01, then it's known. The remaining data is identical
to tdt-6022. There's just a longer frame with 0000 0000 0002 at the end, checksum DD4B. I just don't know what 0002 is at the end.

Baalfrog, 2025-02-15:
Dear both. At first congratulation for your work. 
I have a xdzn-6032 (seems to be a tdt-6032). I’m really interested by this subject. I have to interface the BMS with jeedom thru Mqtt. I have some rs232, rs485 and CAN to uart. I have a really limited knowledge on software but if needed I can provide some information (with your help). I have the windows software which communicate with the BMS (only works thru rs232). Tell me if I can help

Baalfrog, 2025-02-19:
Hello, I guess I've highlited the issue : 0xBC is converted to "c2 bc" on UTF-8. 0xBC means a quarter (1/4).

wojtbar, 2025-02-19:
Yes. But in the case of tdt-6022 I send pure hex and get the correct response.

wojtbar, 2025-02-19:
[image]

wojtbar, 2025-02-19:
Jak możesz to podepnij rs232 do laptopa i baterii .
Programem  Advanced Serial Port Monitor 4.4.21ver. wyślij bc303130313436343245303032303146443336 jako hex z zakończeniem lini CR .

Baalfrog, 2025-02-19:
Me again. modification from tdt_modbus.cpp seems to work. I have at least an answer from the BMS. 
regarding your message. It's exactely the same than those sent by the ESP. 
Answer is : 
bc 30 31 30 31 34 36 34 32 45 30 30 32 30 31 46 44 33  ¼01014642E00201FD36.
36 0d                                                                                                 
7e 30 31 30 31 34 46 34 32 46 30 39 38 30 30 30 31 31  ~01014F42F09800011
30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30  000000000000000000
30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30  000000000000000000
30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30  000000000000000000
30 30 30 30 30 30 30 30 30 30 30 30 36 30 30 30 30 30  000000000000600000
30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30  000000000000000000
30 30 30 30 30 30 30 30 30 32 42 43 30 30 30 36 44 36  0000000002BC0006D6
30 30 30 30 30 36 44 36 30 30 30 30 30 30 30 30 30 32  000006D60000000002
38 30 33 38 30 30 30 30 30 30 30 30 30 00 00 00 02 30  8038000000000....0
30 30 30 45 31 33 31 0d 7e 30 31 30 31 34 36 30 39 30  000E131.~010146090
30 30 30 46 44 41 42 0d 7e 30 31 30 31 34 36 30 34 30  000FDAB.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d 7e 30 31 30 31 34 36 30 34 30  000FDB0.~010146040
30 30 30 46 44 42 30 0d                                                   000FDB0.

Please note the BMS is not connected to battery or temp sensors.

wojtbar, 2025-02-19:
Not  really.
167 payload[0] : 0xBC; // SOF (0xBC)  ????
167 payload[0] = 0xBC;

wojtbar, 2025-02-19:
Something strange. I changed myself at tdt-6022:

166 std::string payload(1,'0')
167 payload[0] = 0xBC; // SOF (0xBC)

And this is the effect.
[19:39:08][D][uart_debug:158]: >>> "\xBC01014642E00201FD36\r"
[19:39:08][D][uart_debug:158]: <<< "~25014F42808000010E0F7C0F7B0F7E0F800F7A0F7C0F7D0F7F0F7E0F7F0F7F0F7F0F7E0F80060B150AD80AC80AC80AC80AD0FDCE15B043B500498000964E20159D0C705C0409E0C9\r"
[19:39:08][I][tdt_bms:030]: Telemetry frame (70 bytes) received
[19:39:08][D][sensor:093]: 'cell voltage 1': Sending state 3.96400 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'cell voltage 2': Sending state 3.96300 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'cell voltage 3': Sending state 3.96600 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'cell voltage 4': Sending state 3.96800 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'cell voltage 5': Sending state 3.96200 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'cell voltage 6': Sending state 3.96400 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'cell voltage 7': Sending state 3.96500 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'cell voltage 8': Sending state 3.96700 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'cell voltage 9': Sending state 3.96600 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'cell voltage 10': Sending state 3.96700 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'cell voltage 11': Sending state 3.96700 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'cell voltage 12': Sending state 3.96700 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'cell voltage 13': Sending state 3.96600 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'cell voltage 14': Sending state 3.96800 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'min cell voltage': Sending state 3.96200 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'max cell voltage': Sending state 3.96800 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'max voltage cell': Sending state 4.00000  with 0 decimals of accuracy
[19:39:08][D][sensor:093]: 'min voltage cell': Sending state 5.00000  with 0 decimals of accuracy
[19:39:08][D][sensor:093]: 'delta cell voltage': Sending state 0.00600 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'average cell voltage': Sending state 3.96571 V with 3 decimals of accuracy
[19:39:08][D][sensor:093]: 'environment temperature': Sending state 10.70000 °C with 0 decimals of accuracy
[19:39:08][D][sensor:093]: 'mosfet temperature': Sending state 4.60000 °C with 0 decimals of accuracy
[19:39:08][D][sensor:093]: 'temperature 1': Sending state 3.00000 °C with 0 decimals of accuracy
[19:39:08][D][sensor:093]: 'temperature 2': Sending state 3.00000 °C with 0 decimals of accuracy
[19:39:08][D][sensor:093]: 'temperature 3': Sending state 3.00000 °C with 0 decimals of accuracy
[19:39:08][D][sensor:093]: 'temperature 4': Sending state 3.80000 °C with 0 decimals of accuracy
[19:39:08][D][sensor:093]: 'current': Sending state -5.62000 A with 2 decimals of accuracy
[19:39:08][D][sensor:093]: 'total voltage': Sending state 55.52000 V with 2 decimals of accuracy
[19:39:08][D][sensor:093]: 'power': Sending state -312.02240 W with 2 decimals of accuracy
[19:39:08][D][sensor:093]: 'charging power': Sending state 0.00000 W with 2 decimals of accuracy
[19:39:08][D][sensor:093]: 'discharging power': Sending state 312.02240 W with 2 decimals of accuracy
[19:39:08][D][sensor:093]: 'residual capacity': Sending state 173.33000 Ah with 2 decimals of accuracy
[19:39:08][D][sensor:093]: 'battery capacity': Sending state 188.15999 Ah with 2 decimals of accuracy
[19:39:08][D][sensor:093]: 'charging cycles': Sending state 150.00000  with 0 decimals of accuracy
[19:39:08][D][sensor:093]: 'rated capacity': Sending state 200.00000 Ah with 2 decimals of accuracy
[19:39:08][D][sensor:093]: 'port voltage': Sending state 55.33000 V with 2 decimals of accuracy
[19:39:08][D][sensor:093]: 'sampling voltage': Sending state 3.18400 V with 2 decimals of accuracy
[19:39:08][D][sensor:093]: 'state of charge': Sending state 92.00000 % with 1 decimals of accuracy

Baalfrog, 2025-02-19:
It seems the 6022 understand the 6032 start of frame. Then fc or bc seems to be accepted. I tried on mine. He didn’t answer. Note the \x doesn’t appear on the uart

wojtbar, 2025-02-19:
My software is described in the tdt-bms.yaml file - and the appropriate directories copied to the HA /config/esphome directory.

external_components:
 - source:
     type: local
     path: tdt-bms
   components: [ tdt_bms, tdt_modbus ]
   refresh: 0s

Baalfrog, 2025-02-20:
As I'm not using HomeAssistant but Jeedom I'll have to perform lot of modification (Mqtt information are not the same, then I have many component which are appearing when connecting the ESP). 

Regarding data sent by the BMS it seems to be conform (I still have to connect the BMS with temp sensors and batteries). 
I confirm the 6032 is sending more data than the 6022. I have to make a print srceen but the debug mode on ESP is displaying "Unhandled data received". I'll send asap those data.

Baalfrog, 2025-02-21:
Hi, 
It seems there is some strange answers from the BMS. Thru Esp debug I have lots of : [tdt_bms:021]: Unhandled data received (data_len: 0x00): 01.01.46.04.00.00 (6)

I tried to return to the original software (SOF : 0xFC) The 6032 seems to anwser (it didn't in the past). There is still many unhandled data received. 

according yo documents it seems to be a CID2 invalid : 01 protocol 01 addr 46 LIFEPO4 04 CID2 invalid

wojtbar, 2025-02-22:
Hello , on the second bms also the same:
[20:28:54][D][uart_debug:158]: >>> "\xBC01014642E00201FD36\r"
[20:28:54][D][uart_debug:158]: <<< "\xBC01014642E00201FD36\r"
[20:28:54][W][tdt_bms:021]: Unhandled data received (data_len: 0x02): 01.01.46.42.E0.02.01 (7)

wojtbar, 2025-02-22:
For bms tdt-6022

I have dumped wireshark pcapng:
1. from xdzn-2.4 software
2. from xdzn-3.4 software

from 1. data
~25014642E00201FD30
~25014F42808000010E0F1A0FC0F1C0F1C0F170F160F170F190F190F1A0F1A0F1A0F1A0F1B060B030AD50AC80ABE0AC80ACD0EC215232DC800451900984E2015100BD9420409E16F

from 2. data
�01014642E00201FD36
~25014F42808000010E0F2C0F2B0F2E0F300F290F280F290F2B0F2D0F2A0F2A0F2B0F2C0F2D060B040AD80AC80AC80AC80AD60EC2153C2F3700451900984E20152C0CA7440409E152

I will try again with the xdzn-3.6 software.
It just doesn't want to start.

wojtbar, 2025-02-22:
TDT 6022 for XDZN 3.6 software

�01014642E00201FD36
~25014F42808000010E0F720F720F740F730F6D0F6E0F6D0F700F700F700F700F700F700F71060B0C0AE00AD20AC80AD20AD80EB6159D355D00451900984E2015890C914D0409E195

wojtbar, 2025-02-22:
The software does not accept data.

[image]

Wireshark captures all data.

wojtbar, 2025-02-22:
So if TDT-6032 works properly on xdzn 3.6 software version then it should not work on version 3.4 and 2.4.
Can someone check this.

Baalfrog, 2025-02-23:
On my side BMS tool v3.6 is working with 6032. 
v3.4 is working too. 
Note I updated the 6032 firmware with 6032_20016S000_L_49_m.
perhaps it's the reason it answer now with esp.

wojtbar, 2025-02-23:
And you have version 2.4. Do you know how to do a packet capture via wireshark (.pcapng)?
I would like to analyze them.

Baalfrog, 2025-02-23:
I don't have version 2.4.I'll download it and test

Baalfrog, 2025-02-23:
Thanks, I just tried, so v2.4 is connected, the horizontal bar is moving (close to the com port selection) but any bos is filled. 
It seems to confirm 6032 is not compliant with v2.4.
thus sniffer the BMS is responding (with protocol 25 ?? ).

So on my side :
v2.4 : ~25014642E00201FD30. => ~25014600004C0000100101010101010101010101010101010106010101010001000100000000000000000000EF3A.

v3.4 : ¼01014642E00201FD36. => ~01014F42F098000110000000000000000000000000000000000000000000000000000000000000000006000000000000000000000000000000002BC0006D6000006D600000000028038000000000....0000E131.

v3.6 : ¼01014642E00201FD36. => ~01014F42F098000110000000000000000000000000000000000000000000000000000000000000000006000000000000000000000000000000002BC0006D6000006D600000000028038000000000....0000E131.

So v3.x send and receive the same data. According to datasheet from here I don't know why I receive 01014F and not 010146 = 46 shall be battery type (LIFEPO4).

wojtbar, 2025-02-23:
tdt 6032   v2.4 : ~25014642E00201FD30 
<< ~01 01 4F 42 F0 98 00 01 10 

000000000000000000000000000000000000000000000000000000000000000006000000000000000000000000000000002BC0006D6000006D600000000028038000000000....0000E131

tdt-6022   v2.4 : ~25014642E00201FD30
 << ~25 01 4F 42 80 80 00 01 0E 

0F1A0FC0F1C0F1C0F170F160F170F190F190F1A0F1A0F1A0F1A0F1B060B030AD50AC80ABE0AC80ACD0EC215232DC800451900984E2015100BD9420409E16F

wojtbar, 2025-02-23:
See a few posts above, for each version the same 4F

wojtbar, 2025-02-23:
Or return header ~01014F42F098000110
(7e 30 31 30 31 .....)

tdt_modbus.cpp
(89)    if (raw[0] != 0x7E) {

(165)   //std::string payload = "~";  // SOF (0x7E)
(166)  std::string payload(1,'0');  // SOF (0xBC)
(167)  payload[0] =  0xBC; // SOF (0XBC)
(168)  payload.append(byte_to_ascii_hex(data.data(), frame_len));

wojtbar, 2025-02-23:
You can also try in
/tdt_bms/__init__.py

(13) DEFAULT_PROTOCOL_VERSION = 0X01
(14) DEFAULT_ADDRESS = 0X01

wojtbar, 2025-02-23:
Send me the entire answer from version 3.4

Baalfrog, 2025-02-23:
others anwser from 6032 with software v3.4 are : 
~010146090000FDAB.
~01014F91F0100000000000000000FA83.
~01014FC1202C363033325F3230303136533030305F4C5F343900018CF462.
~010146006028363033325F3230303136533030305F4C5F343900F569.
~010146040000FDB0.
~01014F4470540001100101010101010101010101010101010106010101010001000100000000C0000018000000000200ED90.
~01014FF27018000000000000000000000000F8FC.
~01014FC2F05C36303332363031363230373237303030310000002020202000000000000000000000000000000000FFFFFFFF0000E B0B.
~01014600B05036303332363031363230373237303030310000002020202000000000000000000000000000000000EE37.

Baalfrog, 2025-02-23:
> Send me the entire answer from version 3.4

Perhaps I idin't understood your request. on my previous message there is "...." <= decoded frame are 00 00 00 02

Baalfrog, 2025-02-23:
> See a few posts above, for each version the same 4F

on 6032 : 
v2.4 : ~25014**6** <= 7E 32 35 30 31 34 **36**
v3.4 : ~01014**F** <= 7E 30 31 30 31 34 **46**
v3.6 : ~01014**F** <= 7E 30 31 30 31 34 **46**

Baalfrog, 2025-02-23:
> You can also try in /tdt_bms/**init**.py
> 
> (13) DEFAULT_PROTOCOL_VERSION = 0X01 (14) DEFAULT_ADDRESS = 0X01

Beware here is not the ESP but the windows software, 2.4, 3.4 and 3.6

wojtbar, 2025-02-23:
response to frame \01014642E00201FD36

wojtbar, 2025-02-23:
In windows xdzn-v3.4 software
After setting com port and pressing run take a screenshot . Don't press anything else.

Baalfrog, 2025-02-24:
Hi, I won't be able to do this test this week (business trip).

Baalfrog, 2025-02-25:
Hi, What will you say if I say I just receive from the manufacturer the CAN and RS485 protocol datasheet ?

Baalfrog, 2025-02-25:
Here is the link to download the RS485 and CAN protocol for XDZN BMS. 
https://drive.google.com/drive/folders/16qVqtz_XlFcYGk2lvxerKafr0BRC0yMP?usp=drive_link.

Perhaps it will be easier to comunicate directly thru RS485 due to standard.

Baalfrog, 2025-02-25:
here is the print screen

[image]

Baalfrog, 2025-03-08:
Hi, I tried to connect the BMS thru RS485 and to modify the ESP program according to datasheet (protocol 20 and address 2).
the BMS answers always without any error on CIP2. but now I have each time an issue on CRC. 
Sent data : ~20124642E00212FD31\r
received data : ~20124600B07E1102100000000000000000000000000000000000000000000000000000000000000000050000000000000000000000000000B58004FFFF000001B5800445C0E51E\r

with each time : CRC check failed! 0xE54E != 0xE51E

I calculate the CRC from the received data and it is correct to those sent by the BMS (E51E)

wojtbar, 2025-03-08:
What reads this code.
If error data  01014644E00201FD34

Baalfrog, 2025-03-08:
the answer is ~200146900000FDAA

Baalfrog, 2025-03-08:
If I pass thru the RS232 I have again many many pages with ~200146040000FDAF.

Baalfrog, 2025-03-08:
It seems sometimes the crc is correct then I have the updated data. 
I added on tdt_modbus.cpp a line before the CRC warning message to check the data lenght (data_len).
when CRC is bad the data_len is 0x8E. when CRC is correct data_len is 0x8F.
I check if there is a data difference but I can't find any diffence. 
good anwser data_len is :0x8F ~20124600B07E1102100000000000000000000000000000000000000000000000000000000000000000050000000000000000000000000000B58004FFFF000001B5800445C0E51E\r
CRC error answer data_len is 0x8E  ~20124600B07E1102100000000000000000000000000000000000000000000000000000000000000000050000000000000000000000000000B58004FFFF000001B5800445C0E51E\r

Baalfrog, 2025-03-08:
(note I whanged the address according to datasheet from 01 to 12

Baalfrog, 2025-03-08:
So after many many tries I found the reason for CRC error : I uncommented "dummy_receiver: true" in tdt-bms.yaml. 
I just tried to recomment it, Now I have no CRC error

Baalfrog, 2025-03-08:
now I'll have to undestand the second part on my needs : change the mqtt configuration to have it compliant with jeedom and not home assistant. 
Have you an idea where does I have to check ? it seems there is some files inside folder "build". I'm pretty sure it is automatically generated then my modification will be discharged. 
My need is to have all information inside an unique topic (for example : "BMS/Daly/{Temp1}" where Temp1 is loaded with the value)

Baalfrog, 2025-05-06:
Dear All, My BMS in installed and the ESP too. 
up to now close to everything is working (sometimes I had to divide 10 or 100 times some measurements). 
I notice two  issues :
- Battery_capacity : I have 65535 as value. as this value is FFFF I assume there is a calculathion issue.
- residual_capacity : I don't know how to handle the value : I have 479.04 On the display I have 96% left on a 280Ah battery
@wojtbar : do you have the same ?

wojtbar, 2025-05-11:
Hello. Can you send me the answers to question ~20124642E00212FD31\r .

Baalfrog, 2025-05-12:
Hello, thank you for your reply. 
I worked a lot during the weekend on this subject. 
I found several things. 
1- on rs485, the BMS adapt itself according to the inverter settings ( I connected the esp on inverter rs485 plug, not on battery rs485 plug). Is is only working with the esp program when the setting is pylontech. 
2- pylontech (inverter) protocol is a bit different than the rs232 : battery capacity is sent thru 2 modus to keep compatibility with old inverter : according to the battery type ( if 4 means >= 65ah) remain capacity and total capacity are ffff, there is others data sent after to give the correct information for battery >= 65ah). 
3- it seems there is an issue on current value sent when the setting is pylontech : current is sent on mA on signed int. Then from -32768mA to +32768mA. As I have 280Ah batteries and as I charge them up to 80A ( and can have a discharging current up to 180A) there is some wrong information sent. I found how to pass thru to add a request on CID2=62
- as the pylontech protocol is not the same than rs232 there is no soc information. By using CID2 too this issue have been solved.
- to be sure to send correct power I recovered the pack voltage from CID2=62  and transferred all power calculation on this too. 

Sources from pylontech found there : https://github.com/Frankkkkk/python-pylontech/blob/master/RS485-protocol-pylon-low-voltage-V3.3-20180821.pdf

As I have a really low knowledge on software my modification are really dirty but if requested I can provide them.

Baalfrog, 2025-05-12:
> Hello. Can you send me the answers to question ~20124642E00212FD31\r .

My battery pack is completely finished then the esp is inside the battery pack. I can’t have an access to the rs485 port. Is there a way to remotely send this ?

wojtbar, 2025-05-17:
Dnia poniedziałek, 12 maja 2025 06:47:23 CEST Baalfrog pisze:
> Baalfrog left a comment (syssi/esphome-seplos-bms#123)
> 
> > Hello. Can you send me the answers to question ~20124642E00212FD31\r .
> 
> My battery pack is completely finished then the esp is inside the battery
> pack. I can’t have an access to the rs485 port. Is there a way to remotely
> send this ?

ESP32 uart
add

 uart:
 .
 . 
 rx_buffer_size: 384
 debug:
     dummy_receiver: false
     direction: BOTH
     after:
        delimiter: "\r"
     sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
interval:
    - interval: 30s
      then:
        - uart.write: "~20124642E00212FD31\r"

compile, download log
and then go back to what it was before adding.

wojtbar, 2025-05-17:
I have an old inverter and when I set the bms to pylontech and the inverter to pylontech then with my battery (li-ion 14 s) the inverter has an error: high battery voltage (14s 4.05V cell and 56.7V pack). Only voltronic works correctly on the bms and Lib on the inverter.

Baalfrog, 2025-05-18:
Hi Wojtbar, thanks for your help. On my side I have two East SML 4kw in parallel mode but they have no battery connection (only USB and UART connexion to a computer). that's why I had to retrieved some informations on a MQTT server to manage the battery and by using Jeedom sending commands to inverter to pas in bypass mode (they are offgrid inverters). 
The answer from your request is : 
<<< "~20124600B07E1102100DE10DF00DE80DEC0DE70DDC0DEB0DD90DDD0DDA0DFD0DFE0E010E010DFD0E00050BD10B9A0B900B900B9B024EDEDA2E6E04FFFF0008042E6E042E6EE0AF\r"

wojtbar, 2025-05-18:
I have one more request. I need a screenshot of the XDZN software connected to the bms.

wojtbar, 2025-05-18:
If it's from my tdt-bms, why was there data from 6 thermometers before and now there are 5? 
In fact, there are 6 of them.

wojtbar, 2025-05-18:
> here is the print screen
> 
> [image]

Here you can see that the temperature must be in a different place.

[17 more replies]