syssi/esphome-seplos-bms #128: No data with this adapter

broekieman, 2025-03-24:
I tried to make this work but no response sofar.

Could it be this adapter is not suited for this? This adapter works with fine with when I plug it in my Victron Cerbo and the rs485 port.

What do I need to make it work with esphome? 

[image]

broekieman, 2025-03-24:
[image]
Does this go into battery rj45 rs485?

syssi, 2025-03-24:
Please ignore the green connector. You have to attach a RJ45 connector to the B-/A+ lines.

broekieman, 2025-03-26:
> I tried to make this work but no response sofar.
> 
> Could it be this adapter is not suited for this? This adapter works with fine with when I plug it in my Victron Cerbo and the rs485 port.
> 
> What do I need to make it work with esphome?
> 
> [image]

Is this adapter different inside from yours? I assemble a USB extention cable to it for power, rx, tx, but no data.

syssi, 2025-03-26:
Yes. For sure!

```
# Your plug
RS485 communication <--[Your Converter]--> USB interface

# My converter
RS485 communication <--[My Converter]--> UART-TTL interface
```

Does this clarification help? A D+/D- pin of the USB interface is different to the RX/TX pin of an UART-TTL interface.

broekieman, 2025-03-26:
I have 2 packs with Seplos bms. Is it possible to see both when parallel connected?

syssi, 2025-03-26:
I must admit: I don't know. It's possible to read both packs if they act as slaves (f.e. `0x01`, `0x02`). As soon one pack acts as master it aggregates the `state of charge` of both packs and it's hard to ask the master about non aggregated details if I remember correctly.

There are people around with more device specific knowledge here.

Privatecoder, 2025-03-26:
@broekieman at least multiple Seplos 2.0 / V16 BMS can be read when isolating the CAN and RS485 signal on the CAN-Port of the Master and then daisy-chaining the _RS485-part_ of the slaves and connecting an ESP or any RS485 adapter using a Baud of 9600.

Check [this #116](https://github.com/syssi/esphome-seplos-bms/issues/116).

syssi, 2025-03-26:
@Privatecoder Thanks for helping me out. I lost some track here.

broekieman, 2025-03-26:
I have Seplos V3 and read on forums it's not easy to daisy chain multiple cells.
Hope someone can help me out.

syssi, 2025-04-01:
Yes it will work. Flow control using a dedicated GPIO is required here. The schema will look like this:

```
               RS485                        UART
┌─────────┐              ┌─────────────┐           ┌─────────────────┐
│         │              │           DI│<--------->│TX               │
│         │<-----B- ---->│  RS485    DE│<--\       │         ESP32/  │
│    BMS  │<---- A+ ---->│  to TTL   RE│<---+----->│GPIO0   ESP8266  │
│         │<--- GND ---->│  module   RO│<--------->│RX               │
│         │              │             │           │                 │
│         │              │          VCC│<--------->│3.3V          VCC│<--
│         │              │          GND│<--------->│GND           GND│<--
└─────────┘              └─────────────┘           └─────────────────┘
```

DE and RE must be connected and attached to a free GPIO + the YAML configuration must be extended:

```
seplos_modbus:
  id: modbus0
  uart_id: uart_0
  rx_timeout: 150ms
  flow_control: GPIO0
```

broekieman, 2025-04-01:
I get this error :

[image]

syssi, 2025-04-01:
Oh, the new parameter is called `flow_control_pin`.

broekieman, 2025-04-02:
> DE and RE must be connected and attached to a free GPIO + the YAML configuration must be extended:

You wrote:
DE and RE must be connected and attached to a free GPIO + the YAML configuration must be extended:

I have a ESP32 board. Where should i connect TX and RX to?

               RS485                        UART
┌─────────┐              ┌─────────────┐           ┌─────────────────┐
│         │              │           DI│<--------->│TX               │
│         │<-----B- ---->│  RS485    DE│<--\       │         ESP32/  │
│    BMS  │<---- A+ ---->│  to TTL   RE│<---+----->│GPIO0   ESP8266  │
│         │<--- GND ---->│  module   RO│<--------->│RX               │
│         │              │             │           │                 │
│         │              │          VCC│<--------->│3.3V          VCC│<--
│         │              │          GND│<--------->│GND           GND│<--
└─────────┘              └─────────────┘           └─────────────────┘

broekieman, 2025-04-02:
Can I use this max485 module with de Seplos sniffer and which code for that? 

https://github.com/DpunktS/seplos_v3_sniffer/blob/main/seplos_v3_sniffer.Yaml

syssi, 2025-04-03:
Any GPIO can be used as UART pins because the ESP32 is equipped with a GPIO matrix and can route the hardware UARTs to any pin set. The examples are using `GPIO16/GPIO17`:

https://github.com/syssi/esphome-seplos-bms/blob/main/esp32-example.yaml#L5-L6

broekieman, 2025-04-03:
I know. I have it working with the Seplos multi pack from DpunktS.
When I add:

seplos_modbus:
  id: modbus0
  uart_id: uart_0
  rx_timeout: 150ms
  flow_control_pin: GPIO0

The config gives an error. His script is different from yours.

syssi, 2025-04-03:
The DpunktS project doesn't support flow control and it's not required because the setup is `read only`. You have to make sure your RS485 converter is in "RX mode". The flow control pin of the converter is used to toggle between RX and TX mode.

broekieman, 2025-04-03:
Please show schematic diagram for DpunktS script with the max485 module. I can't get it to work. 

This module works fine. 
[image]

syssi, 2025-04-03:
Do you own multiple battery banks and one battery bank is the master?

broekieman, 2025-04-04:
Thank you for your help, it works flawless now.
Do you have homeassistant card with all cell info?

syssi, 2025-04-04:
No. There is no lovelace card I'm aware of.

closed 2025-04-04 by syssi