syssi/esphome-seplos-bms #144: BLE integration is not working properly with Seplos BMS v3

inigoml, 2025-07-16:
For not it appears does not recognize Seplos v3 BLE as a valid device. SEPLOS fimware v1.4

[image]

Config: `substitutions:
  name: seplos-bms-ble
  device_description: "Monitor a Seplos Battery Management System via BLE"
  external_components_source: github://syssi/esphome-seplos-bms@main
  mac_address: c0:d6:3c:4c:71:1d

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-seplos-bms"
    version: 1.0.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "Seplos-Bms Fallback Hotspot"
    password: !secret hotspot

captive_portal:

# Enable Home Assistant API
api:
  reboot_timeout: 0s
  encryption:
    key: "vUZz0wdIzkBLlSkyJ+yLYCSTIPUCbTtYIdKv+SLXQnk="

ota:
  - platform: esphome
    password: "XXXXXXXXXXXXXXXXXXXXX"

logger:
  level: DEBUG

esp32_ble_tracker:

ble_client:
  - mac_address: ${mac_address}
    id: client0

seplos_bms_ble:
  - ble_client_id: client0
    id: bms0
    update_interval: 10s

binary_sensor:
  - platform: seplos_bms_ble
    seplos_bms_ble_id: bms0
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    limiting_current:
      name: "${name} limiting current"
    online_status:
      name: "${name} online status"

sensor:
  - platform: seplos_bms_ble
    seplos_bms_ble_id: bms0
    voltage_protection_bitmask:
      name: "${name} voltage protection bitmask"
    current_protection_bitmask:
      name: "${name} current protection bitmask"
    temperature_protection_bitmask:
      name: "${name} temperature protection bitmask"
    error_bitmask:
      name: "${name} error bitmask"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    capacity_remaining:
      name: "${name} capacity remaining"
    state_of_charge:
      name: "${name} state of charge"
    nominal_capacity:
      name: "${name} nominal capacity"
    charging_cycles:
      name: "${name} charging cycles"
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    average_cell_temperature:
      name: "${name} average cell temperature"
    ambient_temperature:
      name: "${name} ambient temperature"
    mosfet_temperature:
      name: "${name} mosfet temperature"
    state_of_health:
      name: "${name} state of health"
    port_voltage:
      name: "${name} port voltage"
    battery_capacity:
      name: "${name} battery capacity"
    temperature_1:
      name: "${name} temperature 1"
    temperature_2:
      name: "${name} temperature 2"
    temperature_3:
      name: "${name} temperature 3"
    temperature_4:
      name: "${name} temperature 4"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"

text_sensor:
  - platform: seplos_bms_ble
    seplos_bms_ble_id: bms0
    software_version:
      name: "${name} software version"
    device_model:
      name: "${name} device model"
    hardware_version:
      name: "${name} hardware version"
    voltage_protection:
      name: "${name} voltage protection"
    current_protection:
      name: "${name} current protection"
    temperature_protection:
      name: "${name} temperature protection"
    errors:
      name: "${name} errors"

    `

syssi, 2025-07-16:
Bad news: The `seplos_bms_ble` component is the implementation of the v2 protocol. The v3 is different and not supported yet unfortunately.

inigoml, 2025-07-16:
I was fearing about it. ;-)

inigoml, 2025-07-16:
> Are you able to provide a btsnoop capture of the BLE communication between the BMS and the Android app?

Could try but I've never used btsnoop.... if no extra hardware is needed, count with it.

syssi, 2025-07-16:
All you need is an Android phone with developer mode enabled (no rooting required). The btsnoop capture will be part of a bug report bundle. See https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa