syssi/esphome-seplos-bms #156: 4 Seplos V2 on one ESP32

rira2005, 2025-07-29:
Hi Sebastian,

I’d like to revisit [this issue](https://github.com/syssi/esphome-seplos-bms/issues/122).

Is there any way to use 4 serial ports on an ESP32 without using an expander (like [this one](https://esphome.io/components/weikai.html))?

If not, do you have a working YAML configuration for the expander card?

Thanks in advance,
Raphael

syssi, 2025-07-29:
I've pushed an example to the project: https://github.com/syssi/esphome-seplos-bms/blob/main/tests/esp32-uart-expander-example.yaml

closed 2025-07-29 by syssi

rira2005, 2025-07-29:
what WeiKai should i usw...

Chip | Bus | UART | GPIO
WK2132-ISSG | S/I | 2 |  
WK2212-IQNG | S/I | 2 | 8
WK2124-ISSG | S | 4 |  
WK2204-IQNG | S/I | 4 |  
WK2168-IQPG | S/I | 4 | 8

syssi, 2025-07-29:
| Chip | Bus | UART | GPIO | Explanation | |
| ------ | ------ | ---| ---- | --- | --- |
| WK2132-ISSG | S/I | 2 |   | 2 UARTs, SPI+I2C supported  | |
| WK2212-IQNG | S/I | 2 | 8 | 2 UARTs, SPI+I2C supported | |
| WK2124-ISSG | S | 4 |   | 4 UARTs, **SPI**+I2C supported    |  <-- OK |
| WK2204-IQNG | S/I | 4 |   |   4 UARTs, **SPI**+I2C supported   |  <-- OK |
| WK2168-IQPG | S/I | 4 | 8 |   4 UARTs, **SPI**+I2C supported   |  <-- OK |